esphome/issues #2941: Cannot use esp32_ble_tracker on ESP32C3

rpatel3001, 2022-01-15:
### The problem

Trying to use the BLE tracker hub on the C3 hits a couple roadblocks. First, it requires changing /config/.esphome/build/node-name/sdkconfig.node-name by either 1) setting `CONFIG_BT_BLE_50_FEATURES_SUPPORTED=n` or 2) setting `CONFIG_BT_BLE_42_FEATURES_SUPPORTED=y`. Either one allows the build to compile and has the same second problem.

The second problem is that builds including esp32_ble_tracker refuse to be uploaded via OTA. The esphome log shows:
```
========================= [SUCCESS] Took 40.25 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.0.180
INFO Uploading /config/.esphome/build/node-name/.pioenvs/node-name/firmware.bin (1293824 bytes)
ERROR Error receiving acknowledge binary size: timed out
```

and the device log (set to VERY_VERBOSE) shows
```
[21:37:37][D][ota:136]: Starting OTA Update from 192.168.0.10...
[21:37:37][V][ota:167]: OTA features is 0x01
[21:37:37][V][ota:250]: OTA size is 1293824 bytes
INFO node-name.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for node-name.local
WARNING Disconnected from API
INFO node-name.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to node-name.local
```

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32-IDF

### Board

M5Stack Stamp C3

### Component causing the issue

esp32_ble_tracker

### Example YAML snippet

```yaml
esphome:
  name: node-name

esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: esp-idf

logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "xxx"
  password: "xxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Fallback Hotspot"
    password: "xxx"

esp32_ble_tracker:
```

DanteBC, 2022-01-18:
Can confirm. ESP32-C3 fails to compile with esp-idf framework. Haven't tried disabling ONFIG_BT_BLE_50_FEATURES_SUPPORTED yet though

rpatel3001, 2022-01-21:
Pretty sure this has to do with the binary size being too large for OTA. If I take a configuration that works and pad the binary size with a large string buffer, the OTA starts to fail in the same way somewhere around a firmware.bin size of 1.1 to 1.2 MB. 

Flashing over serial isn't working, I probably have the wrong config for esptool and espflasher didn't work. Flashing tasmota and uploading through its web portal works fine, even for binaries including esp_ble_tracker (after the sdkconfig mod above) and I can see broadcasts from various devices. Unfortunately the default scan_parameters seem to prevent wifi from connecting and I get the message
```
[V][esp-idf:000]: Error! Should use default active scan time parameter for WiFi scan when Bluetooth is enabled!!!!!!
```

rpatel3001, 2022-01-21:
got an OTA to work by pushing a build without BLE with the following options, then pushing the build with BLE
```
      CONFIG_BOOTLOADER_WDT_ENABLE: n
      CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE: y
      CONFIG_ESP_INT_WDT: n
      CONFIG_ESP_TASK_WDT: n
      CONFIG_INT_WDT: n
      CONFIG_TASK_WDT: n
```
in esp32: framework: sdkconfig_options:

Probably not all of them are required but I just went wild on everything that said WDT

klim123123, 2022-01-21:
please provide full config.

> got an OTA to work by pushing a build without BLE with the following options, then pushing the build with BLE
>
> ```
>       CONFIG_BOOTLOADER_WDT_ENABLE: n [...]

rpatel3001, 2022-01-21:
Push this first:
```
esphome:
  name: test1

esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_BOOTLOADER_WDT_ENABLE: n
      CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE: y
      CONFIG_ESP_INT_WDT: n
      CONFIG_ESP_TASK_WDT: n
      CONFIG_INT_WDT: n
      CONFIG_TASK_WDT: n

ota:
  password: ""

wifi:
  ssid: "xxx"
  password: "xxx"
  use_address: test.lan

  ap:
    ssid: "Fallback Hotspot"
    password: "xxx"
```

then this:

```
esphome:
  name: test2

esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_BOOTLOADER_WDT_ENABLE: n
      CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE: y
      CONFIG_ESP_INT_WDT: n
      CONFIG_ESP_TASK_WDT: n
      CONFIG_INT_WDT: n
      CONFIG_TASK_WDT: n

ota:
  password: ""

esp32_ble_tracker:

wifi:
  ssid: "xxx"
  password: "xxx"
  use_address: test.lan

  ap:
    ssid: "Fallback Hotspot"
    password: "xxx"

```

klim123123, 2022-01-21:
@rpatel3001 thank you!
Do i need to do anything special on device itself(m5stamp c3)? I get this "Configuration does not match the platform of the connected device. Expected an ESP32 device." I connect it to USB and it's recognized on COM4.

rpatel3001, 2022-01-21:
I have been unable to flash esphome over serial. My solution has been to flash tasmota (the -c3 binary) over serial, then use the firmware upgrade function to bootstrap an esphome build with OTA enabled. I have had to manually put the chip into flash mode by jumpering gpio9 to gnd while resetting, and then using no_reset for the before flash action for esptool. I can provide details in the morning, I should have been asleep hours ago lol.

rpatel3001, 2022-01-21:
1. Grab the c3 binary from here: http://ota.tasmota.com/tasmota32/release/
2. Grab the flash files from here: https://github.com/arendst/Tasmota-firmware/tree/main/static/esp32-c3
3. Reset the board while connecting GPIO9 to ground
4. Run the command with the proper serial port:
'''
esptool --port COM10 --chip esp32c3 --baud 460800 --before no_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x0000 bootloader_dout_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 tasmota32c3.bin
'''
5. Connect to the tasmota wifi network and set up your wifi credentials
6. Connect back to your wifi and visit the tasmota site at the IP displayed at the last step
7. Use the firmware upgrade page to upload an esphome binary

klim123123, 2022-01-23:
@rpatel3001 Thank you again! Very detailed instructions. It was easy to follow.

DanteBC, 2022-01-24:
> then this:
>
> ```
> esphome:
>   name: test2
>
> esp32:
>   board: esp32-c3-devkitm-1
>   variant: esp32c3
>   framework:
>     type: esp-idf
>     sdkconfig_options:
>       CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y [...]

Does it still give you warning in esp32_ble_tracker compilation with this Config or is it just my problem?

klim123123, 2022-01-24:
Also, this worked for me:

```
esphome:
  name: c3-test2
  platformio_options:
    board_build.mcu: esp32c3
    platform_packages:
      - platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2

esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: arduino
    version: 2.0.2
    platform_version: https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2idf/platform-espressif32-2.0.2.zip
    
# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "xxx"

wifi:
  ssid: "xxx"
  password: "xxx"
  fast_connect: true
#  power_save_mode: none

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "C3-Test2 Fallback Hotspot"
    password: "xxx"

esp32_ble_tracker:

web_server:
  port: 80
```

rpatel3001, 2022-01-24:
I do still get the warnings, but it hasn't impacted the functionality as far as I can tell. 

I was trying to use arduino for a bit, but there's more incompatibilities with the C3 chip in various components than with esp-idf, at least among the ones I was trying to use. Are you able to flash OTA through the web server without disabling the watchdogs?

rpatel3001, 2022-01-24:
I just reflashed and I guess somewhere along the way I something changed and those warnings are no longer produced. I have no idea why, but I was definitely getting them before.

LokiMetaSmith, 2022-02-06:
Slightly unrelated, but I was having issues with esp-idf not having support for Arduino http_request. Swapping in ESP32: Platform_version: https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2idf/platform-espressif32-2.0.2.zip
worked for me.

rpatel3001, 2022-02-08:
I narrowed it down to the task watchdog timing out during the OTA. OTAs succeed after increasing the timeout to 10s from the default of 5, using only the following options:
```
    sdkconfig_options:
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"
```

Not sure why a large binary would cause the OTA to timeout when receiving the binary size, it should be receiving 4 bytes no matter the actual value.

SebastianBar, 2022-03-13:
> I narrowed it down to the task watchdog timing out during the OTA. OTAs succeed after increasing the timeout to 10s from the default of 5, using only the following options:
>
> ``` [...]

This snippet allowed me to make the `esp32_improv` component work on a C3. Pre-flashing an image with `CONFIG_ESP_TASK_WDT_TIMEOUT_S` did the final trick. Thank you very much!!

michaelblight, 2022-05-20:
I was having the same problem with m5stamp c3u, and the above fixed it, but read on for further info. Initially I only had `CONFIG_BT_BLE_42_FEATURES_SUPPORTED` set, and not the other two (from a suggestion on another thread). At this point I could flash via USB, but not OTA, as per OP. Also at this point the c3u would lose wifi and reconnect every 10 seconds or so, and it would reflect as online/offline/online etc in ESPHome. I assume this was because dual mode wasn't working, and it was switching the antenna between the two (not that I really know how dual mode is supposed to work).

Once I added `CONFIG_BT_BLE_50_FEATURES_SUPPORTED` and `CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"` I could update OTA, and also wifi and BT worked at the same time.

In all cases I still get the warnings during compilation of esp32_ble_tracker.

Mosher23, 2022-06-04:
I have same issue!
If I specify `esp32_ble_tracker` board doesn't connect to wifi. 
Specifying these lines didn't help.
`    sdkconfig_options:
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"`

sanderlv, 2022-11-02:
I had issues but after falshing like here:
https://community.home-assistant.io/t/shelly-plus-1-esphome-bletracker/363549/43

I can still keep doing OTA now.

chose007, 2022-11-30:
I also had lots of trouble with initial wifi connection with my esp32-c3 board when using BLE tracker, especially when trying to tweak the ble scan parameters to receive more readings from my BT sensors and/or when using mqtt. Apparently, esphome starts scanning bluetooth devices immediately after booting which makes it very problematic to establish initial wifi connection. But I found out that when the wifi connection is successfully established, it then keeps on without disconnections even when using more agresive ble tracker settings (at least with my Ubiquiti Unifi APs).

Which led me to try to somehow postpone starting the ble tracker component. There is this setting: "continuous" which defaults to true, but when set to false, the tracker doesn't start automatically. After some more trial and error I ended up with this code which works fine for me - actually it's working more reliable than when using the default ble tracker scan settings and without mqtt which sort of worked too, but sometimes the wifi still didn't connect on boot and I had to reset to device. With this code, the board connects to wifi every time I start it up and keeps the connection on without problems. And I also get much more BT readings from my sensors with the tweaked scan settings, OTA works as well (thanks to @rpatel3001 findings above) and I'm able to use mqtt too. I also lowered the wifi reboot_timeout parameter to just 3 minutes - I hope that if the wifi connection drops anyway, the board should reboot after 3 minutes and connect to wifi again (before starting ble tracker) - although I haven't tested it yet.

```
esphome:
  name: esp32-c3-ble-tracker
  platformio_options:
    board_build.flash_mode: dio  
  on_boot:
    priority: 250
    then:
      - delay: 15s
      - lambda: |-
          id(ble_tracker).set_scan_continuous(true); 
          id(ble_tracker).start_scan();
      
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"

wifi:
  reboot_timeout: 3min
  ssid: "xxx"
  password: "xxx"

esp32_ble_tracker:
  id: ble_tracker
  scan_parameters:
    continuous: false
    interval: 211ms
    window: 120ms
```

adynis, 2023-01-29:
@chose007  : That thing with on_boot delay for ble is magic !!!  thanks !!
In my case ESP32-C3 was connecting usually to WiFi after 1..2..3 minutes. I was trying to understand what was causing the delay, I fund that was related to ble_tracker ... and then found your solution: Brilliant ! It worked immediatly, in 6..7 seconds it's correctly connected to wifi !! thanks.

NewfD90, 2023-02-09:
@chose007 Your brilliant delayed start of the ble scanner made my ESP32C3 WiFi connection rock solid reliable.  Thank you, thank you, thank you!

hansaya, 2023-02-13:
This is what worked for me. https://github.com/esphome/issues/issues/3668#issuecomment-1287568813

digaus, 2023-12-06:
Wouldnt it be better to use on_connect and on_disconnect from Wifi?

```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  on_connect:
    - esp32_ble_tracker.start_scan:
        continuous: true
  on_disconnect:
    - esp32_ble_tracker.stop_scan:
```

rvalitov, 2024-10-07:
After enabling verbose logging, I see that Bluetooth scan ends automatically after several minutes of working:
```
...
[02:37:47][V][bluetooth_proxy:061]: Proxying 2 packets
[02:37:47][V][esp32_ble:331]: (BLE) gap_event_handler - 3
[02:37:47][V][bluetooth_proxy:059]: Proxying raw packet from XX:XX:XX:XX:XX:XX, length 19. RSSI: -79 dB
[02:37:47][V][bluetooth_proxy:061]: Proxying 1 packets
[02:37:47][V][esp32_ble:331]: (BLE) gap_event_handler - 3
[02:37:47][V][esp32_ble:331]: (BLE) gap_event_handler - 3
[02:37:47][V][esp32_ble:331]: (BLE) gap_event_handler - 3
[02:37:47][V][bluetooth_proxy:059]: Proxying raw packet from XX:XX:XX:XX:XX:XX, length 18. RSSI: -82 dB
[02:37:47][V][bluetooth_proxy:059]: Proxying raw packet from XX:XX:XX:XX:XX:XX, length 31. RSSI: -39 dB
[02:37:47][V][bluetooth_proxy:061]: Proxying 2 packets
[02:37:47][D][esp32_ble_tracker:307]: End of scan.
```

Should we restart BT scan again after some time?
And I'm not sure it it's related, but HA does not get any information from this proxy.

rvalitov, 2024-10-10:
All solutions specified here are working great.
Originally the issue with Wi-Fi issues for single-core chips like ESP32-C3 is described [here](https://esphome.io/components/esp32_ble_tracker.html#use-on-single-core-chips).
I personally like this [solution](https://github.com/esphome/issues/issues/2941#issuecomment-1842369092)
Perhaps, this issue can be marked as closed?