esphome/issues #2942: spiTransferBytesNL: initialization from incompatible pointer type

danielgoepp, 2022-01-15:
### The problem

This is not causing any problems for me, just reporting in case folks have not seen this. I get a warnings about incompatible pointer types when installing. See warning message below for details.

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.12.9

### What platform are you using?

ESP32

### Board

esp32-poe-iso

### Component causing the issue

function spiTransferBytesNL

### Example YAML snippet

```yaml
esphome:
  name: esp-heat-control-1

esp32:
  board: esp32-poe-iso
  framework:
    type: arduino

logger:

ota:
  password: "49f164ecd194b13f337f018628a87e83"

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12
  domain: .home
  
mqtt:
  broker: 10.1.1.2
  username: ****
  password: ****

switch:
  - platform: gpio
    name: "Bathroom Second Floor Heat"
    pin: GPIO3
    id: bathroom_second_floor_heat
    inverted: true
  - platform: gpio
    name: "Greatroom Heat"
    pin: GPIO5
    id: greatroom_heat
    inverted: true
  - platform: gpio
    name: "Guestroom Heat"
    pin: GPIO13
    id: guestroom_heat
    inverted: true
  - platform: gpio
    name: "Studio A Heat"
    pin: GPIO14
    id: studio_a_heat
    inverted: true
```

### Anything in the logs that might be useful for us?

```txt
/config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
```

OttoWinter, 2022-01-18:
Yeah that's a warning in the arduino framework, which we don't control. 

You can check if it's already been reported/fixed here, and report it there if not: https://github.com/espressif/arduino-esp32/

closed 2022-01-18 by OttoWinter

mr-sneezy, 2022-02-02:
I have the same build warning, but my code compiles but doesn't work (ESP32 just boot loops).
It does verify OK in ESPhome GUI before compilation though. My booting issue may not be related to the error message of course...
I have the build log, ESP boot log, and code to show, can I do that here or should I open a new issue ?