esphome/issues #2943: ili9341 with TFT_2.4 display

LGhost1973, 2022-01-15:
### The problem

The display 2,4" is not working properly.
3/4 of the display is shown correctly, and 1/4 shows a repeat of the image on the display.
This is when rotation 0°.
At other rotation , it's even worse.

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.12.9

### What platform are you using?

ESP32

### Board

esp-wroom-32d

### Component causing the issue

ili9341

### Example YAML snippet

```yaml
esphome:
  name: esp32-rgb
  platform: ESP32
  board: nodemcu-32s

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "0fe36647b9eebb4abcb062efc8c72558"

wifi:
  ssid: !secret my_ssid
  password: !secret my_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "esp32-rgb"
    password: !secret my_password

captive_portal:

globals:
  - id: enable_change
    type: bool
    restore_value: no
    initial_value: 'true'

time:
  - platform: sntp
    id: sntp_time
    
http_request:    

sensor:
  - platform: homeassistant
    name: "Temperature sensor t3"
    entity_id: sensor.esp_t3  
    id: ha_esp_t3  
  - platform: homeassistant
    name: "Pressure sensor p3"
    entity_id: sensor.esp_p3  
    id: ha_esp_p3
  - platform: homeassistant
    name: "Humidity sensor h3"
    entity_id: sensor.esp_h3  
    id: ha_esp_h3
  - platform: homeassistant
    name: "CPU temperature"
    entity_id: sensor.cpu_temperature  
    id: ha_cpu_temperature
  - platform: homeassistant
    name: "Memory used"
    entity_id: sensor.memory_use_percent  
    id: ha_memory_use_percent 
  - platform: homeassistant
    name: "CPU used"
    entity_id: sensor.processor_use_percent  
    id: ha_processor_use_percent
  - platform: homeassistant
    name: "Weather code"
    entity_id: sensor.ha_weather_code  
    id: ha_weather_code
    
text_sensor:
  - platform: homeassistant
    name: "Weather condition"
    entity_id: sensor.ha_condition  
    id: ha_condition    
    
spi:
  clk_pin: GPIO25
  mosi_pin: GPIO26
  miso_pin: GPIO12 
  
font:
  - file: "fonts/arial.ttf"
    id: arial_font
    size: 16 
    glyphs: "1234567890.: "    
  - file: "fonts/arialbd.ttf"
    id: arialbd_font
    size: 17
    glyphs: "1234567890СИТЕМАУЛЦГОЯПЬмНаулицеВгспьтной.:-%METPHmg°CUusetpna "
    
image:
  - file: "w_sun.png"
    id: w_sun
    type: RGB24
    resize: 40x40
  - file: "w_moon.png"
    id: w_moon
    type: RGB24
    resize: 40x40    
  - file: "w_rain.png"
    id: w_rain
    type: RGB24
    resize: 40x40
  - file: "w_cloudy.png"
    id: w_cloudy
    type: RGB24
    resize: 40x40
  - file: "w_snow.png"
    id: w_snow
    type: RGB24
    resize: 40x40    
  - file: "w_other.png"
    id: w_other
    type: RGB24
    resize: 40x40 
    
display:
  - platform: ili9341
    model: TFT_2.4
    id: my_display
    reset_pin: GPIO33
    cs_pin: GPIO14
    dc_pin: GPIO27
    led_pin: GPIO32    
    auto_clear_enabled: false
    rotation: 0
    update_interval: 5s
    pages:
      - id: page1    
        lambda: |-
          auto green = Color(0, 255, 0);
          auto blue = Color(200, 200, 200);
          it.filled_rectangle(0, 0, 320, 240, Color::BLACK); 
          it.rectangle(0, 0, 320, 240);
          it.strftime(1, 138, id(arial_font), "%d.%m.%Y %H:%M", id(sntp_time).now());
          it.print(5, 0, id(arialbd_font),green , "СИСТЕМА");
          it.line(0, 20, 320, 20);          
          it.printf(5, 30, id(arialbd_font), "CPU: %.1f °C", id(ha_cpu_temperature).state);
          it.printf(5, 60, id(arialbd_font), "CPU: %.1f %%", id(ha_processor_use_percent).state);
          it.printf(5, 90, id(arialbd_font), "MEM: %.1f %%", id(ha_memory_use_percent).state);
            if (id(enable_change)) {
              it.filled_rectangle(14, 120, 10, 10, blue); 
              it.rectangle(44, 120, 10, 10, blue);
              it.rectangle(74, 120, 10, 10, blue);
              it.rectangle(104, 120, 10, 10, blue);
            }
      - id: page2    
        lambda: |-
          auto green = Color(0, 255, 0);
          auto blue = Color(200, 200, 200);
          it.filled_rectangle(0, 0, 320, 240, Color::BLACK);          
          it.rectangle(0, 0, 320, 240);
          it.strftime(1, 138, id(arial_font), "%d.%m.%Y %H:%M", id(sntp_time).now());
          it.print(5, 0, id(arialbd_font),green , "СПАЛЬНЯ");
          it.line(0, 20, 320, 20);          
          it.printf(5, 30, id(arialbd_font), "%.1f °C", id(ha_esp_t3).state);
          it.printf(5, 60, id(arialbd_font), "%.1f %%", id(ha_esp_h3).state);          
          it.printf(5, 90, id(arialbd_font), "%.1f mmHg", id(ha_esp_p3).state);
            if (id(enable_change)) {  
              it.rectangle(14, 120, 10, 10, blue);
              it.rectangle(44, 120, 10, 10, blue);
              it.rectangle(74, 120, 10, 10, blue); 
              it.filled_rectangle(104, 120, 10, 10, blue); 
            }
            if (id(ha_weather_code).state < 600) {
              it.image(80, 25, id(w_rain));
            } else {
              if (id(ha_weather_code).state < 700) {
                it.image(80, 25, id(w_snow));
              } else { 
                if (id(ha_weather_code).state < 800) {
                  it.image(80, 25, id(w_other));
                } else {              
                  if (id(ha_weather_code).state == 800) {
                    if (id(ha_condition).state == "clear-night") {
                      it.image(80, 25, id(w_moon));
                    } else {
                      it.image(80, 25, id(w_sun));
                    }
                  } else {
                    if (id(ha_weather_code).state > 800) {
                      it.image(80, 25, id(w_cloudy));
                    } 
                  }
                }  
              }  
            }            

interval:
  - interval: 7s
    then:
      lambda: |-
        if (id(enable_change)) {
          id(display_next).execute();
        }

script:
  - id: display_next
    then:
      - display.page.show_next: my_display
      - component.update: my_display
```

### Anything in the logs that might be useful for us?

```txt
no errors
```

firsttcrysis, 2022-01-28:
Hi,
I have exactly the same issue.
[image]

I discover that when You change type on M5STACK in epshome, screen is good filled only colors are inverted and whole screen is mirrored. So I suppose display is connected good, only we have problem with software.

[image]

LGhost1973, 2022-02-03:
Problem solving:
In the file ili9341_display.cpp need to modify the code:
//   24_TFT display
void ILI9341TFT24::initialize() {
  this->init_lcd_(INITCMD_TFT);
  this->width_ = 320; //240
  this->height_ = 240; //320
  this->fill_internal_(Color::BLACK);
}

martijnbuts, 2022-06-02:
> Problem solving: In the file ili9341_display.cpp need to modify the code: // 24_TFT display void ILI9341TFT24::initialize() { this->init_lcd_(INITCMD_TFT); this->width_ = 320; //240 this->height_ = 240; //320 this->fill_internal_(Color::BLACK); }

Can you descript how and where you change this? (should this not be fixed in the source files?) 
I have the same problem and if possible want to fix it.

closed 2022-06-07 by jesserockz