esphome/issues #2946: ESP32 C3 using WS2812 with esp-idf framework

TAMCTec, 2022-01-16:
### The problem

I'm trying to use neopixel on board, and it seem that ws2812 is not support for esp idf? 

As I'm using a raspberry pi 4 for home assistant, and Arduino 2.0.1 or 2.0.2 compiler is not for raspberry pi Armv7 framework. so I kinda stuck on esp idf framework.

Is there any other component supports idf frame work?

or is there a way to use Arduino framework with Raspberry Pi and ESP32 C3

### Which version of ESPHome has the issue?

2021.12.3 dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.8

### What platform are you using?

ESP32-IDF

### Board

esp32-c3-devkitM-1

### Component causing the issue

NeoPixelBus Light

### Example YAML snippet

```yaml
esphome:
  name: c3test
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: esp32-c3-devkitm-1
  variant: ESP32C3
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "C3Test Fallback Hotspot"
    password: "V6OmdKMap4I5"

# captive_portal:

light:
  - platform: fastled_clockless
    rgb_order: GRB
    chipset: WS2812
    pin: 8
    num_leds: 1
    name: "Sense Node Light"
```

martgras, 2022-01-16:
Unfortunately both fastled and neopixelbus need Arduino 
But you can use Arduino on a C3 

this works for me on a [M5 Stamp C3](https://docs.m5stack.com/en/core/stamp_c3)

```yaml 
esphome:
  name: testing
  platformio_options:
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
    platform:
      - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

logger:
  level: DEBUG

wifi:
  ssid: !secret wifi_sid
  password: !secret wifi_password

binary_sensor:
  - platform: gpio
    pin: GPIO3
    name: "M5 Button"

light:
  - platform: neopixelbus
    id: m5led
    type: GRB
    variant: SK6812
    method: ESP32_RMT_1
    num_leds: 1
    pin: GPIO2
    name: "M5 Led"

web_server:

```

OttoWinter, 2022-01-18:
Yeah, as noted the fastled/neopixel bus libraries do not support esp-idf (and we can't change that, support needs to be added upstream). Closing as this is more of a feature request in that case.

closed 2022-01-18 by OttoWinter