esphome/issues #2949: INA226 Power does not show negative values.

Oleg-Sob, 2022-01-16:
### The problem

Why INA226 with negative current readings, the power shows positive values. Should be negative too.

[image]

Please tell me how can I fix this.

### Which version of ESPHome has the issue?

Current version: 2021.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2021.12.9

### What platform are you using?

ESP8266

### Board

NodeMCU

martgras, 2022-01-18:
I looked at the raw data coming back from the sensor and it's returning a positive value for power even if current is negative. 
Is that correct ? I'm unsure if power should be signed or not - seems like TI thought it shouldn't. 
Your setup is probably similar to mine where the INA226 is attached to a battery and positive current  means that the battery   provides power to other devices and negative current that the battery is getting charged. In my case when the solar panel generates more energy than the consumers needs. 

If you prefer power to be negative you can use a lambda and multiply power with -1 if shunt voltage is negative

```yaml
- platform: ina226
    address: 0x40
    shunt_resistance: 0.00025 ohm
    current:
      name: "Battery Shunt Current"
    power:
      name: "Battery Total Power"
      filters:
        - lambda: "return id(shunt_voltage).state > 0 ?  x  : -x;"
    bus_voltage:
      name: "Bus Voltage"
    shunt_voltage:
      name: "Shunt Voltage"
      id: shunt_voltage
      unit_of_measurement: "mV"
      filters:
        - multiply: 1000
    max_current: 300A
```

Oleg-Sob, 2022-01-18:
Thanks, all worked well.

martgras, 2022-01-21:
Closing per last comment

closed 2022-01-21 by martgras