esphome/issues #2950: Heatpumpir library not found

ghjaseppu, 2022-01-16:
### The problem

Hi,

I am trying to remotely control a panasonic AC via the arduino heatpumpIR library.

I cannot install due an error at line 31: 
Platform not found: 'climate.heatpumpir'

Any clue how to fix that?

Thanks!

Joseph

### Which version of ESPHome has the issue?

2021.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2021.12.9

### What platform are you using?

ESP32

### Board

Esp 32 wrover

### Component causing the issue

heatpumpir

### Example YAML snippet

```yaml
esphome:
  name: clim
  platform: ESP32
  board: esp-wrover-kit

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "958523084dc193e278635e92123b2477"

wifi:
  ssid: xx
  password: xx

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Clim Fallback Hotspot"
    password: "N9Haz4kfx3O1"

captive_portal:

remote_transmitter:
  pin: GPIO32
  carrier_duty_percent: 50%
climate:
  - platform: heatpumpir
    name: "Living Room AC"
    protocol: panasonic_ckp
    horizontal_default: middle
    vertical_default: auto
    min_temperature: 10
    max_temperature: 40
```

jesserockz, 2022-01-17:
The reason it says `Platform not found: 'climate.heatpumpir'` is because it was not added until ESPHome 2021.10 and you are on ESPHome 2021.8

closed 2022-01-17 by jesserockz

ghjaseppu, 2022-01-17:
Thank you very much! I thought it was up to date as it doesn't ask for an update. I wiill try to force the update.