esphome/issues #2954: DSMR with Belgian digital meter : invalid string length error

spirann, 2022-01-18:
### The problem

Trying to read from the P1 port reports error :
```
[13:34:50][E][dsmr:265]: 0-0:96.1.4(202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020202020)
           ^
Invalid string length
```
I've already looked at this issue, but the solution doesn't seem to work. https://github.com/esphome/issues/issues/2340

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.12.9

### What platform are you using?

ESP8266

### Component causing the issue

dsmr

### Example YAML snippet

```yaml
---
substitutions:
  device_name: slimmelezer
  device_description: "DIY P1 module to read your smart meter"
     
esphome:
  name: ${device_name}
  comment: "${device_description}"
  platform: ESP8266
  esp8266_restore_from_flash: true
  board: d1_mini
  name_add_mac_suffix: false
  project:
    name: zuidwijk.slimmelezer
    version: "1.0"
  on_boot:
    then:
      - if:
          condition:
            lambda: return id(has_key);
          then:
            - lambda: |-
                std::string key(id(stored_decryption_key), 32);
                id(dsmr_instance).set_decryption_key(key);
          else:
            - logger.log:
                level: info
                format: "Not using decryption key. If you need to set a key use Home Assistant service 'ESPHome:  ${device_name}_set_dsmr_key'"
     
wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
 
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}
    ap_timeout: 15s
 
captive_portal:
 
# Enable logging
logger:
  baud_rate: 0
 
# Enable Home Assistant API
api:
  services:
    service: set_dsmr_key
    variables:
      private_key: string
    then:
      - logger.log:
          format: Setting private key %s. Set to empty string to disable
          args: [private_key.c_str()]
      - globals.set:
          id: has_key
          value: !lambda "return private_key.length() == 32;"
      - lambda: |-
          if (private_key.length() == 32)
            private_key.copy(id(stored_decryption_key), 32);
          id(dsmr_instance).set_decryption_key(private_key);
 
ota:
 
web_server:
  port: 80
 
uart:
  baud_rate: 115200
  rx_pin: D7
  rx_buffer_size: 3000
 
globals:
  - id: has_key
    type: bool
    restore_value: yes
    initial_value: "false"
  - id: stored_decryption_key
    type: char[32]
    restore_value: yes
 
dsmr:
  id: dsmr_instance
  # For Luxembourg users set here your decryption key
  #decryption_key: !secret decryption_key // enable this when using decryption for Luxembourg; key like '00112233445566778899AABBCCDDEEFF'
  max_telegram_length: 3000
  crc_check: true
  receive_timeout: 200ms
 
sensor:
  - platform: dsmr
    energy_delivered_lux:
      name: "Energy Consumed Luxembourg"
      state_class: total_increasing
    energy_delivered_tariff1:
      name: "Energy Consumed Tariff 1"
      state_class: total_increasing
    energy_delivered_tariff2:
      name: "Energy Consumed Tariff 2"
      state_class: total_increasing
    energy_returned_lux:
      name: "Energy Produced Luxembourg"
      state_class: total_increasing
    energy_returned_tariff1:
      name: "Energy Produced Tariff 1"
      state_class: total_increasing
    energy_returned_tariff2:
      name: "Energy Produced Tariff 2"
      state_class: total_increasing
    power_delivered:
      name: "Power Consumed"
      accuracy_decimals: 3
    power_returned:
      name: "Power Produced"
      accuracy_decimals: 3
    electricity_failures:
      name: "Electricity Failures"
      icon: mdi:alert
    electricity_long_failures:
      name: "Long Electricity Failures"
      icon: mdi:alert
    voltage_l1:
      name: "Voltage Phase 1"
    voltage_l2:
      name: "Voltage Phase 2"
    voltage_l3:
      name: "Voltage Phase 3"
    current_l1:
      name: "Current Phase 1"
    current_l2:
      name: "Current Phase 2"
    current_l3:
      name: "Current Phase 3"
    power_delivered_l1:
      name: "Power Consumed Phase 1"
      accuracy_decimals: 3
    power_delivered_l2:
      name: "Power Consumed Phase 2"
      accuracy_decimals: 3
    power_delivered_l3:
      name: "Power Consumed Phase 3"
      accuracy_decimals: 3
    power_returned_l1:
      name: "Power Produced Phase 1"
      accuracy_decimals: 3
    power_returned_l2:
      name: "Power Produced Phase 2"
      accuracy_decimals: 3
    power_returned_l3:
      name: "Power Produced Phase 3"
      accuracy_decimals: 3
    gas_delivered:
      name: "Gas Consumed"
      state_class: total_increasing
    gas_delivered_be:
      name: "Gas Consumed Belgium"
      state_class: total_increasing
  - platform: uptime
    name: "SlimmeLezer Uptime"
  - platform: wifi_signal
    name: "SlimmeLezer Wi-Fi Signal"
    update_interval: 60s
 
text_sensor:
  - platform: dsmr
    identification:
      name: "DSMR Identification"
    p1_version:
      name: "DSMR Version"
    p1_version_be:
      name: "DSMR Version Belgium"
  - platform: wifi_info
    ip_address:
      name: "SlimmeLezer IP Address"
    ssid:
      name: "SlimmeLezer Wi-Fi SSID"
    bssid:
      name: "SlimmeLezer Wi-Fi BSSID"
  - platform: version
    name: "ESPHome Version"
    hide_timestamp: true
```

thed6bel, 2022-02-15:
Hello,

I have exactly the same problem

did you solve your problem?

thanks ;)

Pvlerick, 2022-11-09:
Just received and plugged mine and I have the same issue @zuidwijk

Or an idea where to start to look to submit a fix?

zuidwijk, 2022-11-09:
Field 0-0:96.1.4 is not defined in the Dutch standard: https://www.netbeheernederland.nl/_upload/Files/Slimme_meter_15_a727fce1f1.pdf

yet Google-ing for it brought me to this website: https://jensd.be/1205/linux/data-lezen-van-de-belgische-digitale-meter-met-de-p1-poort

It seems that it is an ID,... not used in all meters apparently.

Pvlerick, 2022-11-09:
As the error states, that field is not defined in here: https://github.com/matthijskooijman/arduino-dsmr/blob/master/src/dsmr/fields.h , which to my understanding is the library used to parse the fields.

So it doesn't look that bad, that nonstandard needs to be added in the fields definitions, I guess...

Pvlerick, 2022-11-09:
Actually, someone already did that in a fork here: https://github.com/rvdbreemen/arduino-dsmr-nl-be

So I suppose now the path forward is to recompile using that library instead of the original one.

Pvlerick, 2022-12-18:
I finally got time to look at this, but it's not yet working. @zuidwijk can you point me to the default configuration that you are using?

I found yours here (https://www.zuidwijk.com/slimmelezer-configuration/) but I'd like to try with the "default" one first (the one that comes pre-installed).

Thanks

Pvlerick, 2022-12-20:
After quite a fit of back and forth, I finally managed to get this to work (took a while because I've no prior experience with any of this stuff).

Turns out the PR https://github.com/glmnet/arduino-dsmr/pull/11 does the trick, but since it hasn't been merged it's not in the lib that is on platform.io

In order to have this working, you need to rebuild esphome using that lib instead of the default one. You can just clone my fork of esphome or pull my `dsmr` branch: https://github.com/Pvlerick/esphome/tree/dsmr

Now you need to build the Docker image to use that version of esphome, here's how (from the root folder of esphome): 

```
docker build -t esphome:dsmr --file docker/Dockerfile --target docker .
```

And now onto using that new version of esphome to flash your slimmelezer, either by USB or OTA:

USB:
```
sudo docker run --rm -v ~/slimmelezer:/config --device=/dev/ttyUSB0 -it esphome:dsmr run /config/config.yaml
```
The config file is located on the host at ~/slimmelezer/config.yaml then mounted in /config/config.yaml 

OTA:
```
docker run --rm -v ~/slimmelezer:/config -it --add-host slimmelezer.local:192.168.1.26 esphome:dsmr run /config/config.yaml
```
Same as USB way for the config, but notice the `--add-host` where I'm specifying the IP address of the slimmelezer on the local network, otherwise Docker will fail resolving it.

You should witness that the deps graph uses the referenced version of the Dsml library:
```
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- ArduinoJson @ 6.18.5
|-- Dsmr @ 0.6.0+sha.9bf164e
|-- Crypto @ 0.4.0
```

Small note on the configuration file: I'm using the one provided by @zuidwijk  at https://www.zuidwijk.com/slimmelezer-configuration/ , but since my branch of esphome is based on the dev branch, it uses a new version of the web server by default, so make sure to set its version to 1 to get the correct layout if you browse to slimmelezer.local : 

```
web_server:
  port: 80
  version: 1
```

In conclusion, https://github.com/glmnet/arduino-dsmr/pull/11 should be merged so we can all benefit ;-) cc @glmnet 

Hope this helps!

pieter0021, 2023-06-08:
Is there any progress on this? I would indeed be amazing to have this modification through a simple update :)
My knowledge is of HA is bit more limited, you see ;-).

Pvlerick, 2023-06-19:
https://github.com/glmnet/arduino-dsmr/pull/11 has been merged, unfortunately no new release of DSMR has been release on platform.io: https://registry.platformio.org/libraries/glmnet/Dsmr

@glmnet do you think you could push 0.7 to platform.io so esphome can be updated with the fix?

glmnet, 2023-06-20:
Just released

glmnet, 2023-06-20:
Ohh I think we still need to upgrade esphome dep

Pvlerick, 2023-06-20:
BTW I got an answer from Sibelga and they sent me the vanilla specs, so what they install doesn't follow the specs and their first like tech support is clueless about it.

closed 2023-06-26 by glmnet