esphome/issues #2955: Can't compile ESPHOME on ESP32-SOLO-1, solution from #2165 doesn't work

cryptotomte, 2022-01-18:
### The problem

I had the same problem and tried the solution provided by jesserockz in the closed issue "#2165 What is the proper way of using ESP32-SOLO-1?".
However it doesn't work and I get the following error message:

```
INFO Reading configuration /config/esphome/esp32test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32test (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 3.3.2)

HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf 3.40300.0 (4.3.0) 
 - tool-cmake 3.16.4 
 - tool-ninja 1.7.1 
Reading CMake configuration...
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/data/esp32test/.pioenvs/esp32test/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:189 (message):
  Failed to resolve component 'cxx'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:215 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:435 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 1.36 seconds ==========================
```
My yaml is as follows:

```
esphome:
  name: esp32test

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      
# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "deleted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32Test Fallback Hotspot"
    password: "deleted"

```

What can be the problem here?

// P

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12

### What platform are you using?

ESP32

cryptotomte, 2022-01-19:
Update: HA had an update of ESPHOME to 2022.01. The problem still there, the Cmake-file complains about can't resolve component 'cxx'. See above for more details.

Br,

// P

jesserockz, 2022-01-19:
I will try to reproduce this and report back.

nhscott, 2022-01-24:
I have the same exact issue tried to comple but I can't. No errors, no prompt, when I try to install nothing happens but when I remove framework then can install it on the device but then it never connects to network.
Id like to switch to esphome from tasmota to use the am43 covers and Ble components I have.

wrobell, 2022-06-28:
Using ESPHome 2022.6.2 and the following works for my ESP 32 solo boards
```
esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
```

mmakaay, 2022-06-30:
Yes, that is how it should work.
I just tried compiling the exact configuration that @cryptotomte used, and it worked out of the box.
Closing this issue.

closed 2022-06-30 by mmakaay