esphome/issues #2956: ESPHome firmware compilation Crashes HA Core

maretodoric, 2022-01-19:
### The problem

Every time after ESPHome is updated and i initiate a firmware upgrade on end device, Home Assistant core instance crashes without notice. This only seems to happen if full compile is needed, not when firmware itself is at the latest version.
Is this due to RAM constraints of Raspberry Pi 3B+ or is something else in play?

### Which version of ESPHome has the issue?

2022.1.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.12.7

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

core

### Anything in the logs that might be useful for us?

```txt
22-01-19 09:04:30 INFO (SyncWorker_3) [supervisor.docker.interface] Updating image ghcr.io/esphome/esphome-hassio-armv7:2021.12.3 to ghcr.io/esphome/esphome-hassio-armv7:2022.1.0
22-01-19 09:04:30 INFO (SyncWorker_3) [supervisor.docker.interface] Downloading docker image ghcr.io/esphome/esphome-hassio-armv7 with tag 2022.1.0.
22-01-19 09:05:41 INFO (SyncWorker_3) [supervisor.docker.interface] Stopping addon_15ef4d2f_esphome application
22-01-19 09:05:43 INFO (SyncWorker_3) [supervisor.docker.interface] Cleaning addon_15ef4d2f_esphome application
22-01-19 09:05:43 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.9/site-packages/sentry_sdk/integrations/aiohttp.py", line 121, in sentry_app_handle
    reraise(*_capture_exception(hub))
  File "/usr/local/lib/python3.9/site-packages/sentry_sdk/_compat.py", line 54, in reraise
    raise value
  File "/usr/local/lib/python3.9/site-packages/sentry_sdk/integrations/aiohttp.py", line 111, in sentry_app_handle
    response = await old_handle(self, request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 139, in system_validation
    return await handler(request)
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 205, in token_validation
    return await handler(request)
  File "/usr/src/supervisor/supervisor/api/utils.py", line 65, in wrap_api
    answer = await method(api, *args, **kwargs)
  File "/usr/src/supervisor/supervisor/api/addons.py", line 408, in stats
    stats: DockerStats = await addon.stats()
  File "/usr/src/supervisor/supervisor/addons/addon.py", line 674, in stats
    return await self.instance.stats()
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/supervisor/supervisor/docker/interface.py", line 536, in _stats
    stats = docker_container.stats(stream=False)
  File "/usr/local/lib/python3.9/site-packages/docker/models/containers.py", line 422, in stats
    return self.client.api.stats(self.id, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/docker/utils/decorators.py", line 19, in wrapped
    return f(self, resource_id, *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/docker/api/container.py", line 1139, in stats
    return self._result(self._get(url, params={'stream': False}),
  File "/usr/local/lib/python3.9/site-packages/docker/api/client.py", line 277, in _result
    return response.json()
  File "/usr/local/lib/python3.9/site-packages/requests/models.py", line 910, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/local/lib/python3.9/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.9/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/lib/python3.9/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
22-01-19 09:05:43 INFO (MainThread) [supervisor.addons] Add-on '15ef4d2f_esphome' successfully updated
22-01-19 09:05:45 INFO (SyncWorker_2) [supervisor.docker.interface] Cleanup images: ['ghcr.io/esphome/esphome-hassio-armv7:2021.12.3']
22-01-19 09:05:54 INFO (SyncWorker_0) [supervisor.docker.addon] Starting Docker add-on ghcr.io/esphome/esphome-hassio-armv7 with version 2022.1.0
22-01-19 09:06:52 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_ssh
22-01-19 09:06:52 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_ssh
22-01-19 09:06:52 INFO (MainThread) [supervisor.api.middleware.security] /host/info access from a0d7b954_ssh
22-01-19 09:06:52 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from a0d7b954_ssh
22-01-19 09:06:59 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_ssh
22-01-19 09:06:59 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a0d7b954_ssh
22-01-19 09:06:59 INFO (MainThread) [supervisor.api.middleware.security] /host/info access from a0d7b954_ssh
22-01-19 09:07:00 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from a0d7b954_ssh
22-01-19 09:09:08 WARNING (MainThread) [supervisor.homeassistant.websocket] Connection is closed
22-01-19 09:09:17 WARNING (MainThread) [supervisor.misc.tasks] Watchdog miss API response from Home Assistant
22-01-19 09:09:32 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with Home Assistant Docker!
22-01-19 09:09:33 INFO (SyncWorker_7) [supervisor.docker.interface] Starting homeassistant
22-01-19 09:09:34 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
22-01-19 09:10:12 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-01-19 09:10:12 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
```

pittbull, 2022-01-25:
I'm having this issue as well. Different nodes (nodemcu and wemos) full crash when updating them. (Full crash as in having to restart vm)

This aldo happens 3 out 5 times when recompiling the code and sending it to the node, but these times HA usually recovers. 

Please advise.

michaelwoods, 2022-02-03:
I'm using the dashboard as a docker container and my host system is crashing when I install an update to a device. I hear the fans kick in, I lose my ssh session and then it beeps and reboots. I'm not able to find anything in the logs yet, I watched `htop` just now and `runc` was over 100% at the time of crash (i7 CPU, 32GB RAM).

The web UI state at the time of crash was listing the .cpp.o component compilations.

maretodoric, 2022-02-12:
In my case it was crashing due to low memory, probably because Raspberry Pi 3 B+ has only 1GB of RAM.
I added SWAP of 2GB and just now managed to compile the device after ESPHome update with success.

KrisRux, 2022-03-16:
Hi all, i have the same problem. @maretodoric how to add a SWAP?

maretodoric, 2022-03-16:
@KrisRux check this guide:
https://community.home-assistant.io/t/how-to-increase-the-swap-file-size-on-home-assistant-os/272226

jimbovp, 2022-07-18:
> @KrisRux check this guide: https://community.home-assistant.io/t/how-to-increase-the-swap-file-size-on-home-assistant-os/272226

Thanks man, you're a live saver... well at least a Raspberry pi saver, since I almost threw it out of the window...

ssieb, 2022-10-27:
A compile option has just been added to dev, so you can limit the number of processes.
https://github.com/esphome/esphome/pull/3952

closed 2022-10-27 by ssieb