esphome/issues #2957: Write Modbus Controller Number writes in false registers

killee, 2022-01-19:
### The problem

If I try to write registers the register address is false. It Seams that the address is with a factor 2.

I try to set the Register 0x18 (24) to 1. But the component write it to the register 0x30 (48). This happens for all other registers to.
For Switches it works correct.

  - platform: modbus_controller
    modbus_controller_id: heatpump
    name: "${pumpname} Turn on/off heat pump"
    address: 0x18
    value_type: S_WORD
    entity_category: config

log:
[10:12:06][D][modbus.number:055]: Updating register: connected Sensor=WP1 Turn on/off heat pump start address=0x0 register count=1 new value=0.00 (val=0.00)
[10:12:06][V][modbus_controller:036]: Sending next modbus command to device 1 register 0x30 count 1
[10:12:06][V][modbus:184]: Modbus write: 01.06.00.30.00.00.89.C5 (8)
[10:12:06][V][modbus_controller:461]: Command sent 6 0x30 1
[10:12:06][V][modbus:058]: Modbus received Byte  1 (0X1)
[10:12:06][V][modbus:058]: Modbus received Byte  6 (0X6)
[10:12:06][V][modbus:058]: Modbus received Byte  0 (0X0)
[10:12:06][V][modbus:058]: Modbus received Byte  48 (0X30)
[10:12:06][V][modbus:058]: Modbus received Byte  0 (0X0)
[10:12:06][V][modbus:058]: Modbus received Byte  0 (0X0)
[10:12:06][V][modbus:058]: Modbus received Byte  137 (0X89)

### Which version of ESPHome has the issue?

ESPHome 2022.1.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.10

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

Modbus Controller Number

### Example YAML snippet

```yaml
substitutions:
  devicename: heatpump-kille-com
  internalname: heatpump-kille-com
  pumpname: WP1

esphome:
  name: heatpump-kille-com
  platform: ESP32
  board: esp32dev

packages:
  sensor_gen: !include common/sensor_generic.yaml
  network: !include common/network_generic.yaml

# Enable Home Assistant API
api:
[... 508 lines cut ...]

[10:11:38][D][modbus.number:055]: Updating register: connected Sensor=WP1 Turn on/off heat pump start address=0x0 register count=1 new value=1.00 (val=1.00)
[10:11:38][V][modbus_controller:036]: Sending next modbus command to device 1 register 0x30 count 1
[10:11:38][V][modbus:184]: Modbus write: 01.06.00.30.00.01.48.05 (8)
[10:11:38][V][modbus_controller:461]: Command sent 6 0x30 1
[10:11:38][V][modbus:058]: Modbus received Byte  1 (0X1)
[10:11:38][V][modbus:058]: Modbus received Byte  6 (0X6)
[10:11:38][V][modbus:058]: Modbus received Byte  0 (0X0)
[10:11:38][V][modbus:058]: Modbus received Byte  48 (0X30)
[10:11:38][V][modbus:058]: Modbus received Byte  0 (0X0)
[10:11:38][V][modbus:058]: Modbus received Byte  1 (0X1)
[10:11:38][V][modbus:058]: Modbus received Byte  72 (0X48)

[10:12:06][D][modbus.number:055]: Updating register: connected Sensor=WP1 Turn on/off heat pump start address=0x0 register count=1 new value=0.00 (val=0.00)
[10:12:06][V][modbus_controller:036]: Sending next modbus command to device 1 register 0x30 count 1
[10:12:06][V][modbus:184]: Modbus write: 01.06.00.30.00.00.89.C5 (8)
[10:12:06][V][modbus_controller:461]: Command sent 6 0x30 1
[10:12:06][V][modbus:058]: Modbus received Byte  1 (0X1)
[10:12:06][V][modbus:058]: Modbus received Byte  6 (0X6)
[10:12:06][V][modbus:058]: Modbus received Byte  0 (0X0)
[10:12:06][V][modbus:058]: Modbus received Byte  48 (0X30)
[10:12:06][V][modbus:058]: Modbus received Byte  0 (0X0)
[10:12:06][V][modbus:058]: Modbus received Byte  0 (0X0)
[10:12:06][V][modbus:058]: Modbus received Byte  137 (0X89)
```

martgras, 2022-01-19:
Can you please check the register addresses 
You are using 24 == 0x18 twice for different numbers 

```yaml
  - platform: modbus_controller
    modbus_controller_id: heatpump
    name: "${pumpname} Target overheat degree(heating)"
    address: 24
    value_type: S_WORD
    entity_category: config

  - platform: modbus_controller
    modbus_controller_id: heatpump
    name: "${pumpname} Turn on/off heat pump"
    address: 0x18
    value_type: S_WORD
    entity_category: config
```

I'm checking why the request is sent to address 30 but it would help to get the duplicate address out of the way first 
@stegm -  anything that rings a bell  - before I start taking a closer look at the "range - making" code

stegm, 2022-01-19:
I have an assumption: During register range merging the start address of the sensors are changed to the one of the first sensor in range. I had to work around this in my select PR #3032. Maybe the switch implementation is not aware of this?

BTW  I  assume due to my last patch of the range algorithm, changing the start address should be no longer necessary at all. Perhaps I should take a look at this.

@killee Can you try it with only the switch sensor, nothing else? Can you also post the logs with "[C]" from the beginning?

killee, 2022-01-19:
> I have an assumption: During register range merging the start address of the sensors are changed to the one of the first sensor in range. I had to work around this in my select PR #3032. Maybe the [...]

Only the switch in the firmware:
[13:44:57][I][app:102]: ESPHome version 2022.1.0 compiled on Jan 19 2022, 13:44:23
[13:44:57][C][wifi:488]: WiFi:
[13:44:57][C][wifi:350]:   Local MAC: 7C:9E:BD:46:51:04
[13:44:57][C][wifi:351]:   SSID: [redacted]
[13:44:57][C][wifi:352]:   IP Address: 192.168.83.76
[13:44:57][C][wifi:354]:   BSSID: [redacted]
[13:44:57][C][wifi:355]:   Hostname: 'heatpump-kille-com'
[13:44:57][C][wifi:357]:   Signal strength: -54 dB ▂▄▆█
[13:44:57][V][wifi:359]:   Priority: 0.0
[13:44:58][C][logger:233]: Logger:
[13:44:58][C][logger:234]:   Level: VERBOSE
[13:44:58][C][logger:235]:   Log Baud Rate: 0
[13:44:58][C][logger:236]:   Hardware UART: UART0
[13:44:58][C][uart.arduino_esp32:105]: UART Bus:
[13:44:58][C][uart.arduino_esp32:106]:   TX Pin: GPIO17
[13:44:58][C][uart.arduino_esp32:107]:   RX Pin: GPIO16
[13:44:58][C][uart.arduino_esp32:109]:   RX Buffer Size: 256
[13:44:58][C][uart.arduino_esp32:111]:   Baud Rate: 9600 baud
[13:44:58][C][uart.arduino_esp32:112]:   Data Bits: 8
[13:44:58][C][uart.arduino_esp32:113]:   Parity: NONE
[13:44:58][C][uart.arduino_esp32:114]:   Stop bits: 1
[13:44:58][C][modbus:129]: Modbus:
[13:44:58][C][modbus:130]:   Flow Control Pin: GPIO5
[13:44:58][C][modbus:131]:   Send Wait Time: 250 ms
[13:44:58][C][uptime.sensor:031]: Uptime Sensor 'heatpump-kille-com Uptime Sensor'
[13:44:58][C][uptime.sensor:031]:   State Class: 'total_increasing'
[13:44:58][C][uptime.sensor:031]:   Unit of Measurement: 's'
[13:44:58][C][uptime.sensor:031]:   Accuracy Decimals: 0
[13:44:58][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[13:44:58][V][uptime.sensor:031]:   Unique ID: '7c9ebd465104-uptime'
[13:44:58][C][template.text_sensor:021]: Template Sensor 'heatpump-kille-com Uptime Human Readable'
[13:44:58][C][template.text_sensor:021]:   Icon: 'mdi:clock-start'
[13:44:58][C][status:034]: Status Binary Sensor 'heatpump-kille-com online'
[13:44:58][C][status:034]:   Device Class: 'connectivity'
[13:44:58][C][modbus_controller.switch:014]: modbus_controller.switchModbus Controller Switch 'WP1 Turn on/off heat pump'
[13:44:58][C][captive_portal:144]: Captive Portal:
[13:44:58][C][web_server:162]: Web Server:
[13:44:58][C][web_server:163]:   Address: heatpump-kille-com.local:80
[13:44:58][C][ota:082]: Over-The-Air Updates:
[13:44:58][C][ota:083]:   Address: heatpump-kille-com.local:3232
[13:44:58][C][ota:086]:   Using Password.
[13:44:58][C][api:134]: API Server:
[13:44:58][C][api:135]:   Address: heatpump-kille-com.local:6053
[13:44:58][C][api:139]:   Using noise encryption: NO
[13:44:58][C][wifi_signal.sensor:009]: WiFi Signal 'heatpump-kille-com WiFi Signal Sensor'
[13:44:58][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[13:44:58][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[13:44:58][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[13:44:58][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[13:44:58][V][wifi_signal.sensor:009]:   Unique ID: '7c9ebd465104-wifisignal'
[13:44:58][C][mdns:084]: mDNS:
[13:44:58][C][mdns:085]:   Hostname: heatpump-kille-com
[13:44:58][V][mdns:086]:   Services:
[13:44:58][V][mdns:088]:   - _esphomelib, _tcp, 6053
[13:44:58][V][mdns:090]:     TXT: version = 2022.1.0
[13:44:58][V][mdns:090]:     TXT: mac = 7c9ebd465104
[13:44:58][V][mdns:090]:     TXT: platform = ESP32
[13:44:58][V][mdns:090]:     TXT: board = esp32dev
[13:44:58][C][modbus_controller:275]: ModbusController:
[13:44:58][C][modbus_controller:276]:   Address: 0x01
[13:44:58][C][modbus_controller:278]: sensormap
[13:44:58][C][modbus_controller:282]:  Sensor type=3 start=0x22 offset=0x0 count=1 size=2
[13:44:58][C][modbus_controller:284]: ranges
[13:44:58][C][modbus_controller:287]:   Range type=3 start=0x22 count=1 skip_updates=0
[13:45:01][D][api:098]: Accepted 192.168.83.74
[13:45:01][V][api.connection:763]: Hello from client: 'Home Assistant 2021.12.10 (192.168.83.74)'
[13:45:01][D][api.connection:779]: Home Assistant 2021.12.10 (192.168.83.74): Connected successfully
[... 5 lines cut ...]
[13:45:08][V][modbus_controller:461]: Command sent 6 0x22 1
[13:45:09][V][modbus:058]: Modbus received Byte  1 (0X1)
[13:45:09][V][modbus:058]: Modbus received Byte  6 (0X6)
[13:45:09][V][modbus:058]: Modbus received Byte  0 (0X0)
[13:45:09][V][modbus:058]: Modbus received Byte  34 (0X22)
[13:45:09][V][modbus:058]: Modbus received Byte  0 (0X0)
[13:45:09][V][modbus:058]: Modbus received Byte  1 (0X1)
[13:45:09][V][modbus:058]: Modbus received Byte  232 (0Xe8)
[13:45:09][V][modbus:058]: Modbus received Byte  0 (0X0)
[13:45:09][V][modbus_controller:055]: Modbus response queued
[13:45:12][D][switch:017]: 'WP1 Turn on/off heat pump' Turning OFF.
[13:45:12][V][modbus_controller.switch:072]: write_state 'WP1 Turn on/off heat pump': new value = OFF type = 3 address = 22 offset = 0
[13:45:12][D][switch:037]: 'WP1 Turn on/off heat pump': Sending state OFF
[13:45:12][V][modbus_controller:036]: Sending next modbus command to device 1 register 0x22 count 1
[13:45:12][V][modbus:184]: Modbus write: 01.06.00.22.00.00.29.C0 (8)
[13:45:12][V][modbus_controller:461]: Command sent 6 0x22 1
[13:45:12][V][modbus:058]: Modbus received Byte  1 (0X1)
[13:45:12][V][modbus:058]: Modbus received Byte  6 (0X6)
[13:45:12][V][modbus:058]: Modbus received Byte  0 (0X0)
[13:45:12][V][modbus:058]: Modbus received Byte  34 (0X22)
[13:45:12][V][modbus:058]: Modbus received Byte  0 (0X0)
[13:45:12][V][modbus:058]: Modbus received Byte  0 (0X0)
[13:45:12][V][modbus:058]: Modbus received Byte  41 (0X29)
[13:45:12][V][modbus:058]: Modbus received Byte  192 (0Xc0)
[13:45:12][V][modbus_controller:055]: Modbus response queued

martgras, 2022-01-19:
Why switch ? 
Isn't number causing the problem while switch is ok?

killee, 2022-01-19:
> Can you please check the register addresses You are using 24 == 0x18 twice for different numbers
>
> ```yaml
>   - platform: modbus_controller
>     modbus_controller_id: heatpump
>     name: "${pumpname} [...]

That was a testing mistake...
but also without dupple register I get the same result:
Some writing of different numbers:
[13:52:09][D][number:010]: 'WP1 Heating target temp' - Setting
[13:52:09][D][number:029]:   Value: 44.000000
[13:52:09][D][modbus.number:055]: Updating register: connected Sensor=WP1 Heating target temp start address=0x0 register count=1 new value=44.00 (val=44.00)
[13:52:09][V][modbus_controller:036]: Sending next modbus command to device 1 register 0x20 count 1
[13:52:09][V][modbus:184]: Modbus write: 01.06.00.20.00.2C.89.DD (8)
[13:52:09][V][modbus_controller:461]: Command sent 6 0x20 1
[13:52:09][V][modbus:058]: Modbus received Byte  1 (0X1)
[13:52:09][V][modbus:058]: Modbus received Byte  6 (0X6)
[13:52:09][V][modbus:058]: Modbus received Byte  0 (0X0)
[13:52:09][V][modbus:058]: Modbus received Byte  32 (0X20)
[13:52:09][V][modbus:058]: Modbus received Byte  0 (0X0)
[13:52:09][V][modbus:058]: Modbus received Byte  44 (0X2c)
[13:52:09][V][modbus_controller:036]: Sending next modbus command to device 1 register 0x26 count 1
[13:52:09][V][modbus:184]: Modbus write: 01.06.00.26.00.31.A9.D5 (8)
[13:52:09][V][modbus_controller:461]: Command sent 6 0x26 1
[13:52:09][V][modbus:058]: Modbus received Byte  1 (0X1)
[13:52:09][V][modbus:058]: Modbus received Byte  6 (0X6)
[13:52:09][V][modbus:058]: Modbus received Byte  0 (0X0)
[13:52:09][V][modbus:058]: Modbus received Byte  38 (0X26)
[13:52:09][V][modbus:058]: Modbus received Byte  0 (0X0)
[13:52:09][V][modbus:058]: Modbus received Byte  49 (0X31)
[13:52:09][V][modbus:058]: Modbus received Byte  169 (0Xa9)
[13:52:09][V][modbus:058]: Modbus received Byte  213 (0Xd5)
[13:52:09][V][modbus_controller:055]: Modbus response queued
[13:52:09][V][modbus_controller:063]: Process modbus response for address 0x26 size: 4
[13:52:09][V][modbus_controller:308]: Command ACK 0x26 9728 
[13:52:09][D][number:036]: 'WP1 Fix frequency at testing': Sending state 49.000000
[13:52:11][D][number:010]: 'WP1 EEV OPEN degree（coolinging)' - Setting
[13:52:11][D][number:029]:   Value: 349.000000
[13:52:11][D][modbus.number:055]: Updating register: connected Sensor=WP1 EEV OPEN degree（coolinging) start address=0x0 register count=1 new value=349.00 (val=349.00)
[13:52:11][V][modbus_controller:036]: Sending next modbus command to device 1 register 0x2C count 1
[13:52:11][V][modbus:184]: Modbus write: 01.06.00.2C.01.5D.88.6A (8)
[13:52:11][V][modbus_controller:461]: Command sent 6 0x2C 1
[13:52:11][V][modbus:058]: Modbus received Byte  1 (0X1)
[13:52:11][V][modbus:058]: Modbus received Byte  6 (0X6)
[13:52:11][V][modbus:058]: Modbus received Byte  0 (0X0)
[13:52:11][V][modbus:058]: Modbus received Byte  44 (0X2c)
[13:52:11][V][modbus:058]: Modbus received Byte  1 (0X1)
[13:52:11][V][modbus:058]: Modbus received Byte  93 (0X5d)
[13:52:11][V][modbus:058]: Modbus received Byte  136 (0X88)
[13:52:11][V][modbus:058]: Modbus received Byte  106 (0X6a)
[13:52:11][V][modbus_controller:055]: Modbus response queued
[13:52:20][V][text_sensor:016]: 'heatpump-kille-com Uptime Human Readable': Received new state 5m 32s
[13:52:20][D][text_sensor:067]: 'heatpump-kille-com Uptime Human Readable': Sending state '5m 32s'
[13:52:20][V][sensor:074]: 'heatpump-kille-com Uptime Sensor': Received new state 331.863007
[13:52:20][D][sensor:125]: 'heatpump-kille-com Uptime Sensor': Sending state 331.86301 s with 0 decimals of accuracy

killee, 2022-01-19:
> Why switch ? Isn't number causing the problem while switch is ok?

Exactly only numbers has incorect registers for the writing process.

martgras, 2022-01-19:
I think I found it can you please try
```
external_components:
  - source:
      type: git
      url: https://github.com/martgras/esphome
      ref: modbus_number_write
    components: [modbus_controller]
```

stegm, 2022-01-19:
@martgras Not sure if your patch solves everything. Assume someone change the offset in yaml to an odd address.

```yaml
  - platform: modbus_controller
    modbus_controller_id: heatpump
    id: switch_heat_pump
    name: "${pumpname} Turn on/off heat pump"
    register_type: holding
    address: 34
    offset: 1
    bitmask: 1
```

martgras, 2022-01-19:
Then offset in c++ will be 2 => should be ok. The byte_offset attribute maps to the C++ members offset
Different topic but I want to depreciate offset in YAML anyways - I don't see a good reason for it anymore and it's confusing

killee, 2022-01-19:
> I think I found it can you please try
> 
> ```
> external_components:
>   - source:
>       type: git
>       url: https://github.com/martgras/esphome
>       ref: modbus_number_write
>     components: [modbus_controller]
> ```

You're made it!
Now it works correct! 
Thank you very much for the fast reply and bug solving!

closed 2022-01-19 by killee

stegm, 2022-01-19:
> Then offset in c++ will be 2 => should be ok. The byte_offset attribute maps to the C++ members offset Different topic but I want to depreciate offset in YAML anyways - I don't see a good reason for it anymore and it's confusing

+1 for removing offset in YAML - see no usage for this. Only problems ;-)