esphome/issues #2958: Error copying file post_build.py.script: no such file or directory

Nardol, 2022-01-19:
### The problem

When running esphome compile, it fails imediatly because build script cannot be copied.
It occurs with ESP32 and ESP8266

### Which version of ESPHome has the issue?

2022.1

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Component causing the issue

core

### Anything in the logs that might be useful for us?

```txt
ERROR Error copying file /home/uset/.local/lib/python3.7/site-packages/esphome/components/esp8266/post_build.py.script to .esphome/build/diffuseur-bureau/post_build.py: [Errno 2] No such file or directory: '/home/user/.local/lib/python3.7/site-packages/esphome/components/esp8266/post_build.py.script'
```
```

### Additional information

Looks like related to esphome/esphome#3057

wrmulf, 2022-01-19:
I had the same issue. I copied the file

https://raw.githubusercontent.com/esphome/esphome/dev/esphome/components/esp8266/post_build.py.script

to the appropriate location, in your case: /home/uset/.local/lib/python3.7/site-packages/esphome/components/esp8266/

Maybe it is necessary, to make the file executable.

mweinelt, 2022-01-19:
I think they're not being picked up, because their extension is now `.script` and setuptools doesn't like that, without it being explicitly configured to pick it up.

AWSb24, 2022-01-19:
Side note: I have the same issue with python 3.9

ottuzzi, 2022-01-19:
> I had the same issue. I copied the file
>
> https://raw.githubusercontent.com/esphome/esphome/dev/esphome/components/esp8266/post_build.py.script
>
> to the appropriate location, in your case: [...]

This solution works for me on Windows11 and python 3.10.2.

mhostyn, 2022-01-19:
> This solution works for me on Windows11 and python 3.10.2.

I had the same issue on Windows 10 using the latest Python release 3.10.2 and was able to resolve by creating a new empty file '%Python310 directory%\Lib\site-packages\esphome\components\esp32\post_build.py.script'

Note: i was compiling for an ESP32

closed 2022-01-19 by jesserockz

jesserockz, 2022-01-19:
I have release a fix in 2022.1.1. Please let me know if that is not working for you

mweinelt, 2022-01-19:
It works, thanks.

Nardol, 2022-01-20:
I also confirm the fix, tested with both ESP32 and ESP8266.
Thanks!

dentra, 2022-01-20:
Now fixed. Thanks.