esphome/issues #2960: publish_json() within lambda function doesn't compile with 2022.1.0

wrmulf, 2022-01-19:
### The problem

The syntax of publish_json() seems to be changed. The code compiles with previous version of ESPHome (2012.12.3), the documentation didn't change.

### Which version of ESPHome has the issue?

2022.1.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.12.9

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
on_code_received:
  - logger.log: "Code received"
  - lambda: |-
      char buffer [10];
      std::string code = itoa(data.code,buffer,16);
      std::string sync = itoa(data.sync,buffer,16);
      std::string high = itoa(data.high,buffer,16);
      std::string low = itoa(data.low,buffer,16);
      std::transform(code.begin(), code.end(), code.begin(), ::toupper);
      std::transform(sync.begin(), sync.end(), sync.begin(), ::toupper);
      std::transform(high.begin(), high.end(), high.begin(), ::toupper);
      std::transform(low.begin(), low.end(), low.begin(), ::toupper);
      id(${devicename}_code).publish_state("0x" + code);
      id(${devicename}_mqtt).publish_json("rf/code", [=](JsonObject &root) {
        root["sync"] = "0x" + sync;
        root["low"] = "0x" + low;
        root["high"] = "0x" + high;
        root["code"] = "0x" + code;
      });
```

### Anything in the logs that might be useful for us?

```txt
root@zbox:~/esphome# esphome compile rf.yaml
INFO Reading configuration rf.yaml...
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Compiling app...
Processing rf (board: esp01_1m; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <AsyncMqttClient-esphome> 0.8.6
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ArduinoJson> 6.18.5
|-- <ESP8266mDNS> 1.2
Compiling .pioenvs/rf/src/main.cpp.o
rf.yaml: In lambda function:
rf.yaml:91:8: error: no matching function for call to 'esphome::mqtt::MQTTClientComponent::publish_json(const char [8], setup()::__lambda18::__lambda19)'
rf.yaml:91:8: note: candidate is:
In file included from src/esphome/components/mqtt/custom_mqtt_device.h:7:0,
                 from src/esphome.h:12,
                 from src/main.cpp:3:
src/esphome/components/mqtt/mqtt_client.h:207:8: note: bool esphome::mqtt::MQTTClientComponent::publish_json(const string&, const json_build_t&, uint8_t, bool)
   bool publish_json(const std::string &topic, const json::json_build_t &f, uint8_t qos = 0, bool retain = false);
        ^
src/esphome/components/mqtt/mqtt_client.h:207:8: note:   no known conversion for argument 2 from 'setup()::__lambda18::__lambda19' to 'const json_build_t& {aka const std::function<void(ArduinoJson6185_91::ObjectRef)>&}'
*** [.pioenvs/rf/src/main.cpp.o] Error 1
===================================================================================================== [FAILED] Took 6.95 seconds =====================================================================================================
```

wrmulf, 2022-01-19:
Fix:

`on_code_received:
  - logger.log: "Code received"
  - lambda: |-
      char buffer [10];
      std::string code = itoa(data.code,buffer,16);
      std::string sync = itoa(data.sync,buffer,16);
      std::string high = itoa(data.high,buffer,16);
      std::string low = itoa(data.low,buffer,16);
      std::transform(code.begin(), code.end(), code.begin(), ::toupper);
      std::transform(sync.begin(), sync.end(), sync.begin(), ::toupper);
      std::transform(high.begin(), high.end(), high.begin(), ::toupper);
      std::transform(low.begin(), low.end(), low.begin(), ::toupper);
      id(${devicename}_code).publish_state("0x" + code);
      id(${devicename}_mqtt).publish_json("rf/code", [=](JsonObject root) {
        root["sync"] = "0x" + sync;
        root["low"] = "0x" + low;
        root["high"] = "0x" + high;
        root["code"] = "0x" + code;
      });`

The JSON-Object ist no longer passed by reference.

closed 2022-01-19 by wrmulf

jesserockz, 2022-01-19:
Yup, as per the section in the changelog =)
https://esphome.io/changelog/2022.1.0.html#json