esphome/issues #2964: SONOFF Basic no longer connects to Unifi AP after update to 2022.1.1

user897943, 2022-01-20:
### The problem

Prior to 2022.1.1 it was working absolutely fine, I had to tweak the framework to "latest", because "recommended" was doing the same thing where the SB was not connecting to my Unifi APs. After changing framework to "latest" it did.

However, whatever this new update to ESPHome did, it bricked the devices again.

My config (replaced private info with "xxxxxxx":

```
########################## SONOFF GENERAL CONFIG #############################
###########################  Aquarium BASIC  #################################

esphome:
  name: aquarium

esp8266:
  board: sonoff_basic
  framework:
    version: latest

wifi:
  ssid: Kumshott
  password: !secret xxxxxxxxxxxx
  power_save_mode: none
  reboot_timeout: 15s
  output_power: 20dB
  fast_connect: true
#  manual_ip:
#    static_ip: xxxxxxx
#    subnet: xxxxxxxx
#    gateway: xxxxxxxxx

  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "Aquarium Fallback Hotspot"
#    password: !secret xxxxxxxx

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret xxxxxxx

ota:
  safe_mode: true
  password: !secret passota

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Aquarium Lights Sonoff Button"
    on_press:
      then:
      - light.toggle: light1

output:
  - platform: gpio
    id: relay
    pin: GPIO12
    
light:
 - platform: binary
   name: "Aquarium Lights"
   id: light1
   output: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes
    
sensor:
  - platform: wifi_signal
    name: "Aquarium Lights WiFi signal"
    update_interval: 60s

  - platform: uptime
    name: "Aquarium Lights device uptime"

text_sensor:
  - platform: version
    name: "Aquarium Lights ESPHome version"
```

### Which version of ESPHome has the issue?

2022.1.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

Sonoff Basic

randybb, 2022-01-20:
Not Sonoff Basic again...
Please connect it via USB-UART or any other 3.3 V PSU and if it will boot up, then the issue is related to dried capacitors (a known issue of Sonoff Basic).

jarkastr, 2022-01-21:
This issue is not exclusive to sonoff basic.  I had the same issue with 3 other esp8266 devices on 2021.12.3.  They were auto connecting to wifi without a problem, I flashed 12.3 and then they would not auto connect to my unifi ap.

randybb, 2022-01-21:
I have bunch of esp8266/esp32 nodes and never had any issue connection to my UniFi WiFi, so please provide some logs. You need to connect directly to UART, ideally powering it via the build in PSU first (connect only rx, tx and gnd) and if it is not able to connect, disconnect AC and connect 3.3V.

user897943, 2022-01-21:
> I have bunch of esp8266/esp32 nodes and never had any issue connection to my UniFi WiFi, so please provide some logs. You need to connect directly to UART, ideally powering it via the build in PSU first (connect only rx, tx and gnd) and if it is not able to connect, disconnect AC and connect 3.3V.

Dude... you are NOT meant to EVER connect to AC power while connected to serial. Like, NEVER EVER. That's a massive electrocution risk. You only need to use the serial to USB adaptor's voltage. I'm not sure if you meant in that way but if not it's really poorly phrased. Also, good for you that you never had issues!

Do you want to send me one of your Sonoff Basics to test? What UniFi gear do you use? What is the version of your Network application? What is the version of the UniFi equipment firmware? What is the version of your ESPHome? I find it hilarious when people ask for logs then say their stuff is fine but they don't give any info either.

My stuff was working fine until I installed this version, the previous version was the one just preceding. The two Sonoff Basics did not fail at the same time, the chance of that happening is slim to none. This is a recurring issue with ESPHome. I already got in touch with Ubiquiti but as always they ignore it.

Someone on the ESPHome project already knows what the issue is, they confirmed it's because of UniFi. If ESPHome knows, and Ubiquiti isn't doing anything, I think it would be easier for ESPHome contributors to develop a workaround in ESPHome than waiting for Ubiquiti to bother. In the meantime users have to be dealing with this huge ball-ache.

I don't want to have to flash a board that lives under my fish tank every time ESPHome updates. It sucks and needs to be fixed. I'm not the only person this is affecting and this is not a "dry capacitor" issue. It's a buggy ESPHome issue. This only ever affects the Sonoff Basic. I have Sonoff POWR2 and it's fine. I have two generic LED controllers using ESP8266 and it's fine too. All the other WiFi stuff connects absolutely fine to my Ubiquiti APs. I'm not and nobody will change their networking for one device that ESPHome refuses to play nice with.

If developers on this project don't want to tackle the problem when they know about it, then please remove support for the device - yep, I mean for everyone, even people who don't use Ubiquiti. If you're not gonna support a major wireless vendor that's discrimination.

randybb, 2022-01-21:
Ha ha. In case of Sonoff Mini which uses a capacitive dropper it would be an issue, but not in Basic with uses a proper transformer. If you are scared, you can use a properly isolated uart - it is in your hands. I know what I do my hands and I am fully confident.
It is just an esp8266. ESPHome is not able to distinguish if it is esp8266 in while box or just a bare pcb like NodeMCU or D1. Which means the issue is in device itself and since people here have some experiences with Sonoff Basic, imost of the times the problem is the same.
You are who is having issues, without logs it is not possible properly react. I can say I have 0 issues with properly configured wifi network. Nobody is saying that Ubiquiti is not supported. On Ubiquiti gear you have more options you can set, which means more things that can go wrong, but this will affect devices sharing the same mcu or software, not only an esp8266 in a white box.

user897943, 2022-01-21:
> Ha ha. In case of Sonoff Mini which uses a capacitive dropper it would be an issue, but not in Basic with uses a proper transformer. If you are scared, you can use a properly isolated uart - it is [...]

Because you have the knowledge you also have the responsibility to make it clear when something could be fatal. This is actually stated right on the Tasmota page for example. You could have newbies coming here from Google with zero clue about the risk, so it's not something to laugh about. No, I am not the only person having the issue but I also won't do homework for you by linking issues that are still open on the bug tracker about this very same thing. My WiFi is properly configured, whatever that means.

The only devices that ever have problems are Sonoff Basic, even the Sonoff POWR2 are fine like I said.

pengelen, 2022-01-21:
Did you try a basic config without any sensors, lights and switches?
Someone mentioned before that the firmware size could be the issue.
https://github.com/esphome/issues/issues/2731

user897943, 2022-01-21:
Nope, because why? The firmware uploads, when uploading it shows the image size is half of the flash size and within memory limits with capacity to spare. Also, it was working on 2022.1.0. I know because I just tested:

```
INFO Reading configuration /config/Backyard String Lights.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing stringlights (board: sonoff_basic; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
Compiling .pioenvs/stringlights/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/stringlights/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/stringlights/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/stringlights/src/esphome/components/version/version_text_sensor.cpp.o
Compiling .pioenvs/stringlights/src/esphome/core/application.cpp.o
In file included from src/esphome/components/version/version_text_sensor.h:4,
                 from src/esphome/components/version/version_text_sensor.cpp:1:
src/esphome/components/version/version_text_sensor.cpp: In member function 'virtual void esphome::version::VersionTextSensor::dump_config()':
src/esphome/components/text_sensor/text_sensor.h:12:3: warning: 'nonnull' argument 'this' compared to NULL [-Wnonnull-compare]
   12 |   if ((obj) != nullptr) { \
      |   ^~
src/esphome/components/version/version_text_sensor.cpp:21:41: note: in expansion of macro 'LOG_TEXT_SENSOR'
   21 | void VersionTextSensor::dump_config() { LOG_TEXT_SENSOR("", "Version Text Sensor", this); }
      |                                         ^~~~~~~~~~~~~~~
Compiling .pioenvs/stringlights/src/esphome/core/util.cpp.o
Compiling .pioenvs/stringlights/src/main.cpp.o
Linking .pioenvs/stringlights/firmware.elf
RAM:   [====      ]  37.9% (used 31060 bytes from 81920 bytes)
Flash: [====      ]  38.1% (used 390565 bytes from 1023984 bytes)
Building .pioenvs/stringlights/firmware.bin
esp8266_copy_factory_bin([".pioenvs/stringlights/firmware.bin"], [".pioenvs/stringlights/firmware.elf"])
========================= [SUCCESS] Took 56.08 seconds =========================
INFO Successfully compiled program.
INFO Connecting to xxxxxxx
INFO Uploading /config/.esphome/build/stringlights/.pioenvs/stringlights/firmware.bin (394720 bytes)
INFO Compressed to 274229 bytes
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from xxxxxx using esphome API
WARNING Can't connect to ESPHome API for xxxxxxxxxx: Timeout while connecting to ('xxxxxxxxxx', 6053)
INFO Trying to reconnect to xxxxxxxxx in the background
INFO Successfully connected to xxxxxxxxxx
[13:54:04][I][app:102]: ESPHome version 2022.1.0 compiled on Jan 21 2022, 13:52:39
[13:54:04][C][status_led:019]: Status LED:
[13:54:04][C][status_led:020]:   Pin: GPIO13
[13:54:04][C][wifi:488]: WiFi:
```

This is not an issue with the device, people need to accept that it's a problem with ESPHome and stop acting so defensive, no code is perfect why are people acting like this one is bug-free? Please fix.

smoke007, 2022-01-22:
@user897943,
I'm here too because one of the last ESPHome updates started corrupting wifi again on my Sonoff Basics when pushing the latest update to them.  I have about 30 of these boards in my home.  The rabbit hole goes deep with the Sonoff Basic's wifi issues and ESPHome.  See https://github.com/esphome/issues/issues/455 and note OttoWinter's listed "common fixes".  This really is about the best advice.  This issue has been around since Jun 2019.

For your issue, I'm guessing if you take your compiled bin and flash it with your FTDI adapter, your wifi will continue to show in logging that it won't connect.  This is where the workaround of disabling logger or changing random things in your config will sometimes fix the bin corruption.  Doing a "Clean build files" will not fix it.  :(

Here's what I've had to do to get mine working again.  Create a new ESPHome device, give it a new or temp name, compile and flash it to your Basic.  It should now communicate with ESPHome again over the wifi.  If you flash back the previous bin, wifi will be broken again.  Something gets corrupted in the compile process.

Now you can copy your whole config from below the "esphome:" section, from the broken device config, and paste it into your new device config.  You must keep the "esphome:" section as is if you want to OTA update it, because it includes the new device's name.  Now OTA flash this to your Basic.  

If you're fine with the device's new name in ESPHome, you're done.  HA should discover your new ESPHome device.

If you want to change the devices name back to the old one, delete the old broken device now from ESPHome and follow this process to rename.  https://esphome.io/components/esphome.html#esphome-changing-node-name

After writing that all out, I realized that you might be able to just copy everything from the bad device's config file, delete the old device, then create a new device using the same old device's name, then paste in the config info.  However, I can't remember if that works.  My theory is that something is corrupted during the compile, so I don't know if the new device with the old device name will using the same old path and cached data to compile another corrupt bin with this method.  Let us know if you test it, however the above 1st method works for me.

nagyrobi, 2022-01-22:
Tasmota has some commands to [reset](https://tasmota.github.io/docs/Commands/#management) the wifi self-calibration data in the chip, which afaik is not overwritten by a simple flash erase or ota.

user897943, 2022-01-22:
> Tasmota has some commands to [reset](https://tasmota.github.io/docs/Commands/#management) the wifi self-calibration data in the chip, which afaik is not overwritten by a simple flash erase or ota.

Are you suggesting that this could potentially resolve? If ESPHome fails to boot, then there should be a routine to auto-reset wifi calibration after maybe 10 failures?

nagyrobi, 2022-01-22:
One has to also power-cycle the unit after resetting the wifi calibration data, at least that's how it's done in Tasmota - so I doubt it can be automated like that.

user897943, 2022-01-22:
I found that this advice worked:
https://github.com/esphome/issues/issues/2912#issuecomment-1013762264

I removed safe_mode from the ota: section, and I removed the entire logger: section and it's working now. So perhaps ESPHome devs could use this as a starting point.

jarkastr, 2022-01-23:
I did not have safe_mode on, but removing the logger fixed my problem as well.

jarkastr, 2022-01-26:
@user897943  what Unifi AP and firmware are you using. I believe I am having the same issue as described in this post on the forums: https://community.ui.com/questions/UAP-AC-LR-All-2-4GHz-clients-lose-communication-while-wifi-connection-persists/73a676e3-14fe-414e-86f7-b545ef9d2eaf

I can only assume it is related to ESPHome devices not connecting to the AP after an update.

user897943, 2022-01-26:
> @user897943 what Unifi AP and firmware are you using. I believe I am having the same issue as described in this post on the forums: [...]

I use a UDM (which contains a built-in Nano HD, and I also have a UAP-AC-HD on the 2nd floor.

The UDM is on Early Access FW 1.11.1 and the Network Application is also Early Access on version 7.0.18.

jov58, 2022-01-26:
I just bought a Sonoff Basic R2, the board inside is labeled  `Sonoff RF R2 POWER V1.3 2019.08.16`. It has a blue LED by the way, in contrast to some sources saying the LED is only blue as of rev 1.4 and green prior to that.

Couldn't get it to work with ESPHome and [this configuration](https://esphome.io/devices/sonoff_basic.html#sample-configuration), but I can with Tasmota or a test program of my own with PlatformIO.

But I finally got ESPHome working by first flashing Tasmota, connecting it to WiFi and then using the OTA upload from Tasmota to upload the firmware.bin produced by ESPHome. After power cycling the blue LED show a short flash and six flashes a second apart. The Sonoff is visible on the network now and I can integrate it in Home Assistant with no problem so far.

A workaround perhaps for some and hope it helps to solve the problem.

BertrumUK, 2022-02-09:
Just updated some of my sonoff basics V1 and lost access to 1 of them. The captive portal came up but I could not connect to it. Tasmota basic bin worked ok until I tried to upload my ESPHome config file. In the end removing the captive portal part of the config allowed me to flash it successfully so it does seem to be related to file size IMO.

MariusLaurinaitis, 2022-02-21:
Same here, Sonoff basic after update wont connect to wifi.

smoke007, 2022-02-24:
> The captive portal came up but I could not connect to it. 

The captive portal doesn't work correctly for me on any Sonoff Basics either.  It seems to crash and reboot as soon as captive portal loads up.  I haven't tested other devices with more memory, but guessing the Basics just can't handle it.  It would be nice if it worked on these devices since it would have saved me a number of times from having to hardwire flash these.

MariusLaurinaitis, 2022-02-27:
Ok, I just remove part from yaml
----
status_led:
  pin:
    number: GPIO13
    inverted: yes
----

and all my SonoffBasic online again and working.

fbloemhof, 2022-04-30:
> ## Ok, I just remove part from yaml
> ## status_led:
> pin:
> number: GPIO13
> inverted: yes
> and all my SonoffBasic online again and working.

Nice! My only SonoffBasic I use was broken for several months. Today I had some time to figure out why and indeed removing the status_led config from the yaml fixed it.

krasatos, 2022-05-05:
Got 4 sonoffs with esphome that stopped working last week suddenly.
Is there some kind of downgrade i can do (preferably on the unifi - as the esphomes are not really accessible) that could revert the whole situation?
Thank you.

nagyrobi, 2022-05-05:
Any setting changed in Unifi lately? Maybe disable WPA3?