esphome/issues #2966: DSMR: Garbage Data with Kaifa MA105C

thedevleon, 2022-01-21:
### The problem

This is a follow-up on Issue #2621. As suggested in the last commit, I'm making a new issue.

So, in a nutshell, I have a Kaifka MA105C I want to read with an ESP8266. For this I've connected the RTS line to 3.3V, the Data to D7 (hardware UART), DATA_GND to GND and the ESP itself is powered externally via an USB power supply.

This is the config I first tried:

```
logger:
  baud_rate: 0
  level: VERBOSE

uart:
  rx_pin: D7
  baud_rate: 115200
  rx_buffer_size: 1700

dsmr:
    max_telegram_length: 1700

sensor:
  - platform: dsmr
    energy_delivered_tariff1:
      name: Energy Consumed Tariff 1

text_sensor:
  - platform: dsmr
    identification:
      name: "DSMR Identification"
    p1_version:
      name: "DSMR Version"
```

With this, all I'm getting in the console is the following:

```
[10:34:50][V][dsmr:142]: Header of telegram found
[10:34:50][V][dsmr:142]: Header of telegram found
[10:34:50][V][dsmr:142]: Header of telegram found
[10:34:51][W][dsmr:092]: Timeout while reading data for telegram
[10:35:00][V][dsmr:142]: Header of telegram found
[10:35:00][V][dsmr:142]: Header of telegram found
[10:35:00][V][dsmr:142]: Header of telegram found
[10:35:00][V][dsmr:142]: Header of telegram found
[10:35:01][W][dsmr:092]: Timeout while reading data for telegram
```

As such, I followed the things mentioned in #2621, and added debugging for the UART port as well as the different branch with fixes for DSMR:

```
logger:
  baud_rate: 0
  level: VERBOSE

uart:
  rx_pin: D7
  baud_rate: 115200
  rx_buffer_size: 1500
  debug:
    direction: RX
    after:
      delimiter: "\r\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
  
external_components:
  - source:
      type: git
      url: https://github.com/mmakaay/esphome
[... 147 lines cut ...]
  }
}
```

Any idea what could be going wrong? I really don't get why the ESPHOME and FTDI adapter only read garbage, while the same ESP8266 with a super simple sketch reads the data (mostly) fine...

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

dsmr

thedevleon, 2022-01-21:
@mmakaay maybe you can shed some insight on this?

thedevleon, 2022-01-21:
Well, turns out the fix was pretty easy, only took a lot of digging to find out... Turns out the signal still needs to be inverted. I had thought this was standard for DSMR, but guess it's specific to the Kaifa meters.

```
uart:
  rx_pin:
    number: D7
    inverted: true
  baud_rate: 115200
  rx_buffer_size: 1500
```

closed 2022-01-21 by thedevleon

mmakaay, 2022-01-21:
It's part of the P1 port specification that the signal is inverted. So inverting it in the DSMR reader device is indeed needed, good catch.

Note that you can drop the `external_components` configuration when you're using a recent version of ESPHome. The changes have all been accepted into the production code.