esphome/issues #2967: ota.on_begin runs after the file is uploaded

maruel, 2022-01-22:
### The problem

[ota.on_begin](https://esphome.io/components/ota.html?highlight=ota#ota-on-begin) event is called after the upload is completed while it's being applied. While it makes sense semantically, it's not what I expected from a user experience, I would consider the upload phase to be part of the OTA.

With the snippet below, notice that the LED changes colors after the upload, not before.

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

m5stak pico32

### Component causing the issue

ota

### Example YAML snippet

```yaml
# With a colored LED set as "led":
ota:
  (...)
  on_begin:
    then:
      - light.control:
          id: led
          state: on
          brightness: 100%
          red: 100%
          green: 0%
          blue: 100%
```

martgras, 2022-01-22:
`on_begin` is called when uploading starts.

You are probably running into the issue described in the documentation:

An OTA update blocks the main loop during its operation. This means that you won’t be able to represent state changes using components that update their output only from within their loop() method. Explained differently: if you try to display the OTA progress using component X, but the update only appears after the OTA update finished, then component X cannot be used for providing OTA update feedback.

To verify try 

```yaml
on_begin:
    then:
      - logger.log: "OTA start"
      - light.control:
          id: led
          state: on
          brightness: 100%
          red: 100%
          green: 0%
          blue: 100%
```

maruel, 2022-01-22:
Would forcibly running a single loop() (only the inner parts for output components) alleviate some of the problem?

martgras, 2022-01-22:
I don't know. 
GPI0 output should be ok
Also I'm a bit hesitant to dig into the code since we had a couple of OTA issues recently. My personal opinion is that OTA stability is extremely important. Doing other work while OTA is running isn't something I see a pressing need. It's not something done very frequently (if it works why update very often) and usually takes very little time.
Aa I said that's my personal view only - I may be wrong

maruel, 2022-01-22:
Fair point, I'm fine with closing considering wontfix given the risks.

martgras, 2022-01-23:
Cloing per last comment

closed 2022-01-23 by martgras