esphome/issues #2973: ac_dimmer trigger too slow

Th3M1k3y, 2022-01-23:
### The problem

When using the ac_dimmer component I can't get it to trigger at the zero crossing.

I am not sure if it is a software or hardware problem, because it could very well also be the software triggering where it should, but the voltage needing come into the right rage for the zero point detect to notice it.

The problem I see is that it very consistently trigger 1.5ms too late, but if there were an option to delay the trigger I could just delay it by 8.5ms (50hz). That would solve the problem for me, as it would then just wait, and when it then trigger it would be at the next half wave.

[image]
.

### Which version of ESPHome has the issue?

2022.1.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

ac_trigger

### Example YAML snippet

```yaml
output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: D2
    method: leading
    min_power: 0.5
    zero_means_zero: true
    zero_cross_pin:
      number: D8
      mode:
        input: true
      inverted: true
      
fan:
  platform: speed
  name: "${friendly_name} - Motor speed"
  id: fan_speed
  output: dimmer1
```

Th3M1k3y, 2022-01-25:
I found a closed issue about some of the same behavior, in that one it was suggested to use another chip to detect the zero crossing, so I will start by trying a H11L1M instead of the one on the dimmer module.

Th3M1k3y, 2022-01-25:
> I found a closed issue about some of the same behavior, in that one it was suggested to use another chip to detect the zero crossing, so I will start by trying a H11L1M instead of the one on the dimmer module.

On my quest to fix the behavior of the delayed trigger on the waveform, I contacted a friend to ask him about how to use the H11L1M properly. During that he could calculate the trigger value on the RobotDyn module I use, and it turns out it is doing exactly what is to expected of it, so the waveform I was able to capture, correspond to the voltage the module trigger at. The problem is the ability to trigger directly at zero, because of too high heat dissipation in the resistors. Triggering a bit higher solve that issue, but leads to the little spike in the beginning of the wave. The "easy" solution is to be able to offset the triggerpoint.

ErikT80, 2025-03-31:
> On my quest to fix the behavior of the delayed trigger on the waveform, I contacted a friend to ask him about how to use the H11L1M properly. During that he could calculate the trigger value on the [...]

Did you ever find a solution for the robotdyn dimmer?