esphome/issues #2974: lambda expression for count of a repeat action is not working within an interval.

MarkKananen, 2022-01-23:
### The problem

#create a global variable counter and set it to 3
interval:
  - interval: 30s
    then:
      - repeat:
          count: !lambda 'return id(counter);'
          then:
            - switch.turn_toggle: buzzer
            - delay: 1s
count is never evaluated correctly.

### Which version of ESPHome has the issue?

2022.1.1 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.10

### What platform are you using?

ESP8266

### Board

NodeMCU

### Component causing the issue

interval

### Example YAML snippet

```yaml
#create a global variable counter and set it to 3
interval:
  - interval: 30s
    then:
      - repeat:
          count: !lambda 'return id(counter);'
          #count is never evaluated correctly.
          then:
            - switch.turn_toggle: buzzer
            - delay: 1s
```

### Anything in the logs that might be useful for us?

```txt
I've had trouble trying to determine what it is actually doing, seems to be evaluating to infinite or some other large number.  If you let it go too long, the intervals start stacking up on top of one another.
```

### Additional information

As a work around, the same code has gone into a switch that is toggled by the interval.

oxan, 2022-01-24:
What is `counter`? Usually you want to use the state of the object, not the ID itself (which is a pointer, not a value).

martgras, 2022-01-26:
I tried to repoduce the issue with the yaml below 
Works as expected for me. the global counter is updated to a random value after interval

```yaml
esphome:
  name: esp32s2-01

globals:
  - id: counter 
    type: int
    restore_value: no
    initial_value: '5'
esp32:
  board: nodemcu-32s

interval:
  - interval: 30s
    then:
      - repeat:
          count: !lambda 'return id(counter);'
          then:
            - lambda: |-
                static int i = 0 ; 
                ESP_LOGI("demo","Repeat count=%d , iteration=%d",id(counter),++i);
      - lambda: |-
          // set counter to a random value 
          id(counter) = (random_uint32() % 4) + 1;
```
logs
```
12:25:47][I][demo:026]: Repeat count=1 , iteration=6
[12:26:17][I][demo:026]: Repeat count=3 , iteration=7
[12:26:17][I][demo:026]: Repeat count=3 , iteration=8
[12:26:17][I][demo:026]: Repeat count=3 , iteration=9
[12:26:47][I][demo:026]: Repeat count=4 , iteration=10
[12:26:47][I][demo:026]: Repeat count=4 , iteration=11
[12:26:47][I][demo:026]: Repeat count=4 , iteration=12
[12:26:47][I][demo:026]: Repeat count=4 , iteration=13
[12:27:17][I][demo:026]: Repeat count=4 , iteration=14
[12:27:17][I][demo:026]: Repeat count=4 , iteration=15
[12:27:17][I][demo:026]: Repeat count=4 , iteration=16
[12:27:17][I][demo:026]: Repeat count=4 , iteration=17
[12:27:47][I][demo:026]: Repeat count=4 , iteration=18
[12:27:47][I][demo:026]: Repeat count=4 , iteration=19
[12:27:47][I][demo:026]: Repeat count=4 , iteration=20
[12:27:47][I][demo:026]: Repeat count=4 , iteration=21
[12:28:17][I][demo:026]: Repeat count=2 , iteration=22
[12:28:17][I][demo:026]: Repeat count=2 , iteration=23
```

michaelblight, 2022-05-26:
I came here because I was having the same problem. However, I was resetting the counter to 1 at the end of the loop, but my indentation was bad such that it was inside the loop, not after it. So it looks like 'count:' is re-evaluated after each loop, and resetting it to a number less than the iterations will stop the loop. You can see this if you add `id(counter) = 1;` after the `ESP_LOGI` above. then you get something like:
```
[18:13:36][I][demo:031]: Repeat count=2 , iteration=1
[18:14:06][I][demo:031]: Repeat count=4 , iteration=1
[18:14:36][I][demo:031]: Repeat count=3 , iteration=1
```
Maybe the OP had the same sort of issue?

ssieb, 2022-12-05:
There doesn't appear to be an issue here.  If there is, then reopen and provide more information.

closed 2022-12-05 by ssieb