esphome/issues #2975: remote network device stays offline

redstorm1, 2022-01-24:
### The problem

Home Assistant is on a 192.168.1.0/24 network and i have a second remote network 192.168.2.0/24, routing between networks is in place.
A ESP8266 on the 192.168.2.x network stay offline. 
Setting a static IP 192.168.2.200 on the ESP the device will stream logs and update OTA but stays offline in HA.
Pinging the ip of the device works from the home assistant server, pinging via name.local does. Adding an entry in the servers hosts file for the device enables ping via name.local but remains offline in HA.

### Which version of ESPHome has the issue?

2022.1.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

core-2021.12.10

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

esphome API

### Example YAML snippet

```yaml
esphome:
  name: terrace
  platform: ESP8266
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "0492ba9a250392029a52db84bd15c8dd"

wifi:
  ssid: "wifi"
  password: "pass"
  use_address: 192.168.2.200
  manual_ip:
    static_ip: 192.168.2.200
    gateway: 192.168.2.254
    subnet: 255.255.255.0
```

### Additional information

updating to status_use_ping, the device is shown online but not device or entity is available to configure in HA.

The networks are configured via a wireguard VPN with routing and firewalls updated and ping works from each network to each device and vice versa.

I suspect HA is not connecting out to192.168.2.200:6053 to the API on the ESP8266 even though it can ping it, and stream the logs and update over the air just does not connect to the API

redstorm1, 2022-01-24:
opening a shell in the esphome docker container and verified the container can ping the device on the remote network.
`root@a0d7b954-esphome:/# ping 192.168.2.200
PING 192.168.2.200 (192.168.2.200) 56(84) bytes of data.
64 bytes from 192.168.2.200: icmp_seq=1 ttl=252 time=12.9 ms
64 bytes from 192.168.2.200: icmp_seq=2 ttl=252 time=23.6 ms
`

jesserockz, 2022-01-24:
Do you have your HA logs showing it not being able to connect to the device?
If you have to set `status_use_ping` in dashboard, then I suspect the mDNS is not being repeated from remote network into main network. Its not a case of firewall rules, but you specifically need a repeater.

redstorm1, 2022-01-25:
I have setup on configured mdnstunneller and have partial success. 
After a restart of ESPHome they are discoverable then go offline in about 60 seconds, then periodically they will come online briefly.

The remote devices are able to connect to the API on 6053 and are controllable via HA, just a issue with staying contactable from HA to the device.

how long do mDNS broadcasts get cached for?

redstorm1, 2022-01-25:
To get mdnstunneller working you need to add

```
serviceWhitelist:

"_esphomelib._tcp.local"
```
To both the server and the client to get it partially working.

dhover, 2022-02-16:
I am using the "status_use_ping" option. After removing ESPHome from the Community Addons and installing the ESPHome from the ESPHome Repo (see ESPHome 2022.2.0 release notes) the configuration option "status_use_ping" is removed.
So you have to add that option again. I think it is good to add a small remark in the release notes!