esphome/issues #2976: Add Momentary Switch example and link to Button Documentation

borpin, 2022-01-24:
### The problem

There is no example of a momentary switch on the Button documentation.  I'd suggest this is a very likely use of this so would be useful to include it.

Links between the Momentary switch docs and the Button docs would also benefit users.

### Which version of ESPHome has the issue?

Docs

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
A configuration that works for me and creates a Button Entity in HomeAssistant.

switch:
  - platform: gpio
    pin: GPIO4
    name: "Garage Door 2 Relay"
    id: relay
    restore_mode: ALWAYS_OFF
    internal: true

button:
  - platform: template
    name: "Garage Door 2"
    icon: "mdi:gate"
    on_press:
    - switch.turn_on: relay
    - delay: 500ms
    - switch.turn_off: relay
```
```

oxan, 2022-01-24:
A better implementation would be something like this, that avoids the unnecessary switch:
```yaml
output:
  - platform: gpio
    id: out
    pin: GPIO4

button:
  - platform: template
    on_press:
      - output.turn_on: out
      - delay: 500ms
      - output.turn_off: out
```

Momentary switches in general are a hack from when buttons didn't exist, so I'd rather remove them from the documentation than make them more prominent.

An `output` button would probably also be useful, so that you can do things like this:
```yaml
output:
  - platform: gpio
    id: out
    pin: GPIO4

button:
  - platform: output
    output: out
    duration: 500ms
```

borpin, 2022-01-24:
> A better implementation would be something like this, that avoids the unnecessary switch:

I have no problem with this. My thoughts were mainly that there should be something in the docs as it seems a common task that needs a solution.

My only thought is ensuring the relay is in a known state at start. Would that example hide it from HA?

> An `output` button would probably also be useful

As I say, I am not fussed what the correct solution is, just I think there should be something in the docs 😀

oxan, 2022-01-25:
> My thoughts were mainly that there should be something in the docs as it seems a common task that needs a solution.

Yes, I agree. I've implemented the `output` button with documentation for the next release (see [here](https://next.esphome.io/components/button/output.html)), but I'll leave this one open as something should also be done with the Momentary Switch section.

> My only thought is ensuring the relay is in a known state at start. Would that example hide it from HA?

Yes, that's true, but maybe instead of exposing the state to allow the user to check it, it's better if we force it to be off on startup.

borpin, 2022-01-25:
> > My only thought is ensuring the relay is in a known state at start. Would that example hide it from HA?
> 
> Yes, that's true, but maybe instead of exposing the state to allow the user to check it, it's better if we force it to be off on startup.

That was what I did in the original switch setup - can the same be done for a `output` setup? I'd suggest adding it to the example if you can.

I'll put this out here - the `output` docs are really confusing! Too many different things all called `output`!

I would have guessed this `output` would be the Generic Switch Output. If so, then I can see there is a `restore_mode`. However, it looks like the doc you have created is a GPIO Output, that doesn't.

Is there a robust solution to ensure 'off' at startup? I wouldn't want a random 'on' (which would open the garage door) on boot 😆 

A note from the Momentary docs to this button might suffice? When I first did this (as a momentary switch) it took me quite a while to find the right bit of the docs.

oxan, 2022-01-25:
> I'll put this out here - the `output` docs are really confusing! Too many different things all called `output`!

Well, there are many different outputs. An output is anything that can output a signal: a GPIO pin, an AC dimmer, various I/O port expanders, or even something more higher level such as Bluetooth signal. It makes sense that they're all called `output`, because at a high level they all do the same. All these components are interchangable everywhere an output is used. For example, with the output button I made it's also possible to trigger a Bluetooth signal. 

> I would have guessed this `output` would be the Generic Switch Output. If so, then I can see there is a `restore_mode`. However, it looks like the doc you have created is a GPIO Output, that doesn't.

There's no Generic Switch Output, there's a Generic Output Switch. The difference seems subtle, but it's important: it is a switch that uses an output, not an output that uses a switch. 

The button I created is a Generic Output Button: it's a button that uses an output. There's no `restore_mode`, as there's no restore mode for buttons (nor for outputs), and there's no switch involved at all. The example I show uses a GPIO Output, but it could use any other output as well. 

I do agree it would be nice if there's some high-level introduction about how the different platforms (outputs/switches/buttons/etc) relate to each other, but that's not related to this particular implementation.
 
> Is there a robust solution to ensure 'off' at startup? I wouldn't want a random 'on' (which would open the garage door) on boot 😆

ESPHome always initializes outputs to off on boot. Depending on how you've wired the ESP and garage door it might be high when the ESP is off or on boot, but there's nothing ESPHome can do about that from software.

> [edit] Just to say in the doc you have created, it is a little unclear which block the 'Configuration Variables' refers to (`output` or `button`).

Hmm, I see your point, maybe I should move the GPIO stuff to a separate example below.

borpin, 2022-01-25:
> Well, there are many different outputs.

I can see that now and while a Generic Switch Output is different to a Generic Output Switch, the logic doesn't follow through. Example Binary Light - firstly it is defined in a different order in the example (which I realise in itself doesn't matter but I'm a sucker for consistency), but also should it be an Output Light (rather than Binary Light) to follow the same logic?

Sorry if it seems I am splitting hairs, but I do find the docs a bit of a maze and not very consistent (and yes, docs are hard!).

> Hmm, I see your point, maybe I should move the GPIO stuff to a separate example below.

I feel all of the `output` needs a bit of rework so it is consistent and I think a general intro section to `outputs` would also help, including things like how they are initialised, the fact they need 2 parts etc. One that links an action to the hardware and a second that initiates that link (perhaps not the best description).

From my perspective, I'm driving a relay which is a form of switch. The platform used is really an HA thing and helps HA determine what to do with the entity which is also an important concept.

Thanks for your time and effort to improve the docs - it is appreciated 👍

disconn3ct, 2022-12-14:
> there should be something in the docs

Unfortunately [there is](https://esphome.io/components/switch/gpio.html?highlight=momentary#momentary-switch). Lots of issues (HA never sees the toggle unless it is a LONG momentary) but its documented.

I just switched over to button and its definitely more correct. What can I do to help get the docs updated with button?

closed 2023-02-10 by borpin