esphome/issues #2977: OSError: [Errno 105] No buffer space available

JeffrySteegmans, 2022-01-24:
### The problem

When I try to connect wireless to the logs of my espHome device I got the following exception:

`Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 825, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 812, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 362, in command_logs
    return show_logs(config, args, port)
  File "/esphome/esphome/__main__.py", line 284, in show_logs
    return run_logs(config, port)
  File "/esphome/esphome/components/api/client.py", line 72, in run_logs
    asyncio.run(async_run_logs(config, address))
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/esphome/esphome/components/api/client.py", line 24, in async_run_logs
    zc = zeroconf.Zeroconf()
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_core.py", line 448, in __init__
    listen_socket, respond_sockets = create_sockets(interfaces, unicast, ip_version, apple_p2p=apple_p2p)
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_utils/net.py", line 373, in create_sockets
    if add_multicast_member(cast(socket.socket, listen_socket), i):
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_utils/net.py", line 288, in add_multicast_member
    listen_socket.setsockopt(socket.IPPROTO_IP, socket.IP_ADD_MEMBERSHIP, _value)
OSError: [Errno 105] No buffer space available`

This is happening on different versions.

### Which version of ESPHome has the issue?

Multiple versions

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

na

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

esphome dashboard

### Anything in the logs that might be useful for us?

```txt
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 825, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 812, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 362, in command_logs
    return show_logs(config, args, port)
  File "/esphome/esphome/__main__.py", line 284, in show_logs
    return run_logs(config, port)
  File "/esphome/esphome/components/api/client.py", line 72, in run_logs
    asyncio.run(async_run_logs(config, address))
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/esphome/esphome/components/api/client.py", line 24, in async_run_logs
    zc = zeroconf.Zeroconf()
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_core.py", line 448, in __init__
    listen_socket, respond_sockets = create_sockets(interfaces, unicast, ip_version, apple_p2p=apple_p2p)
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_utils/net.py", line 373, in create_sockets
    if add_multicast_member(cast(socket.socket, listen_socket), i):
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_utils/net.py", line 288, in add_multicast_member
    listen_socket.setsockopt(socket.IPPROTO_IP, socket.IP_ADD_MEMBERSHIP, _value)
OSError: [Errno 105] No buffer space available
```

martgras, 2022-01-24:
Probably https://github.com/esphome/issues/issues/2669#issuecomment-980811981 resolves the `No buffer space available` error

JeffrySteegmans, 2022-01-24:
I tried that solution but my docker installation doesn't allow recognize sudo or sysctl command.
I get the following exception:

`bash: sudo: command not found`

I create my connection to docker with the following command:

` docker exec -it esphome /bin/bash`

martgras, 2022-01-25:
The settings are for the host not the docker container

JeffrySteegmans, 2022-01-25:
Thanks allot! 
Never thought about executing the commands on my host environment.

closed 2022-01-25 by JeffrySteegmans