esphome/issues #2979: Light component: brightness not showing actual brightness but set brightness (they differ when using large transition times)

trantoriana, 2022-01-25:
### The problem

The problem in short is that the brightness attribute in home assistant does not match the actual brightness of the lights.

Consider following (exaggerated) example indicative for the problem:

*) transition time of the lights set to 100 minutes
*) Having a brightness of 0 at the beginning
*) Setting new brightness to 100
*) Actual brightness of the lights steps nicely from 0 to 100 over the course of the 100 minutes transition time
*) Brightness attribute of the light in home assistent shows 100 from minute 0 (which is the target brightness rather than actual brightness

While this may seem practically a rubbish example, I do have extended transition times as I am controlling my aquarium lights using esphome. As I do light the estatic of slowly increasing the brightness (for me and the fish), the transition time is a very welcome feature which actually works pretty neat.

However, in home assistant I graph the red, bleu and white values to get a insight of the lighting schedule. This graph is based on the light brightness attribute.

As the brightness attribute of the light does not show "actual" brightness but rather "target" brightness, the graph does not really show how the light schedule works in combination with a slow transition time, instead it shows sharp increases in light while in reality the lights slowly increase in brightness over the transition period. If the graph could show actual brightness, it would be much easier to create better light schedules.

Expectation:

*) Either have the "brightness" attribute follow the actual light brightness
*) Introduce a new attribute "current brightness" to follow the actual light brightness and leave the current attribute to keep the target brightness

Either work for me, but in 'light' of backward compatibility adding a new attribute for actual brightness may be more sensible solution.

I understand that in normal cases, this seems like a minor issue... and it is... it is just that my lighting graph would look much more to reality than now.

Thanks, Trant

P.S. If someone would point me to where this could be implemented, I could look myself to see if I can improve the light component myself and then request a merge.
[image]

### Which version of ESPHome has the issue?

2021.12.01

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12

### What platform are you using?

ESP32

### Board

esp32 on a quinled quad board (monochromatic)

### Component causing the issue

light component

nagyrobi, 2022-01-25:
It will likely still not be like you desire, as even the current brightness of the light will only be updated periodically (relatively rarely), say, every 60 seconds. You'll still see stairs, but smaller ones and more of them.

Instead of this, I'd suggest to try alternative graphing methods which would suit better for this task.
The ultimate graphing engine is Grafana, available for HA as an addon.
Also try this CC: https://github.com/alexarch21/history-explorer-card

trantoriana, 2022-01-26:
> It will likely still not be like you desire, as even the current brightness of the light will only be updated periodically (relatively rarely), say, every 60 seconds. 

This is exactly what I want. Having access to the actual value of the brigthness as it goes through transition time.

> You'll still see stairs, but smaller ones and more of them.

but it will be more exact than now, and depending on the graph resolution will mimic a increasing line anyway.

> Instead of this, I'd suggest to try alternative graphing methods which would suit better for this task.

Using a different graphing method like grafana does not make the grap look any better if home assistant is not provided with the actual brightness rather than target brighness. The graph will still be the same.

Perhaps the last picture is a bit misleading, as that is a graph of sending 4 different target brightnesses starting at around 7, then 8, etc...

The should be part is just to show that the sharp increase should be not so steep, even stairs, if the actual brightness was available

The request still stands, how to get access to the actual brightness given ESPHome using transition periods.

nagyrobi, 2022-01-26:
What platform do you use to dim your lights?
Perhaps you could use a lambda inside to pipe out the brightness value to a number component, and graph that instead.

trantoriana, 2022-01-26:
> What platform do you use to dim your lights?

ESPHome, Loght component - Monochromatic

> Perhaps you could use a lambda inside to pipe out the brightness value to a number component, and graph that instead.

As long as the brightness esphome keeps internally as brightness is the target brigthness, also a lambda function will only get the internal (target) brighness... still does not help me.

I would love to use a number sensor based on the actual brigthness, the problem is that actual brightless is not available.

nagyrobi, 2022-01-26:
Is the monocromatic acting on a simple gpio output?

trantoriana, 2022-01-26:
yes, it is defined in esphome like

`light:
  - platform: monochromatic
    name: ${deviceName} ${output2Name}
    id: ${deviceName}_${output2Name}
    default_transition_length: 600s
    output: LED_gpio_17
output:
   - platform: ledc
     pin: 17
     id: LED_gpio_17
     frequency: 5000Hz
`

brightnes is set on the light through the home assistent ui slider for instance, or within a scenaria

brightness is read thrugh the light attribute in hme assistant

nagyrobi, 2022-01-26:
Try something similar: https://community.home-assistant.io/t/esphome-reading-incrementing-output-value/104086/6

So you'd increment step by step manually the output level of your ledc. You can monitor this with a number sensor.

trantoriana, 2022-01-26:
I guess what you are proposing is to set the brightness myself, or to use a counter mimicing current increasing...
Sounds a bit far fetched to me.

I found however, in issue 343, a lambda function using ""remote_values.as_brightness" function.

I created three text sensors like:

`text_sensor:`
`  - platform: template`
`    name: quinledquad_1_red_brightness`
`    lambda: |-`
`             float brightness;`
`             char  brt[20];`
`             id(quinledquad_1_red).remote_values.as_brightness(&brightness);`
`             gcvt(brightness*255, 3, brt);`
`             return{brt};`
`    update_interval: 20s`

which gives me sensors with values...

Let's see when I adjust the brightness whether this will give me the correct stepping down values.

Will report later.

trantoriana, 2022-01-26:
Correction, the right line to use to get the current value is:

'id(quinledquad_1_red).current_values.as_brightness(&brightness);'

will keep an eye and report later.

trantoriana, 2022-01-26:
Absolutely wonderful sigmoid curve:

[image]

Will post a complete picture tomorrow to show the differences..
Works like a charm.

Thanks @nagyrobi for trying to help me, highly appreciate the input.

trantoriana, 2022-01-27:
Way better than before ;-)

[image]