esphome/issues #2980: Watchdog crashing device with custom cover

Erman2, 2022-01-25:
### The problem

I have built a windows opener using custom cover component. But since ESPHome 2021.12, (right now I am using 2022.1 and it is still not working) the device always crashes before it can close/open the window. The opening/closing sequence takes approximately 10 seconds. In that time a while loop runs that measures motor current and window endstops.

This a dummy code that represents my situation. 
`#include "esphome.h"

class WindowOpener : public Component, public Cover
{
public:
    void setup() override
    {
        // Setup pin modes
    }

    CoverTraits get_traits() override
    {
        auto traits = CoverTraits();
        traits.set_is_assumed_state(false);
        traits.set_supports_position(false);
        traits.set_supports_tilt(false);
        return traits;
    }
    
    void control(const CoverCall &call) override
    {
        // This will be called every time the user requests a state change.
        if (call.get_position().has_value())
        {
            float pos = *call.get_position();

            // Simulate the time that it takes to open/close the window
            long start = millis();
            for (;;)
            {
                if (millis() - start > 10000)
                {
                    break;
                }
            }

            this->position = pos;
            this->publish_state();
        }
        if (call.get_stop())
        {
            // User requested cover stop
        }
    }
};`

### Which version of ESPHome has the issue?

2022.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.12

### What platform are you using?

ESP32

### Board

Custom PCB

### Component causing the issue

Custom cover

### Example YAML snippet

```yaml
cover:
- platform: custom
  lambda: |-
    auto window_opener = new WindowOpener();
    App.register_component(window_opener);
    return {window_opener};

  covers:
    name: "Bedroom window"
    device_class: window
```

martgras, 2022-01-26:
Probably because of  https://github.com/esphome/esphome/pull/2846

This should work :
```c++
App.feed_wdt();
yield();
```

Erman2, 2022-01-27:
Thank you very much this has solved my issue. However I am now running into another issue. When the window is opening/closing, the device disconnects from HomeAssistant and from ESPHome logs console and a few seconds. Is there any way to solve that too?

martgras, 2022-01-27:
The tight loop for 10 seconds isn't a good idea.  Since esphome is single threaded you are basically blocking all other components. 
yield() and feedwdt just prevent the worst effects.

```C++
// This will be called every time the user requests a state change.
        if (call.get_position().has_value()) {
            float pos = *call.get_position();

            // Simulate the time that it takes to open/close the window
            this->set_timeout("COVER_OPEN_CLOSE", 10000, [this, pos]() {
                this->position = pos;
                this->publish_state();
            });
        }
```

However for general questions the discord chat is a better place  https://discord.gg/KhAMKrd

Erman2, 2022-01-27:
Okay, thank  you again.

closed 2022-01-27 by Erman2