esphome/issues #2981: Internal buffer full message

seblang, 2022-01-25:
### The problem

I a trying to use teleinfo ESP.
All is going weel but when I start my module i have a message in logs "[teleinfo:060]: Internal buffer full" and no information are sended to HA

### Which version of ESPHome has the issue?

ESPHome version 2022.1.2 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2021.12.10

### What platform are you using?

ESP8266

### Board

esp_01s

### Example YAML snippet

```yaml
esphome:
  name: teleinfokit
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "XXXXXXXX"
  password: "XXXXXXXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "TeleInfoKit"
    password: !secret ap_key

captive_portal:

# Enable logging
logger:     
  baud_rate: 0   # disable logging via UART, help to avoid numerous crash with ESP_LOGD
  level: INFO   # INFO for less log, put DEBUG to view all the linky's "étiquettes" received  in the logs
  esp8266_store_log_strings_in_flash: False     #  :doc:`recommanded for ESP8266 </components/sensor/custom>`

# Enable Home Assistant API
api:
  password: XXXXXXXXXX

ota:
  password: !secret ota_pass
  id: my_ota

uart:
  id: uart_bus
  rx_pin: GPIO3
  baud_rate: 9600
  parity: EVEN
  data_bits: 7

teleinfo:
  id: myteleinfo
  update_interval: 30s
  historical_mode: false

sensor:
  - platform: teleinfo
    tag_name: "HCHC"
    id: hchc
    name: "Index Heures Creuses"
    unit_of_measurement: "kWh"
    accuracy_decimals: 3
    icon: mdi:flash
    device_class: energy
    state_class: total_increasing
    teleinfo_id: myteleinfo
    filters:
      - multiply: 0.001
  - platform: teleinfo
    tag_name: "HCHP"
    id: hchp
    name: "Index Heures Pleines"
    unit_of_measurement: "kWh"
    accuracy_decimals: 3
    icon: mdi:flash
    device_class: energy
    state_class: total_increasing
    teleinfo_id: myteleinfo
    filters:
      - multiply: 0.001
  - platform: teleinfo
    tag_name: "PAPP"
    id: papp
    name: "Puissance Actuelle"
    unit_of_measurement: "VA"
    state_class: "measurement"
    icon: mdi:flash
    teleinfo_id: myteleinfo
  - platform: teleinfo
    tag_name: "IINST"
    id: iinst
    name: "Intensité Actuelle"
    unit_of_measurement: "A"
    device_class: current
    state_class: "measurement"
    icon: mdi:flash
    teleinfo_id: myteleinfo
  - platform: teleinfo
    tag_name: "EAIT"
    id: EAIT
    name: "Energie active injectée totale"
    unit_of_measurement: "Wh"
    state_class: "measurement"
    icon: mdi:flash
    teleinfo_id: myteleinfo
  - platform: teleinfo
    tag_name: "EAST"
    id: EAST
    name: "Energie active soutirée totale"
    unit_of_measurement: "Wh"
    state_class: "measurement"
    icon: mdi:flash
    teleinfo_id: myteleinfo
  - platform: teleinfo
    tag_name: "SINSTS"
    id: SINSTS
    name: "Puissance app. Instantanée soutirée"
    unit_of_measurement: "W"
    device_class: current
    state_class: "measurement"
    icon: mdi:flash
    teleinfo_id: myteleinfo  
  - platform: teleinfo
    tag_name: "SINSTI"
    id: SINSTI
    name: "Puissance app. Instantanée Injecté"
    unit_of_measurement: "W"
    device_class: current
    state_class: "measurement"
    icon: mdi:flash
    teleinfo_id: myteleinfo
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/teleinfo.yaml...
INFO Starting log output from teleinfokit.local using esphome API
INFO Successfully connected to teleinfokit.local
[20:23:08][I][app:102]: ESPHome version 2022.1.2 compiled on Jan 25 2022, 20:12:19
[20:23:32][W][teleinfo:060]: Internal buffer full
```

0hax, 2022-01-28:
Hi,
Thanks for reporting the issue. Does it happen with earlier versions?
Like `2021.10.3` for example.

gbrd, 2024-01-02:
I have this problem with esp32 not with esp8266 (standard mode)

0hax, 2024-01-11:
Can you try to bump `MAX_BUF_SIZE` to 4096 in `esphome/components/teleinfo/teleinfo.h` ?
Also please make sure there is no garbage on the lines due to incorrect wiring. (It happens often)

gbrd, 2024-01-11:
I found 2 ideas that I didn't had time to test here : https://forum.hacf.fr/t/teleinfo-via-wifi/1077/286

- Increase uart buffer size with rx_buffer_size: 2048
- keep only one tag (just to test)
- try software serial (just to test)

gbrd, 2024-01-11:
+ link to this maybe related closed not merged  pr

https://github.com/esphome/esphome/pull/3855

0hax, 2024-01-12:
Can you share your yaml config ?

gbrd, 2024-01-12:
Here is my yaml config :
It work perfectly with my wemos d1 mini, not with a wemos esp32
```
esphome:
  name: 22-teleinfo
  friendly_name: Teleinfo

esp8266:
  board: d1_mini

# esp32:
#   board: esp32dev
#   framework:
#     type: arduino

# Enable logging
logger:
  #hardware_uart: UART1
  baud_rate: 0
  level: DEBUG
  #esp8266_store_log_strings_in_flash: False

# Enable Home Assistant API
api:
  encryption:
    key: "xxxx"

ota:
  password: "yyy"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.1.172
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.254

uart:
  id: uart_0
  #tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 9600 
  parity: EVEN
  data_bits: 7
  stop_bits: 1
  #debug: 
  #  direction: RX
  #  dummy_receiver: false
  #  after:
  #    delimiter: "\n"
  #  sequence:
  #    - lambda: UARTDebug::log_string(direction, bytes);
  

teleinfo:
  id: myteleinfo
  uart_id: uart_0
  update_interval: 10s
  historical_mode: false 
  
sensor:
  - platform: teleinfo #Heures Creuses Jours Bleus
    id: teleinfo_hc_blue
    tag_name: "EASF01"
    name: "HC bleu"
    unit_of_measurement: "Wh"
    icon: mdi:flash
[... 222 lines cut ...]
  - platform: teleinfo
    id: teleinfo_NJOURF
    tag_name: "NJOURF"
    name: "Numéro du jour en cours"
    teleinfo_id: myteleinfo
    
  - platform: teleinfo
    id: teleinfo_NJOURFp1
    tag_name: "NJOURF+1"
    name: "Numéro du prochain jour"
    teleinfo_id: myteleinfo
    
  - platform: teleinfo
    id: teleinfo_NGTF
    tag_name: "NGTF"
    name: "calendrier tarifaire"
    teleinfo_id: myteleinfo

  - platform: teleinfo
    id: teleinfo_LTARF
    tag_name: "LTARF"
    name: "tarif en cours"
    teleinfo_id: myteleinfo

```

0hax, 2024-01-12:
> Here is my yaml config : It work perfectly with my wemos d1 mini, not with a wemos esp32
>
> ```
> esphome:
>   name: 22-teleinfo
>   friendly_name: Teleinfo
>
> esp8266:
>   board: d1_mini
>
>
> # esp32:
> #   board: [...]
What is the UART config you are using when running it with esp32?
I have decoded the debug dump you pasted in a previous post and it all looks garbage. It's like the UART is not correctly configured.

0hax, 2024-01-12:
```
esphome:
  name: linky

esp8266:
  board: nodemcuv2
```

This is the esp8266 config which works right?

gbrd, 2024-01-12:
> What is the UART config you are using when running it with esp32? 
I think it's just the same, I will have to retry to confirm

> I have decoded the debug dump you pasted in a previous post and it all looks garbage. It's like the UART is not correctly configured.
Warning, it is my not my debug dump but the one of @vytdv  
It seems we both have the same problem (or a similar one), I'm just trying to give info to help.

0hax, 2024-01-12:
> Warning, it is my not my debug dump but the one of @vytdv
> It seems we both have the same problem (or a similar one), I'm just trying to give info to help.
Ah sorry for the confusion.
To debug further, it would be great if one of you can rebuild esphome with a different `MAX_BUF_SIZE`
To do so, you need to clone https://github.com/esphome/esphome
Checkout the most recent tag, edit `esphome/components/teleinfo/teleinfo.h` to change `MAX_BUF_SIZE`
And then rebuild esphome using `pip install .` for example. Better if you do it in a virtualenv.
Once done, you can reflash your board.

I have 2 setups here, one in historical mode and one with standard mode.
Both works since years but I may have less data in the standard mode one as it's for solar panel.

gbrd, 2024-01-12:
I did that with MAX_BUF_SIZE => same but.
Other note : I had to use another io (IO32), because I was not able to receive any byte on IO3 (RX)

gbrd, 2024-01-12:
I added some logs. The code is just waiting for char 0x03 (frame separator) and never finds it. 
Maybe it's a hardware problem (my optocoupler/resistors stuff  would work with esp8266 but not with esp32 ?!)

gbrd, 2024-01-12:
Ok. I just enabled logs on my esp8266 to compare and I understood what you said about garbage :)
I did not realised that I used log_string (and not log hex) and I shoud see readable text... 
Thus 2 possible causes I think :
- misconfiguration on uart as you said (but I don't understand what is specific to esp32 ?)
- hardware (resistor values ?) that does not interract well with esp32 

I should try to build another "adapter" using a LTV814 (I will read again https://hallard.me/utinfo/ ... )

gbrd, 2024-01-13:
@vytdv @0hax what kind of electronic adapter do you use ? (with esp32 standard mode on linky) ? What optocoupler, resistors values, schematic?

gbrd, 2024-01-13:
This one should be the latest (with 220 ohm resistor) I built it, I will test

gbrd, 2024-01-13:
It works !... but I'have a lot of "bad crc" messages like this : 
```
[E][teleinfo:038]: bad crc: got XX except YY
```

0hax, 2024-01-13:
> @vytdv @0hax what kind of electronic adapter do you use ? (with esp32 standard mode on linky) ? What optocoupler, resistors values, schematic?

You can see exactly what I use here https://0hax.github.io/post/teleinformation/ (and in french :))

gbrd, 2024-01-13:
> I've just tried with a lolin_c3_mini, the Wemos TIC module and the MAX_BUF_SIZE at 4096, no difference, still the same error.
> @gbrd, how did you get it to work?

@vytdv you should check the value of the resistor between the linky and the optocoupler. Mine is 220. @0hax .. 0 :)

 I understood that it should be 4.7k (?) to be enedis compliant, hallard made a big work to build a compliant module but finally reduced the value to 220 ohm in the very last version...  and @0hax has none. I think the only drawback is it eats all the signal and you could not add something else on TIC plug.

gbrd, 2024-01-14:
> After checking, I do have the latest version of hallard's WeMos-TIC module (the 1.1) with the 220 ohm resistor, but it doesn't seem to be more functional.

What do you with log string debug on uart ?
```
  debug: 
    direction: RX
    dummy_receiver: false
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
 ```

You could try another gpio, short circuit the 220 resistor, or using a wemos D1 mini on your WeMos tic (because on my side it worked better with D1 mini, your lolin is esp32 and it did not work for me Rx pin of esp32)... but not sure if it's the same problem

Dkhil, 2024-10-07:
i had the same problem, it was an error in the schematic a restor of 3.3k ohm instead of 10kohm should placed between GND and LTV814 and mosfet
https://forum.hacf.fr/uploads/default/original/3X/4/5/45f1f6188992092d2deb3d372719fc1649cebbe7.png
wrong shematic for standard mode 
[image]

robin-thoni, 2025-04-18:
> i had the same problem, it was an error in the schematic a restor of 3.3k ohm instead of 10kohm should placed between GND and LTV814 and mosfet schematic for standard mode https://forum.hacf.fr/uploads/default/original/3X/4/5/45f1f6188992092d2deb3d372719fc1649cebbe7.png shematic for history mode [image]

That also fixed my issue when switching to the `standard` mode. 220 and 3.3k Ohm, instead of 1k and 10k, as mentioned [here](https://www.hacf.fr/esphome-cartes-energy-teleinformation/).

I still have some `[E][teleinfo:038]: bad crc: got 70 except 52` errors, but it was already the case in `historic` mode.

Maybe I'll try to play w/ different resistor values to see if I can get something better 🤷‍♂

[image]

[image]