esphome/issues #2985: M5Stack STAMP-C3 Flashing problem

legowarrior, 2022-01-26:
### The problem

I try to flash the M5Stack STAMP-C3 via the ESPHOME-flasher.
The flashing looks good, but after there is an only error on serial output: [14:27:03]invalid header: 0xffffffff
[image]

### Which version of ESPHome has the issue?

2022.1.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.10

### What platform are you using?

ESP32

### Board

M5Stack STAMP-C3 ESP32-C3 Rev 3

### Example YAML snippet

```yaml
esphome:
  name: testing
  platformio_options:
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
    platform:
      - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

logger:
  level: DEBUG

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

binary_sensor:
  - platform: gpio
    pin: GPIO3
    name: "M5 Button"

light:
  - platform: neopixelbus
    id: m5led
    type: GRB
    variant: SK6812
    method: ESP32_RMT_1
    num_leds: 1
    pin: GPIO2
    name: "M5 Led"

web_server:
```

### Anything in the logs that might be useful for us?

```txt
Using 'COM16' as serial port.
Connecting....
Detecting chip type... ESP32-C3
Connecting....

Chip I(96 %)Writing at 0x000cfd1f... (100 %)Wrote 798464 bytes (482880 compressed) at 0x00010000 in 13.9 seconds (effective 459.9 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[14:27:03]fff
[14:27:03]invalid header: 0xffffffff
[14:27:03]invalid header: 0xffffffff
[14:27:03]invalid header: 0xffffffff
[14:27:03]invalid header: 0xffffffff
```

### Additional information

I've tried 2 boards, and 2 USB cables

legowarrior, 2022-01-26:
update: i tried the command line esptool for flashing.
the parameters: esptool --chip esp32c3 --port COM17 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m 0x0 teszt4.bin
there are still errors on serial output after reboot:
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380000
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c0a0020,len:0x178c8
load:0x3fc8e000,len:0x3220
load:0x40380000,len:0x5500
load:0x42000020,len:0x9a528
load:0x40385500,len:0x8980
load:0x50000000,len:0x10
SHA-256 comparison failed:
Calculated: edcff9835de1bae4da3a8221271202f7151fcb0ab5f197f9f8bdb2a787b49b64
Expected: 209d916f1c2d6e3d42d129876c4417c902a74c9e8e⸮ESP-ROM:esp32c3-api1-20210207

randybb, 2022-01-26:
If I remember correctly the new images are not compatible with esphome-flasher and you have to use https://web.esphome.io (+ there are some other methods, but this will be easiest for you).

legowarrior, 2022-01-27:
> If I remember correctly the new images are not compatible with esphome-flasher and you have to use https://web.esphome.io via Google Chrome/Chromium or MS Edge (+ there are some other methods, but this will be easiest for you).

thank you! it's working fine!

closed 2022-01-27 by legowarrior