esphome/issues #2987: Error updating NodeMCU

andrelunardelli, 2022-01-27:
### The problem

Hello!
First of all, I am completely new and have no experience with terminals SUDOS, BASH, ˜, docker (the unfriendly whale)...etc.
All I am trying is to create a ESPHome DIY device for pushbuttons, all of a sudden it just blocked my update attempts.
Has anyone been through this issue trying to UPDATE or EDIT a NodeMCU ESP8266, either wirellessly or connected to the usb???

INFO Reading configuration /config/esphome/buttonand.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 825, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 803, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
  File "/esphome/esphome/config.py", line 974, in read_config
    res = load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 829, in load_config
    return _load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 817, in _load_config
    result = validate_config(config, command_line_substitutions)
  File "/esphome/esphome/config.py", line 744, in validate_config
    result.run_validation_steps()
  File "/esphome/esphome/config.py", line 124, in run_validation_steps
    task.step.run(self)
  File "/esphome/esphome/config.py", line 297, in run
    component = get_component(self.domain)
  File "/esphome/esphome/loader.py", line 177, in get_component
    assert "." not in domain
AssertionError

Please if it envolves terminal help me by telling which and what do I have to type... pleaseeeee!
Thanks a lot!!!

### Which version of ESPHome has the issue?

2022.1.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.10

### What platform are you using?

ESP8266

### Board

NODEMCU

### Component causing the issue

buttons

### Example YAML snippet

```yaml
esphome:
  name: buttonode

esp8266:
  board: nodemcuv2

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "bda0688b29aee8d21aa69fce6c3b7253"

wifi:
  ssid: "VICTORIA"
  password: "naotemsenha"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Buttonode Fallback Hotspot"
    password: "tlsf7L6ddGFv"

captive_portal:

binary.switch:
  - platform: gpio
    pin:
      number: D7
      mode: INPUT_PULLUP
      inverted: true
    name: "Green switch"
    filters:
      - delayed_on: 10ms
      - delayed_off: 10ms
      
  - platform: gpio
    pin:
      number: D5
      mode: INPUT_PULLUP
      inverted: true
    name: "Red switch"
    filters:
      - delayed_on: 10ms
      - delayed_off: 10ms
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/buttonand.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 825, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 803, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
  File "/esphome/esphome/config.py", line 974, in read_config
    res = load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 829, in load_config
    return _load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 817, in _load_config
    result = validate_config(config, command_line_substitutions)
  File "/esphome/esphome/config.py", line 744, in validate_config
    result.run_validation_steps()
  File "/esphome/esphome/config.py", line 124, in run_validation_steps
    task.step.run(self)
  File "/esphome/esphome/config.py", line 297, in run
    component = get_component(self.domain)
  File "/esphome/esphome/loader.py", line 177, in get_component
    assert "." not in domain
AssertionError
```

### Additional information

No dev experience, or terminal, or docker...

randybb, 2022-01-27:
Compiler doesn't like your "binary.switch" as there is no such component. Instead of
```yaml
binary.switch:
```
use
```yaml
binary_sensor:
```

andrelunardelli, 2022-01-27:
Geez… I messed up so much with the yaml using switches that I just used the wrong component.
Thankssss a bunch!!! They gotta write a Hass for Dummies!!! lol

closed 2022-01-27 by andrelunardelli