esphome/issues #2988: Servo control with pca9685 as output

kobold81, 2022-01-27:
### The problem

From reading the description of the servo component, I think controlling servos with the pca9685 channels as outputs seems possible. Connection to the pca9685 works as I also control some lights with it, but moving sliders in home assistant does not make the servos move.

Did I configure something wrong or is the servo compoent not intended to work with pca9685 as an output?

### Which version of ESPHome has the issue?

2022.1.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.12.10

### What platform are you using?

ESP32

### Board

esp32cam

### Component causing the issue

pca9685, servo

### Example YAML snippet

```yaml
# Enable Home Assistant API
api:
  password: ""
  services:
    - service: control_servo_1
      variables:
        level: float
      then:
        - servo.write:
            id: my_servo_1
            level: !lambda 'return level / 100.0;'
    
    - service: control_servo_2
      variables:
        level: float
      then:
        - servo.write:
            id: my_servo_2
            level: !lambda 'return level / 100.0;'

i2c:
  sda: 12
  scl: 13
  scan: true
  id: bus_a

pca9685:
  id: pca9685_1
  address: 0x40
  frequency: 60 #500

output:
  - platform: pca9685 #Ch5 - servo 1
    id: 'pca_channel5'
    channel: 5
  - platform: pca9685  #Ch6 - servo 2
    id: 'pca_channel6'
    channel: 6

servo:
  - id: my_servo_1
    output: pca_channel5

  - id: my_servo_2
    output: pca_channel6

Home Assistant:
automation.yaml:

#automation:
  - alias: Write Servo Value to ESP
    trigger:
      platform: state
      entity_id: input_number.servo_control1
    action:
      # Replace livingroom with the name you gave the ESP
      - service: esphome.esp32camera_control_servo1
        data_template:
          level: '{{ trigger.to_state.state | int }}'
          
  - alias: Write Servo2 Value to ESP
    trigger:
      platform: state
      entity_id: input_number.servo_control2
    action:
      # Replace livingroom with the name you gave the ESP
      - service: esphome.esp32camera_control_servo2
        data_template:
          level: '{{ trigger.to_state.state | int }}'

configuration.yaml:

input_number:
  servo_control1:
    name: Servo Control1
    initial: 0
    min: -100
    max: 100
    step: 1
    mode: slider
    
    
  servo_control2:
    name: Servo Control2
    initial: 0
    min: -100
    max: 100
    step: 1
    mode: slider
```

randybb, 2022-01-27:
Since you have `id:` under your `pca9685:`, adding `pca9685_id: 'pca9685_1'` to your output definition will make any difference?
```yaml
output:
  - platform: pca9685 #Ch5 - servo 1
    id: 'pca_channel5'
    pca9685_id: 'pca9685_1'
    channel: 5
  - platform: pca9685  #Ch6 - servo 2
    id: 'pca_channel6'
    pca9685_id: 'pca9685_1'
    channel: 6
```

kobold81, 2022-01-27:
I added pca9685_id: 'pca9685_1' to channel 5 &6, but no difference. I have attachen channels 1-4 to switch component and it works even without the "pca9685_id: 'pca9685_1'" added to each output.

Giancky79, 2022-03-31:
I’m thinking about that 3V3/5V you mentioned. In your case I would use 4k7 resistors (ussual value for single I2C master-slave config as what I’ve seen) and I would tie them to 5V. In case you have them on 3V3 it may not have enough voltage for the PCA9685 to recognize the logic “1”. Just checked datasheet and it indeed states that logic HIGH is 0.7*Vcc (as ussual), which is 3.5V if you power the PCA9685 from 5V supply.

maybe the solution is here?

Giancky79, 2022-03-31:
[image]

randybb, 2022-03-31:
Based on the datasheet Vdd can be 2.5, 3.3 and 5.0V + pull-up resistors are not needed as these are already on the board (check 10k resistors near OE and SCL pins). And never had any issue with 3.3V pullups on i2c while IC is powered from 5V rail.

Giancky79, 2022-03-31:
ok, you think if i put on 5v to vcc, we resolve the problem with servo on pca9685?

Giancky79, 2022-03-31:
my configuration is similar to @kobold81, but i try with 5v on vcc........ worked, because the 10k resistors near oe and scl/sda, eureka

BerkshireBugsy, 2022-12-24:
> my configuration is similar to @kobold81, but i try with 5v on vcc........ worked, because the 10k resistors near oe and scl/sda, eureka

Hi there...just come across this thread as I am trying to get servos working via ESPHome, PCA9685 and ESP32...but nothing - it seems dead. Did you get it working in the end?

TIA