esphome/issues #2991: ESPHome UART code for Daly BMS not working on ESP32

derekisastro, 2022-01-27:
### The problem

ESPHome UART code for Daly BMS not working on ESP32. Using the code from the ESPHome site. It seems to connect correctly and the sensors have been created but produce no data for the sensors.
I am sure the Rx-Tx connections are correct.

### Which version of ESPHome has the issue?

2022.1.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.10

### What platform are you using?

ESP32

### Board

ESP32 ESP-32S NodeMCU-32S ESP-WROOM-32

### Component causing the issue

ESP32

### Example YAML snippet

```yaml
uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 9600

daly_bms:
  update_interval: 20s

sensor:
  - platform: daly_bms
    voltage:
      name: "Battery Voltage"
    current:
      name: "Battery Current"
    battery_level:
      name: "Battery Level"
    max_cell_voltage:
      name: "Max Cell Voltage"
    max_cell_voltage_number:
      name: "Max Cell Voltage Number"
    min_cell_voltage:
      name: "Min Cell Voltage"
    min_cell_voltage_number:
      name: "Min Cell Voltage Number"
    max_temperature:
      name: "Max Temperature"
    max_temperature_probe_number:
      name: "Max Temperature Probe Number"
    min_temperature:
      name: "Min Temperature"
    min_temperature_probe_number:
      name: "Min Temperature Probe Number"
    remaining_capacity:
      name: "Remaining Capacity"
    cells_number:
      name: "Cells Number"
    temperature_1:
      name: "Temperature 1"
    temperature_2:
      name: "Temperature 2"

text_sensor:
  - platform: daly_bms
    status:
      name: "BMS Status"

binary_sensor:
  - platform: daly_bms
    charging_mos_enabled:
      name: "Charging MOS"
    discharging_mos_enabled:
      name: "Discharging MOS"
```

### Anything in the logs that might be useful for us?

```txt
[09:33:26][C][wifi:355]:   Hostname: 'solar-bms'
[09:33:26][C][wifi:357]:   Signal strength: -60 dB ▂▄▆█
[09:33:26][C][wifi:361]:   Channel: 10
[09:33:26][C][logger:234]:   Level: DEBUG
[09:33:26][C][logger:235]:   Log Baud Rate: 0
[09:33:26][C][logger:236]:   Hardware UART: UART0
[09:33:26][C][uart.arduino_esp32:105]: UART Bus:
[09:33:26][C][uart.arduino_esp32:106]:   TX Pin: GPIO1
[09:33:26][C][uart.arduino_esp32:107]:   RX Pin: GPIO3
[09:33:26][C][uart.arduino_esp32:109]:   RX Buffer Size: 256
[09:33:26][C][uart.arduino_esp32:111]:   Baud Rate: 9600 baud
[09:33:26][C][uart.arduino_esp32:112]:   Data Bits: 8
[09:33:26][C][uart.arduino_esp32:113]:   Parity: NONE
[09:33:26][C][uart.arduino_esp32:114]:   Stop bits: 1
[09:33:26][C][daly_bms:024]: Daly BMS:
```

### Additional information

Sensors
Array 1
109.52 Amps
Array 2
33.95 Amps
Array 3
30.40 Amps
Array 4
15.00 Amps
BMS Status
Unknown
Battery Current
Unknown
Battery Level
Unknown
Battery Voltage
Unknown
Cells Number
Unknown
Charging MOS
Off
Discharging MOS
Off
Max Cell Voltage
Unknown
Max Cell Voltage Number
Unknown
Max Temperature
Unknown
Max Temperature Probe Number
Unknown
Min Cell Voltage
Unknown
Min Cell Voltage Number
Unknown
Min Temperature
Unknown
Min Temperature Probe Number
Unknown
Remaining Capacity
Unknown
Temperature 1
Unknown
Temperature 2
Unknown
Automations
No Automations have been added using this device yet. You can add one by clicking the + button above.
Scenes
No Scenes have been added using this device yet. You can add one by clicking the + button above.
Scripts
No Scripts have been added using this device yet. You can add one by clicking the + button above.

moesizlag, 2022-02-18:
Exact same problem. Using a M5 Stack Atom Lite, connects perfectly fine to HA but no data from the BMS gets read.

moesizlag, 2022-02-25:
I was able to get it working using an ESP8266 and thx to the guidance here: https://community.home-assistant.io/t/esphome-daly-bms-using-uart-guide/394429
No luck still with the M5Stack Lite or an ESP32. Latter one did kind of work but missed a few sensors.

insipiens, 2022-04-02:
It worked for me on an ESP32, the only difference is I made GPIO21 and GPIO22 the uart connections physically and in the UART: section

```
uart:
  tx_pin: GPIO22
  rx_pin: GPIO21
  baud_rate: 9600
```