esphome/issues #2995: Impossible to set api_key in secrets.yaml

raidnet-ms, 2022-01-28:
### The problem

Every password can be substituted with !secret, except the api_key

### Which version of ESPHome has the issue?

2022.1.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.10

### What platform are you using?

ESP32

### Board

wemos battery

### Component causing the issue

api key

### Example YAML snippet

```yaml
api:
  password: !secret api_password
  encryption:
    key: !secret apy_key
```

OttoWinter, 2022-01-28:
1. There's a spelling error in the yaml, it should be ap**i**_key
2. Please post the error you're getting

raidnet-ms, 2022-01-28:
Sorry Otto, was a long day on esphome :-) the error was only here, in the yaml is correct, I've checked again of course.

What I get is that the API is not loggin in

raidnet-ms, 2022-01-28:
In the meanwhile removing and adding the device again I found this other issue:

https://ibb.co/bszXyTW

raidnet-ms, 2022-01-29:
Ok there's something more weird happening here.

With the api key I keep disconnecting from api. With verbose log I noticed this after 2 minutes:

`INFO majordomo.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for majordomo.local
WARNING Disconnected from API
INFO majordomo.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to majordomo.local`

If i restart esphome integration, the device connects again.

If I disable the apy key, none of this happens.
I've tryied to remove and add the device back many times, but this happens over and over.

closed 2022-01-30 by raidnet-ms