esphome/issues #2997: Growatt modbus sensor - incorrect data reading

slawekzz, 2022-01-29:
### The problem

The sensor works incorrectly with the new Growatt inverters. From the model 10KTL3-X it only reads the following values:
  Growatt PV Active Power
  Growatt PV1 Active Power
  Growatt PV1 Current
  Growatt PV1 Voltage
  Growatt PV2 Active Power
  Growatt PV2 Current
  Growatt PV2 Voltage
  Growatt Status Code 

### Which version of ESPHome has the issue?

2022.1.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.10

### What platform are you using?

ESP8266

slawekzz, 2022-01-30:
This is some mistake.  I'm not a developer.  I'm just the user.  if I can help you somehow, please write.  for example, if you send me the appropriate instructions, I can read the communication of my inverter.  it is possible that they are under other registers.

theoxhead, 2022-02-03:
Hello, I am experience the same issues with the Growatt 7000TL3-X, I am using the ESP32 module and ESPHome version 2022.1.3

Skevdude, 2022-02-10:
I also have the before mentioned problem. My inverter is a Growatt 8000TL3-X.
I'm not good with code, but if there's a way I can help, let me know.

djwlindenaar, 2022-03-16:
Same problem here, using a MIC-3300TL-X. Seems like [this ](http://www.photovoltaicsolar.in/Growatt_Manual/MAX%20Series%20Modbus%20RTU%20Protocol.pdf)is the right specification for these inverters. I'll have a look if I can find time to write a pull request for it. I don't have a clue whether it is easy to detect automatically which version is used, so I guess we need to add a configuration item to specify RTU-I or RTU-II

djwlindenaar, 2022-03-16:
Update: got it to work by changing the registers read in line with the data in the PDF for "RTU-II". I've now just hacked that into the existing code, so that's not a good way to be submitted as a patch, but it does prove that this is actually going to solve it. I'll see if I can find some time to make a nice patch over the weekend.

djwlindenaar, 2022-03-20:
@slawekzz , @theoxhead , @Skevdude , I've submitted a pull request as can be seen above. 

In the meantime, you could test this by checking the growatt_solar code from my branch [djwlindenaar/esphome](/djwlindenaar/esphome) . That can be done by cloning the repository locally and referring to that by adding to your device yaml:
```
external_components:
  - source: my_components
```
my_components should be replaced by the path to the cloned repository.

also you should add to the growatt_solar section:
```  
  - platform: growatt_solar
    protocol_version: RTU2
```

Skevdude, 2022-03-20:
Thanks @djwlindenaar for your efforts!

slawekzz, 2022-03-21:
Thanks for the quick change. Everything looks fine except Growatt Total Energy Production. The bus return value is less than what is shown on the inverter display. For example, from 6828 kWh is read and it should be 6832 kWh.

slawekzz, 2022-03-22:
I have a Growatt 10KTL3-X inverter. The reading from the inverter display and HA integration is the same 6865.1 kWh. Reading from modbus is different.
'Growatt Total Energy Production': Sending state 6860.50000 kWh with 0 decimals of accuracy

djwlindenaar, 2022-03-22:
nothing that makes sense from a math p.o.v. the difference is quite small, though. only 0.1%. That might explain why I cannot see it, since I've only just started using my inverter...

I'm wondering whether we should be reading these registers instead: 
1050. Etogrid _total H Energy to grid total high 0.1kWh
1051. Etogrid _ total L Energy to grid total high 0.1kWh

instead of 
55. Eac total H Total generate energy (high) 0.1kWH
56. Eac total L Total generate energy (low) 0.1kWH

djwlindenaar, 2022-03-23:
@slawekzz , could you test whether the branch I created works? First if it works at all ;) , second whether it shows the right value for you? It reads the register at 1050 and 1051 for Total Energy Production....

[growat_solar_totalissue](https://github.com/djwlindenaar/esphome/tree/growatt_solar_totalissue)

djwlindenaar, 2022-03-23:
Just tested on mine (MIC 3300TL-X), reads 0.0, so probably not the solution

djwlindenaar, 2022-03-25:
@slawekzz , could you run the code in my growatt_solar_totalissue with VERBOSE logging and copy the result of one sensor update cycle here? It logs the raw data from the first 125 registers, so hopefully, this will give some insight in whether something is going wrong...

slawekzz, 2022-03-25:
[12:02:42][V][Growatt:107]: Modbus received Byte  0 (0X0001)
[12:02:42][V][Growatt:107]: Modbus received Byte  1 (0X0001)
[12:02:42][V][Growatt:107]: Modbus received Byte  2 (0X2FED)
[12:02:42][V][Growatt:107]: Modbus received Byte  3 (0X1247)
[12:02:42][V][Growatt:107]: Modbus received Byte  4 (0X005E)
[12:02:42][V][Growatt:107]: Modbus received Byte  5 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  6 (0XABCE)
[12:02:42][V][Growatt:107]: Modbus received Byte  7 (0X1326)
[12:02:42][V][Growatt:107]: Modbus received Byte  8 (0X0045)
[12:02:42][V][Growatt:107]: Modbus received Byte  9 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  10 (0X841F)
[12:02:42][V][Growatt:107]: Modbus received Byte  11 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  12 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  13 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  14 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  15 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  16 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  17 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  18 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  19 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  20 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  21 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  22 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  23 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  24 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  25 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  26 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  27 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  28 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  29 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  30 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  31 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  32 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  33 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  34 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  35 (0X0001)
[12:02:42][V][Growatt:107]: Modbus received Byte  36 (0X280D)
[12:02:42][V][Growatt:107]: Modbus received Byte  37 (0X1388)
[12:02:42][V][Growatt:107]: Modbus received Byte  38 (0X08E9)
[12:02:42][V][Growatt:107]: Modbus received Byte  39 (0X006D)
[12:02:42][V][Growatt:107]: Modbus received Byte  40 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  41 (0X611E)
[12:02:42][V][Growatt:107]: Modbus received Byte  42 (0X094F)
[12:02:42][V][Growatt:107]: Modbus received Byte  43 (0X006D)
[12:02:42][V][Growatt:107]: Modbus received Byte  44 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  45 (0X6576)
[12:02:42][V][Growatt:107]: Modbus received Byte  46 (0X090D)
[12:02:42][V][Growatt:107]: Modbus received Byte  47 (0X006D)
[12:02:42][V][Growatt:107]: Modbus received Byte  48 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  49 (0X62A7)
[12:02:42][V][Growatt:107]: Modbus received Byte  50 (0X0FCC)
[12:02:42][V][Growatt:107]: Modbus received Byte  51 (0X0FCB)
[12:02:42][V][Growatt:107]: Modbus received Byte  52 (0X0F9E)
[12:02:42][V][Growatt:107]: Modbus received Byte  53 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  54 (0X0117)
[12:02:42][V][Growatt:107]: Modbus received Byte  55 (0X0001)
[12:02:42][V][Growatt:107]: Modbus received Byte  56 (0X136C)
[12:02:42][V][Growatt:107]: Modbus received Byte  57 (0X0143)
[12:02:42][V][Growatt:107]: Modbus received Byte  58 (0X5D32)
[12:02:42][V][Growatt:107]: Modbus received Byte  59 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  60 (0X00C3)
[12:02:42][V][Growatt:107]: Modbus received Byte  61 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  62 (0X9891)
[12:02:42][V][Growatt:107]: Modbus received Byte  63 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  64 (0X004C)
[12:02:42][V][Growatt:107]: Modbus received Byte  65 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  66 (0X7355)
[12:02:42][V][Growatt:107]: Modbus received Byte  67 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  68 (0X0000)
[12:02:42][V][Growatt:107]: Modbus received Byte  69 (0X0000)
[... 30 lines cut ...]
[12:02:43][V][Growatt:107]: Modbus received Byte  100 (0X4E20)
[12:02:43][V][Growatt:107]: Modbus received Byte  101 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  102 (0X0001)
[12:02:43][V][Growatt:107]: Modbus received Byte  103 (0XB206)
[12:02:43][V][Growatt:107]: Modbus received Byte  104 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  105 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  106 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  107 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  108 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  109 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  110 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  111 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  112 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  113 (0X0044)
[12:02:43][V][Growatt:107]: Modbus received Byte  114 (0X001E)
[12:02:43][V][Growatt:107]: Modbus received Byte  115 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  116 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  117 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  118 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  119 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  120 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  121 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  122 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  123 (0X0000)
[12:02:43][V][Growatt:107]: Modbus received Byte  124 (0X0000)

djwlindenaar, 2022-03-25:
Could you also add the lines with info on the sensors being updated (so with the calculated float values)? For checking

slawekzz, 2022-03-25:
- platform: growatt_solar
    protocol_version: RTU2
    update_interval: 10s
    address: 2
    inverter_status:
      name: "Growatt Status Code"
    phase_a:
      voltage:
          name: "Growatt Voltage Phase A"
      current:
          name: "Growatt Current Phase A"
      active_power:
          name: "Growatt Power Phase A"
    phase_b:
      voltage:
          name: "Growatt Voltage Phase B"
      current:
          name: "Growatt Current Phase B"
      active_power:
          name: "Growatt Power Phase B"
    phase_c:
      voltage:
          name: "Growatt Voltage Phase C"
      current:
          name: "Growatt Current Phase C"
      active_power:
          name: "Growatt Power Phase C"
    pv1:
      voltage:
          name: "Growatt PV1 Voltage"
      current:
          name: "Growatt PV1 Current"
      active_power:
          name: "Growatt PV1 Active Power"
    pv2:
      voltage:
          name: "Growatt PV2 Voltage"
      current:
          name: "Growatt PV2 Current"
      active_power:
          name: "Growatt PV2 Active Power"
    active_power:
      name: "Growatt Grid Active Power"
    pv_active_power:
      name: "Growatt PV Active Power"
    frequency:
      name: "Growatt Frequency"
    energy_production_day:
      name: "Growatt Today's Generation"
    total_energy_production:
      name: "Growatt Total Energy Production"
    inverter_module_temp:
      name: "Growatt Inverter Module Temp"

Johnwulp, 2022-03-25:
I also tried your update @djwlindenaar, but the only thing I get is this output and no sensor logging. Maybe the output from my inverter is somehow different? I have the **Growatt MIN 2500 TL-XH**

```
[16:05:20][V][modbus:184]: Modbus write: 01.04.00.00.00.5F.B0.32 (8)
[16:05:20][V][modbus:058]: Modbus received Byte  0 (0X0)
[16:05:20][V][modbus:058]: Modbus received Byte  127 (0X7f)
[16:05:20][V][modbus:058]: Modbus received Byte  223 (0Xdf)
[16:05:20][V][modbus:058]: Modbus received Byte  254 (0Xfe)
[16:05:20][V][modbus:058]: Modbus received Byte  253 (0Xfd)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  177 (0Xb1)
[16:05:20][V][modbus:058]: Modbus received Byte  199 (0Xc7)
[16:05:20][V][modbus:058]: Modbus received Byte  123 (0X7b)
[16:05:20][V][modbus:058]: Modbus received Byte  254 (0Xfe)
[16:05:20][V][modbus:058]: Modbus received Byte  173 (0Xad)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  177 (0Xb1)
[16:05:20][V][modbus:058]: Modbus received Byte  199 (0Xc7)
[16:05:20][V][modbus:058]: Modbus received Byte  191 (0Xbf)
[16:05:20][V][modbus:058]: Modbus received Byte  248 (0Xf8)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[16:05:20][V][modbus:058]: Modbus received Byte  255 (0Xff)
[... 181 lines cut ...]
      current:
          name: "Growatt Current Phase"
      active_power:
          name: "Growatt Power Phase"
    pv1:
      voltage:
          name: "Growatt PV1 Voltage"
      current:
          name: "Growatt PV1 Current"
      active_power:
          name: "Growatt PV1 Active Power"
    active_power:
      name: "Growatt Grid Active Power"
    pv_active_power:
      name: "Growatt PV Active Power"
    frequency:
      name: "Growatt Frequency"
    energy_production_day:
      name: "Growatt Today's Generation"
    total_energy_production:
      name: "Growatt Total Energy Production"
      accuracy_decimals: 1
    inverter_module_temp:
      name: "Growatt Inverter Module Temp"
```

djwlindenaar, 2022-03-25:
@slawekzz I'm stumped... The logs prove that the code basically works. The EacTotal register at position 55 is read and correctly calculated into the total energy produced. 

One thing I did notice is that if I add together the PV1 and PV2 total produced energy, it's a lower number than the grand total 

PV1 total = 3905.7 kWh
PV2 total = 2952.5 kWh
PV total = 6858.2 kWh
AC total = 7050.8 kWh

So I'm wondering where it got the additional 200 kWh (free energy!). It does make me wonder how accurate all of these measurements really are. Having said that, I have no explanation why the logging through the Growatt server is different from the modbus registers. Also studying the specification, I can't really see other registers that could be read for that.

Only thing I was considering is whether there is a time lag or something playing a role here. Could you check if the difference is still there at a time when only a few watts are being generated? or compare the last value of the day of modbus with that of Growatt server. 
If that still leaves a difference, I just don't know. Only thing I could imagine is that some kind of calculation is done in the Growatt server/on the display, which we simply don't know about.

djwlindenaar, 2022-03-25:
@Johnwulp , looks like the received data is inverted somehow... Could you try with the Rx pin inverted? Not sure if that will work at all, but with all the 0xff it seems that way...

Johnwulp, 2022-03-25:
> @Johnwulp , looks like the received data is inverted somehow... Could you try with the Rx pin inverted? Not sure if that will work at all, but with all the 0xff it seems that way...

When I reversed tx and rx pin I get no response at all. Going to search on the interwebs…

```
20:40:47][V][modbus:184]: Modbus write: 01.04.00.00.00.5F.B0.32 (8)
[20:40:57][V][modbus:184]: Modbus write: 01.04.00.00.00.5F.B0.32 (8)
[20:41:07][V][modbus:184]: Modbus write: 01.04.00.00.00.5F.B0.32 (8)
[20:41:17][V][modbus:184]: Modbus write: 01.04.00.00.00.5F.B0.32 (8)
```

Johnwulp, 2022-03-25:
I use pin 3 and 4 connected to rx and tx. That should work right?
[image]

Johnwulp, 2022-03-25:
Im at the point that I have `Modbus CRC Check failed! C071!=300` but I think I need logic level shifters in order to make this work.

djwlindenaar, 2022-03-25:
Maybe you should open a new issue, since this doesn't seem to relate to the original topic. 

Also please share your hardware setup as that gives us a chance to understand what you're doing.

Btw when I said invert the Rx pin, I meant the literally. Not switch Tx and Rx. 

So: 
```
uart:
  - id: uart1
    baud_rate: 9600
    tx_pin: GPIO1
    rx_pin: 
      number: GPIO3
      inverted: true
```

djwlindenaar, 2022-03-26:
> Im at the point that I have `Modbus CRC Check failed! C071!=300` but I think I need logic level shifters in order to make this work.

Ah... Did you connect the rs485 signals directly to the UART? Interesting that this gives any result at all.

@Johnwulp , you could consider buying a ShineWifi-X. It has ESP8266 inside and I simply use that with esphome. Works great, neat install. Those have a USB connector to connect to the inverter, but you can flash esphome via that connector and a PC. There's more info on that on the web.

closed 2022-04-10 by jesserockz