esphome/issues #2999: INA226 Max measured current its 0.818A

zac99, 2022-01-30:
### The problem

The configuration has a current consumption above 1.5A and the INA226 mi shows a measurement of 0.818A. I set up two modals and they have the same reading.

### Which version of ESPHome has the issue?

2022.1.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.1

### What platform are you using?

ESP32

### Board

ESP-32WROOM-32

### Component causing the issue

INA226

### Example YAML snippet

```yaml
i2c:
  sda: 21
  scl: 22
  scan: true
  id: bus_a    

  
sensor:
  - platform: ina226
    address: 0x45
    shunt_resistance: 0.1 ohm
    current:
      name: "INA226 Current"
    power:
      name: "INA226 Power"
    bus_voltage:
      name: "INA226 Bus Voltage"
    shunt_voltage:
      name: "INA226 Shunt Voltage"
    max_current: 5.0A
    update_interval: 2s

  - platform: ina226
    address: 0x40
    shunt_resistance: 0.1 ohm
    current:
      name: "INA226_2 Current"
    power:
      name: "INA226_2 Power"
    bus_voltage:
      name: "INA226_2 Bus Voltage"
    shunt_voltage:
      name: "INA226_2 Shunt Voltage"
    max_current: 5.0A
    update_interval: 2s
```

mjess, 2022-02-13:
I'm not an expert for this device but it looks like expected behavior, as the INA226 can only measure up to ~819mA.

Unfortunately I found only a german explanation but basically it says that the INA226 has a resolution of 2.5 µV per bit. The register is 16bit wide but one bit is for the sign, which leaves 15bit for the value: (2^15 -1)* 2.5µV = 81.9175 mV (max value)
The typical INA226 has 0.1Ohm shunt resistor.
81.9175mV / 0.1 Ohm = 819.175mA max current range. 
Explanation (in German): https://wolles-elektronikkiste.de/ina226

ryny24, 2023-08-29:
I'm having the same problem. I set the max current to 3.2A, but it will not read over 0.82A.   Everything under 0.82 is accurate, but my meter is showing 2A but INA226 won't go above 0.82A.

ryny24, 2023-08-29:
I had the same issue. Could not get above 0.82 A, until I removed the R100 resistor and used an external shunt. Just FYI for anyone who finds this thread...