esphome/issues #3000: [V][component:199]: Component esp32_ble_tracker took a long time for an operation (0.16 s)

MaxPeal, 2022-01-31:
### The problem

with a simple yaml
esp32_ble_tracker took allways "too" a long time for an operation

```
$ export CVER=2022.2.0-dev20220131
$ docker run --rm -v "${PWD}":/config --device=/dev/ttyUSB0 -it esphome/esphome:$CVER run --device /dev/ttyUSB0 esp32-devkitv1-ble-debug.yaml

esphome:
  name: esp32-devkitv1-ble-debug

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

# Enable logging
logger:
  level: VERY_VERBOSE
  logs:
    mqtt.component: DEBUG
    mqtt.client: ERROR

light:
  - platform: status_led
    name: "Switch state"
    pin:
      number: GPIO2

# Enable Home Assistant API
api:

ota:
  password: "pas$word"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Devkitv1-Ble-Debug"
    password: "pas$word"

esp32_ble_tracker:
```

```
$ docker run --rm -v "${PWD}":/config --device=/dev/ttyUSB0 -it esphome/esphome:$CVER logs --device /dev/ttyUSB0 esp32-devkitv1-ble-debug.yaml
...
[08:55:35][V][component:199]: Component esp32_ble_tracker took a long time for an operation (0.16 s).
[08:55:35][V][component:200]: Components should block for at most 20-30ms.
[08:55:35][VV][esp32_ble_tracker:497]: Parse Result:
[08:55:36][VV][esp32_ble_tracker:514]:   Address: 15:E1:86:98:B9:80 (RANDOM)
[08:55:36][VV][esp32_ble_tracker:516]:   RSSI: -77
[08:55:36][VV][esp32_ble_tracker:517]:   Name: ''
[08:55:36][VV][esp32_ble_tracker:531]:   Manufacturer data: 01.09.21.0A.D2.09.4E.34.E4.C0.4C.41.50.54.4F.50.2D.30.54.54.41.4C.30.31.52 (25)
[08:55:36][VV][esp32_ble_tracker:547]: Adv data: 1C.FF.06.00.01.09.21.0A.D2.09.4E.34.E4.C0.4C.41.50.54.4F.50.2D.30.54.54.41.4C.30.31.52 (29)
[08:55:36][VV][esp32_ble_tracker:497]: Parse Result:
[08:55:36][VV][esp32_ble_tracker:514]:   Address: 15:E1:86:98:B9:80 (RANDOM)
[08:55:36][VV][esp32_ble_tracker:516]:   RSSI: -72
[08:55:36][VV][esp32_ble_tracker:517]:   Name: ''
[08:55:36][VV][esp32_ble_tracker:531]:   Manufacturer data: 01.09.21.0A.D2.09.4E.34.E4.C0.4C.41.50.54.4F.50.2D.30.54.54.41.
...
```

### Which version of ESPHome has the issue?

testet with 2022.1.2 and 2022.2.0-dev20220131

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

esp32_ble_tracker

### Example YAML snippet

```yaml
esphome:
  name: esp32-devkitv1-ble-debug

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

# Enable logging
logger:
  level: VERY_VERBOSE
  logs:
    mqtt.component: DEBUG
    mqtt.client: ERROR

light:
  - platform: status_led
    name: "Switch state"
    pin:
      number: GPIO2

# Enable Home Assistant API
api:

ota:
  password: "pas$word"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Devkitv1-Ble-Debug"
    password: "pas$word"

esp32_ble_tracker:
```

### Anything in the logs that might be useful for us?

```txt
$ docker run --rm -v "${PWD}":/config --device=/dev/ttyUSB0 -it esphome/esphome:$CVER logs --device /dev/ttyUSB0 esp32-devkitv1-ble-debug.yaml
...
[08:55:35][V][component:199]: Component esp32_ble_tracker took a long time for an operation (0.16 s).
[08:55:35][V][component:200]: Components should block for at most 20-30ms.
[08:55:35][VV][esp32_ble_tracker:497]: Parse Result:
[08:55:36][VV][esp32_ble_tracker:514]:   Address: 15:E1:86:98:B9:80 (RANDOM)
[08:55:36][VV][esp32_ble_tracker:516]:   RSSI: -77
[08:55:36][VV][esp32_ble_tracker:517]:   Name: ''
[08:55:36][VV][esp32_ble_tracker:531]:   Manufacturer data: 01.09.21.0A.D2.09.4E.34.E4.C0.4C.41.50.54.4F.50.2D.30.54.54.41.4C.30.31.52 (25)
[08:55:36][VV][esp32_ble_tracker:547]: Adv data: 1C.FF.06.00.01.09.21.0A.D2.09.4E.34.E4.C0.4C.41.50.54.4F.50.2D.30.54.54.41.4C.30.31.52 (29)
[08:55:36][VV][esp32_ble_tracker:497]: Parse Result:
[08:55:36][VV][esp32_ble_tracker:514]:   Address: 15:E1:86:98:B9:80 (RANDOM)
[08:55:36][VV][esp32_ble_tracker:516]:   RSSI: -72
[08:55:36][VV][esp32_ble_tracker:517]:   Name: ''
[08:55:36][VV][esp32_ble_tracker:531]:   Manufacturer data: 01.09.21.0A.D2.09.4E.34.E4.C0.4C.41.50.54.4F.50.2D.30.54.54.41.
...
```

OttoWinter, 2022-02-05:
Yeah that's during the BlE controller setup, which can take a bit. It only happens once so it's not a real problem. It's intentionally a verbose message because in some cases it's hard to remove blocking operations and if it only happens once it's not a problem

OttoWinter, 2022-02-05:
Closing as not a problem

closed 2022-02-05 by OttoWinter