esphome/issues #3002: FastLED appears to include delay when light is static or off

JumpMaster, 2022-01-31:
### The problem

Hi,

When adding multiple FastLED lights there is a noticeable delay inserted per light.  For example.  If you add 7 FastLED lights the ESP32 will no longer connect to Home Assistant.  At six it responds but an effect such as addressable_scan is very laggy.  Initially it was suspected that the ESP32 was overloaded.  But the more of the six lights that are set to do constant updates the better they all perform. ie. Setting all six lights to addressable_scan results in all lights performing well.  With my dangerous level of knowledge it appears there is a delay in the code when a FastLED light is either a static color or off.

Thanks for your help!

### Which version of ESPHome has the issue?

2022.1.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.2.0b4

### What platform are you using?

ESP32

### Board

Adafruit Huzzah32 (ESP-WROOM-32 module)

### Component causing the issue

FastLED Light

### Example YAML snippet

```yaml
esphome:
  name: test-device

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

ota:
  password: "22d15afa02eb34cd5b835e5b23f35b12"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Fallback Hotspot"
    password: "icZiqgSYRW3p"

captive_portal:

light:
  - platform: fastled_clockless
    chipset: WS2813
    rgb_order: GRB
    pin: GPIO14
    num_leds: 10
    name: "14-GPIO14"
    effects:
      - addressable_scan:
          name: Addressable Scan
          move_interval: 20ms
  - platform: fastled_clockless
    chipset: WS2813
    rgb_order: GRB
    pin: GPIO25
    num_leds: 10
    name: "A1-GPIO25"
    effects:
      - addressable_scan:
          name: Addressable Scan
          move_interval: 20ms
  - platform: fastled_clockless
    chipset: WS2813
    rgb_order: GRB
    pin: GPIO26
    num_leds: 10
    name: "A0-GPIO26"
    effects:
      - addressable_scan:
          name: Addressable Scan
          move_interval: 20ms
  - platform: fastled_clockless
    chipset: WS2813
    rgb_order: GRB
    pin: GPIO32
    num_leds: 10
    name: "32-GPIO32"
    effects:
      - addressable_scan:
          name: Addressable Scan
          move_interval: 20ms
  - platform: fastled_clockless
    chipset: WS2813
    pin: GPIO33
    num_leds: 10
    rgb_order: GRB
    name: "33-GPIO33"
    effects:
      - addressable_scan:
          name: Addressable Scan
          move_interval: 20ms
  - platform: fastled_clockless
    chipset: WS2813
    pin: GPIO27
    num_leds: 10
    rgb_order: GRB
    name: "27-GPIO27"
    effects:
      - addressable_scan:
          name: Addressable Scan
          move_interval: 20ms
```

### Anything in the logs that might be useful for us?

```txt
Log appear completely normal.
```

OttoWinter, 2022-01-31:
Try with the `neopixelbus` component instead, that library is better maintained at the moment.

OttoWinter, 2022-02-05:
Yeah I'm thinking we should add a banner at the top of fastled because neopixelbus appears to be working better nowadays (for other isssues too).

closed 2022-07-07 by github-actions[bot]