esphome/issues #3006: Restoring preferences from flash on ESP32

marshn, 2022-02-01:
### The problem

State of components not being restored correctly.

With three switches defined, changing the state of any one results in the state of all three being restored to the new saved state at next boot.

Appears that only one switch state is being stored in flash, and all three are restored with the same saved state.

### Which version of ESPHome has the issue?

2022.1.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

LILYGO T-Micro32 Module

### Component causing the issue

ESP32 Preferences

### Example YAML snippet

```yaml
switch:
  - platform: template
    name: ${device_name} Sw 1
    optimistic: true
    id: sw_1
    restore_state: true
  - platform: template
    name: ${device_name} SW 2
    id: sw_2
    optimistic: true
    restore_state: true
  - platform: template
    name: ${device_name} SW 3
    id: sw_3
    optimistic: true
    restore_state: true
```
```

### Anything in the logs that might be useful for us?

```txt
[11:28:20][I][logger:214]: Log initialized
[11:28:20][C][ota:458]: There have been 2 suspected unsuccessful boot attempts.
[11:28:20][D][esp32.preferences:114]: Saving preferences to flash...
[11:28:20][I][app:029]: Running through setup()...
[11:28:20][V][app:030]: Sorting components by setup priority...
[11:28:20][VV][scheduler:057]: set_interval(name='', interval=60000, offset=13597)
[11:28:20][D][template.switch:050]:   Restored state OFF
[11:28:20][D][switch:017]: 'esp-014 Sw 1' Turning OFF.
[11:28:20][D][switch:037]: 'esp-014 Sw 1': Sending state OFF
[11:28:20][D][template.switch:050]:   Restored state OFF
[11:28:20][D][switch:017]: 'esp-014 SW 2' Turning OFF.
[11:28:20][D][switch:037]: 'esp-014 SW 2': Sending state OFF
[11:28:20][D][template.switch:050]:   Restored state OFF
[11:28:20][D][switch:017]: 'esp-014 SW 3' Turning OFF.
[11:28:20][D][switch:037]: 'esp-014 SW 3': Sending state OFF
[11:28:20][VV][scheduler:057]: set_interval(name='update', interval=60000, offset=4674)
[11:28:20][VV][scheduler:057]: set_interval(name='update', interval=60000, offset=13205)
[11:28:20][VV][scheduler:057]: set_interval(name='update', interval=60000, offset=3980)
[11:28:20][C][wifi:037]: Setting up WiFi...
```

### Additional information

It appears that the 'key' generated in ESP32 ESPPreferenceObject make_preference() is identical for all three switch components. Adding some additional debugging to make_preferences() shows this in the logfile:

```
[11:33:25][C][ota:458]: There have been 3 suspected unsuccessful boot attempts.
[11:33:25][D][esp32.preferences:133]: Saving preferences to flash...
[11:33:25][D][esp32.preferences:141]: Save blob key='233825511', size 4
[11:33:25][D][esp32.preferences:142]: 04.00.00.00
[11:33:25][I][app:029]: Running through setup()...
[11:33:25][V][app:030]: Sorting components by setup priority...
[11:33:25][VV][scheduler:057]: set_interval(name='', interval=60000, offset=31)
[11:33:25][D][esp32.preferences:124]: make_preference type=c9882864, offset=5 keyval=-913823647, key='-913823647', size=1
[11:33:25][D][esp32.preferences:068]: Load blob key='-913823647', size=1
[11:33:25][D][esp32.preferences:069]: 00
[11:33:25][D][template.switch:050]:   Restored state OFF
[11:33:25][D][switch:017]: 'esp-014 Sw 1' Turning OFF.
[11:33:25][D][switch:037]: 'esp-014 Sw 1': Sending state OFF
[11:33:25][D][esp32.preferences:124]: make_preference type=c9882867, offset=6 keyval=-913823647, key='-913823647', size=1
[11:33:25][D][esp32.preferences:049]: Load pending save, size 1
[11:33:25][D][esp32.preferences:050]: 00
[11:33:25][D][template.switch:050]:   Restored state OFF
[11:33:25][D][switch:017]: 'esp-014 SW 2' Turning OFF.
[11:33:25][D][switch:037]: 'esp-014 SW 2': Sending state OFF
[11:33:25][D][esp32.preferences:124]: make_preference type=c9882866, offset=7 keyval=-913823647, key='-913823647', size=1
[11:33:25][D][esp32.preferences:049]: Load pending save, size 1
[11:33:25][D][esp32.preferences:050]: 00
[11:33:25][D][template.switch:050]:   Restored state OFF
[11:33:25][D][switch:017]: 'esp-014 SW 3' Turning OFF.
[11:33:25][D][switch:037]: 'esp-014 SW 3': Sending state OFF
```
Although the 'type' and 'length' parameters passed to make_preferences() are different for each of the three switches, they 'key' generated by XOR'ing them together is identical (-91382364). All three switch states are restored from the same flash blob.

The additional debugging is from this modified code:

```
  ESPPreferenceObject make_preference(size_t length, uint32_t type) override {
    auto *pref = new ESP32PreferenceBackend();  // NOLINT(cppcoreguidelines-owning-memory)
    pref->nvs_handle = nvs_handle;
    current_offset += length;

    uint32_t keyval = current_offset ^ type;
//    uint32_t keyval = (type << 16) + current_offset;
    char keybuf[16];

    snprintf(keybuf, sizeof(keybuf), "%d", keyval);
//    snprintf(keybuf, sizeof(keybuf), "%x", keyval);
    pref->key = keybuf;  // copied to std::string

	ESP_LOGD(TAG, "make_preference type=%08x, offset=%d keyval=%d, key='%s', size=%u", type, current_offset, keyval, pref->key.c_str(), length);

    return ESPPreferenceObject(pref);
  }
```

As a test I tried changing the line of code which creates the key. From:

```
    uint32_t keyval = current_offset ^ type;
```
to:

```
    uint32_t keyval = (type << 16) + current_offset;
```

This generates a unique key for each switch and their states are individually restored from flash. I am not aware of the broader consequences of doing this, so I have no confidence that this is the appropriate fix.

I haven't noticed this issue in more complex esphome configurations. Maybe is is a consequence of the three similarly names switches?

wizzor, 2022-02-03:
I may have a related issue: I have text sensors which I set based on an external HTTP call with a lambda using the following code:
`id(location_id).publish_state(locationid);
              ESP_LOGD("main", "Flushing values to flash");
              global_preferences->sync();`

On reboot the states of the text sensors return a null value. I suspect this issue is related.

marshn, 2022-03-16:
I have tried using 2022.3, and this issue is still present. 

This configuration creates three switches, all of which restore to the same state from flash, even if only one of them has its state changed. 

The first logfile extract shows that having the switches to "Switch 1"=OFF, "Switch 2"=ON, "Switch 3"=OFF all three are restored to the 'ON' state when the board is restarted.

Changing the names of the three switches in the configuraiton to "Switch 11", "Switch 22" and "Switch 33" result in all three being restored to their correct states when the board is restarted.

```
substitutions:
  wifi_password: "xxxxxxxx"

###############################################################################
# Main Config
###############################################################################
# esp8266:
esp32:
  board: pico32
  framework:
    type: arduino

esphome:    
  name: flash-test

wifi:
  ssid: "BF IoT"
  password: ${wifi_password}
  use_address: "192.168.0.129"

ota:
  password: password

web_server:
  port: 80
  version: 1

# The debug component can be used to debug problems with ESPHome. At startup, it 
# prints a bunch of useful information like reset reason, free heap size, ESPHome 
# version and so on.
# Logger must be at least debug (default)
debug:
  update_interval: 30s

# Enable logging
logger:
    
###############################################################################
# Switches
###############################################################################
switch:
  - platform: template
    name: Switch 1
    id: "sw_1"
    optimistic: true
    restore_state: true

  - platform: template
    name: Switch 2
    id: "sw_2"
    optimistic: true
    restore_state: true

  - platform: template
    name: Switch 3
    id: "sw_3"
    optimistic: true
    restore_state: true
    
###############################################################################
# Buttons
[... 106 lines cut ...]
[12:58:18][D][switch:037]: 'Switch 22': Sending state ON
[12:58:21][D][button:013]: 'Restart' Pressed.
[12:58:21][I][restart.button:012]: Restarting device...
[12:58:22][I][app:133]: Rebooting safely...
[12:58:22][D][esp32.preferences:114]: Saving preferences to flash...
...
[12:58:22][I][logger:214]: Log initialized
[12:58:22][C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[12:58:22][D][esp32.preferences:114]: Saving preferences to flash...
[12:58:22][I][app:029]: Running through setup()...
[12:58:22][D][template.switch:050]:   Restored state OFF
[12:58:22][D][switch:017]: 'Switch 11' Turning OFF.
[12:58:22][D][switch:037]: 'Switch 11': Sending state OFF
[12:58:22][D][template.switch:050]:   Restored state ON
[12:58:22][D][switch:013]: 'Switch 22' Turning ON.
[12:58:22][D][switch:037]: 'Switch 22': Sending state ON
[12:58:22][D][template.switch:050]:   Restored state OFF
[12:58:22][D][switch:017]: 'Switch 32' Turning OFF.
[12:58:22][D][switch:037]: 'Switch 32': Sending state OFF
[12:58:22][C][wifi:037]: Setting up WiFi...
```

I think that the issue is is a consequence of the three similarly named switches (names only differing by the last byte of the ID) generating similar hashed IDs, which only differ in the three least dignificant bits. In 'components\esp32\prefferences.cpp' the hashed IDs are XORed with the the flash offsets (which also only differ in the three LSB) and results in identical key values for the three switches. All three switches then have their state restored from the same flash data.

Changing the name of the switches, adding a second byte with a different value, creates a completely different hashed ID. This means that the three switched have completely different key values and are each resored from their own flash data.

cinadr, 2022-04-22:
Hi!

I have made some custom scheduler component and wanted to save/load schedules to/from flash. I experienced that I can only restore the last component from flash. I also dug through espressif NVS docs. I think the offset tracking has no use at all, as the NVS itself works solely key based. 

Also there is no guarantee the the order of written components will be the same while reading back data and the offset will differ from the original one. This resulting in a messed up key calculation which will not read back the data. Furthermore the offset is based on the length of the data which get more complicated while writing different sized object to flash. 

So I have an idea that the whole offset approach should be left out of key calculation.

P.S.: The <current_offset> field is not used in the whole codebase of ESPHome only in this file and only in make_preference method.