esphome/issues #3010: OTA updates failing with `IndexError: list index out of range`

arantes555, 2022-02-02:
### The problem

Hello there ! I have a problem with OTA updates on ESPHome for a couple of D1 Mini boards : whenever I try an OTA install or update, it fails with an `IndexError: list index out of range` error.

### Which version of ESPHome has the issue?

v2022.1.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

D1 Mini

### Anything in the logs that might be useful for us?

```txt
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 825, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 812, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 380, in command_run
    exit_code = upload_program(config, args, port)
  File "/esphome/esphome/__main__.py", line 272, in upload_program
    return espota2.run_ota(host, remote_port, password, CORE.firmware_bin)
  File "/esphome/esphome/espota2.py", line 325, in run_ota
    return run_ota_impl_(remote_host, remote_port, password, filename)
  File "/esphome/esphome/espota2.py", line 313, in run_ota_impl_
    perform_ota(sock, password, file_handle, filename)
  File "/esphome/esphome/espota2.py", line 272, in perform_ota
    receive_exactly(sock, 1, "receive OK", RESPONSE_RECEIVE_OK)
  File "/esphome/esphome/espota2.py", line 92, in receive_exactly
    check_error(data, expect)
  File "/esphome/esphome/espota2.py", line 108, in check_error
    dat = data[0]
IndexError: list index out of range
```
```

### Additional information

I have an unusual network setup, with the ESPHome devices being on a separate network, with a static IP. However, routing is configured correctly, and the static IP in question is reachable from my local network.

LuSkOb, 2022-02-25:
Hello, I have exactly the same problem.

Warter21, 2022-04-03:
Hello, same here with the latest version: 2022.03.02.
or frozen on this state:
INFO Uploading .esphome/build/epaper-display/.pioenvs/epaper-display/firmware.bin (500608 bytes)

AalianKhan, 2022-08-29:
Getting the same error, any update on it?

raccettura, 2022-09-22:
I'm getting this error as well since updating to 2022.9:
```Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/site-packages/esphome/__main__.py", line 931, in main
    return run_esphome(sys.argv)
  File "/usr/local/lib/python3.9/site-packages/esphome/__main__.py", line 918, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/usr/local/lib/python3.9/site-packages/esphome/__main__.py", line 384, in command_run
    exit_code = upload_program(config, args, port)
  File "/usr/local/lib/python3.9/site-packages/esphome/__main__.py", line 276, in upload_program
    return espota2.run_ota(host, remote_port, password, CORE.firmware_bin)
  File "/usr/local/lib/python3.9/site-packages/esphome/espota2.py", line 336, in run_ota
    return run_ota_impl_(remote_host, remote_port, password, filename)
  File "/usr/local/lib/python3.9/site-packages/esphome/espota2.py", line 324, in run_ota_impl_
    perform_ota(sock, password, file_handle, filename)
  File "/usr/local/lib/python3.9/site-packages/esphome/espota2.py", line 245, in perform_ota
    receive_exactly(sock, 1, "binary size", RESPONSE_UPDATE_PREPARE_OK)
  File "/usr/local/lib/python3.9/site-packages/esphome/espota2.py", line 93, in receive_exactly
    check_error(data, expect)
  File "/usr/local/lib/python3.9/site-packages/esphome/espota2.py", line 109, in check_error
    dat = data[0]
IndexError: list index out of range
```

AalianKhan, 2022-09-22:
I tried clearing the build and that worked for me