esphome/issues #3011: INA226 is not work

sirarikako, 2022-02-02:
### The problem

INA226 is not work in 2022.1.2，but is work in 2021.12.3

I tested ver:
2021.12.3 work
2022.1.0 not work
2022.1.2 not work

### Which version of ESPHome has the issue?

2022.1.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.12.5

### What platform are you using?

ESP8266

### Board

nodemcu or ESP-12E

### Component causing the issue

INA226

### Example YAML snippet

```yaml
i2c:
  sda: D6
  scl: D7
  scan: True
  id: i2c_1
  
sensor:
  - platform: ina226
    address: 0x40
    shunt_resistance: 0.002 ohm
    # current:
    #   name: "INA226 Current"
    # power:
    #   name: "INA226 Power"
    bus_voltage:
      name: "INA226 Bus Voltage"
      filters:
        - lambda: |-
            if( x < 1.00 ){
              return 0.00;
            }else{
              return x;
            }
      # id: "ina226_bus_voltage"
      # on_value:
      #   then:
      #     - lambda: |-
      #         if( id(ina226_bus_voltage).state < 1.00 ){
      #           id(ina226_bus_voltage).publish_state(0.00);
      #         }  
    # shunt_voltage:
    #   name: "INA226 Shunt Voltage"
    max_current: 3.2A
    update_interval: 5s
```

### Anything in the logs that might be useful for us?

```txt
[I][app:102]: ESPHome version 2022.1.2 compiled on Feb  2 2022, 19:57:51
[C][wifi:488]: WiFi:
[C][wifi:350]:   Local MAC: DC:4F:22:55:D4:32
[C][wifi:351]:   SSID: [redacted]
[C][wifi:352]:   IP Address: xxxxxx
[C][wifi:354]:   BSSID: [redacted]
[C][wifi:355]:   Hostname: 'ina226-test'
[C][wifi:357]:   Signal strength: -61 dB ▂▄▆█
[V][wifi:359]:   Priority: 0.0
[C][wifi:361]:   Channel: 11
[C][wifi:362]:   Subnet: 255.255.255.0
[C][wifi:363]:   Gateway: xxxxxx
[C][wifi:364]:   DNS1: xxxxxx
[C][wifi:365]:   DNS2: 0.0.0.0
[C][logger:233]: Logger:
[C][logger:234]:   Level: VERY_VERBOSE
[C][logger:235]:   Log Baud Rate: 115200
[C][logger:236]:   Hardware UART: UART0
[C][i2c.arduino:037]: I2C Bus:
[C][i2c.arduino:038]:   SDA Pin: GPIO12
[C][i2c.arduino:039]:   SCL Pin: GPIO13
[C][i2c.arduino:040]:   Frequency: 50000 Hz
[C][i2c.arduino:043]:   Recovery: bus successfully recovered
[I][i2c.arduino:053]: Results from i2c bus scan:
[I][i2c.arduino:059]: Found i2c device at address 0x40
[C][switch.gpio:048]: GPIO Switch 'menjin'
[C][switch.gpio:049]:   Pin: GPIO5
[C][switch.gpio:071]:   Restore Mode: Restore (Defaults to OFF)
[C][switch.gpio:048]: GPIO Switch 'door_bee'
[C][switch.gpio:049]:   Pin: GPIO12
[C][switch.gpio:071]:   Restore Mode: Restore (Defaults to OFF)
[C][ina226:080]: INA226:
[C][ina226:081]:   Address: 0x40
[E][ina226:084]: Communication with INA226 failed!
[C][captive_portal:144]: Captive Portal:
[C][ota:082]: Over-The-Air Updates:
[C][ota:083]:   Address: ina226-test.local:8266
[C][ota:086]:   Using Password.
[C][api:134]: API Server:
[C][api:135]:   Address: ina226-test.local:6053
[C][api:139]:   Using noise encryption: NO
[C][mdns:084]: mDNS:
[C][mdns:085]:   Hostname: ina226-test
```

### Additional information

2021.12.3: platformio/tool-scons @ ~4.40200.0
2022.1.0: platformio/tool-scons @ ~4.40300.0

tmarkson, 2022-05-13:
I'm seeing this too after upgrading two different nodes to 2022.3.2.  Previous firmware that did work was from last year, unsure the exact version.  Both of the nodes are Nodemcu V3.

OM-3kChcBjgFR2COfP, 2024-03-13:
Same problem too: **Communication with INA226 failed!**

NodeMCU lolin v3
Home Assistant Core 2024.3.0
Esphome [2024.2.2]

```
[09:17:58][C][logger:448]:   Level: DEBUG
[09:17:58][C][logger:449]:   Log Baud Rate: 115200
[09:17:58][C][logger:451]:   Hardware UART: UART0
[09:17:58][C][i2c.arduino:053]: I2C Bus:
[09:17:58][C][i2c.arduino:054]:   SDA Pin: GPIO4
[09:17:58][C][i2c.arduino:055]:   SCL Pin: GPIO5
[09:17:58][C][i2c.arduino:056]:   Frequency: 50000 Hz
[09:17:58][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[09:17:58][I][i2c.arduino:069]: Results from i2c bus scan:
[09:17:58][I][i2c.arduino:075]: Found i2c device at address 0x0C
[09:17:58][I][i2c.arduino:075]: Found i2c device at address 0x40
[09:17:58][C][dallas.sensor:075]: DallasComponent:
[09:17:58][C][dallas.sensor:076]:   Pin: GPIO12
[09:17:58][C][dallas.sensor:077]:   Update Interval: 120.0s
[09:17:58][W][dallas.sensor:080]:   Found no sensors!
[09:17:58][C][dallas.sensor:089]:   Device 'NAVIEN - 1'
[09:17:58][C][dallas.sensor:089]:     Device Class: 'temperature'
[09:17:58][C][dallas.sensor:089]:     State Class: 'measurement'
[09:17:58][C][dallas.sensor:089]:     Unit of Measurement: '°C'
[09:17:58][C][dallas.sensor:089]:     Accuracy Decimals: 1
[09:17:58][C][dallas.sensor:097]:     Address: 0x4e3ce10457af7b28
[09:17:58][C][dallas.sensor:098]:     Resolution: 12
[09:17:58][C][dallas.sensor:089]:   Device 'NAVIEN - 2'
[09:17:58][C][dallas.sensor:089]:     Device Class: 'temperature'
[09:17:58][C][dallas.sensor:089]:     State Class: 'measurement'
[09:17:58][C][dallas.sensor:089]:     Unit of Measurement: '°C'
[09:17:58][C][dallas.sensor:089]:     Accuracy Decimals: 1
[09:17:58][C][dallas.sensor:097]:     Address: 0x053ce104571e4428
[09:17:58][C][dallas.sensor:098]:     Resolution: 12
[09:17:58][C][ina226:081]: INA226:
[09:17:58][C][ina226:082]:   Address: 0x40
[09:17:58][E][ina226:085]: Communication with INA226 failed!
```

onurody, 2024-03-22:
Communication with INA226 failed!
[08:29:37][C][wifi:421]:   Hostname: 'dc-monitoring'
[08:29:37][C][wifi:423]:   Signal strength: -61 dB ▂▄▆█
[08:29:37][C][wifi:427]:   Channel: 6
[08:29:37][C][wifi:428]:   Subnet: 255.255.255.0
[08:29:37][C][wifi:429]:   Gateway: 192.168.1.1
[08:29:37][C][wifi:430]:   DNS1: 192.168.1.1
[08:29:37][C][wifi:431]:   DNS2: 0.0.0.0
[08:29:37][C][logger:166]: Logger:
[08:29:37][C][logger:167]:   Level: DEBUG
[08:29:37][C][logger:169]:   Log Baud Rate: 115200
[08:29:37][C][logger:170]:   Hardware UART: UART0
[08:29:37][C][i2c.arduino:059]: I2C Bus:
[08:29:37][C][i2c.arduino:060]:   SDA Pin: GPIO4
[08:29:37][C][i2c.arduino:061]:   SCL Pin: GPIO5
[08:29:37][C][i2c.arduino:062]:   Frequency: 50000 Hz
[08:29:37][C][i2c.arduino:065]:   Recovery: bus successfully recovered
[08:29:37][I][i2c.arduino:075]: Results from i2c bus scan:
[08:29:37][I][i2c.arduino:077]: Found no i2c devices!
[08:29:37][C][ina226:087]: INA226:
[08:29:37][C][ina226:088]:   Address: 0x40
[08:29:37][E][ina226:091]: Communication with INA226 failed!
[08:29:37][C][captive_portal:088]: Captive Portal:
[08:29:37][C][sntp:055]: SNTP Time:
[08:29:37][C][sntp:056]:   Server 1: '0.pool.ntp.org'
[08:29:37][C][sntp:057]:   Server 2: '1.pool.ntp.org'
[08:29:37][C][sntp:058]:   Server 3: '2.pool.ntp.org'
[08:29:37][C][sntp:059]:   Timezone: '<+03>-3'
[08:29:37][C][mdns:115]: mDNS:
[08:29:37][C][mdns:116]:   Hostname: dc-monitoring
[08:29:38][C][ota:096]: Over-The-Air Updates:
[08:29:38][C][ota:097]:   Address: dc-monitoring.local:8266
[08:29:38][C][ota:100]:   Using Password.
[08:29:38][C][ota:103]:   OTA version: 2.
[08:29:38][C][api:139]: API Server:
[08:29:38][C][api:140]:   Address: dc-monitoring.local:6053
[08:29:38][C][api:142]:   Using noise encryption: YES

ssieb, 2024-03-22:
@onurody the i2c scan doesn't see the device, so you need to fix that first.

onurody, 2024-03-22:
I used a very simple code, but I'm having a communication problem with ina226. I'm a bit clumsy. What exactly is the error?
i2c:
  sda: GPIO4
  scl: GPIO5
  scan: true
  id: bus_a   
sensor:
  
  - platform: ina226
    address: 0x40
    shunt_resistance: 0.00025 ohm
    current:
      name: "Battery Shunt Current"
    power:
      name: "Battery Total Power"
      filters:
        - lambda: "return id(shunt_voltage).state > 0 ?  x  : -x;"
    bus_voltage:
      name: "Bus Voltage"
    shunt_voltage:
      name: "Shunt Voltage"
      id: shunt_voltage
      unit_of_measurement: "mV"
      filters:
        - multiply: 1000
    max_current: 300A

time:
  - platform: sntp
    id: my_time

ssieb, 2024-03-22:
Probably something with your wiring.  But this isn't the place.  Come ask on discord if you need help.
As long as you see `[I][i2c.arduino:077]: Found no i2c devices!`, this issue isn't relevant.

onurody, 2024-03-22:
thank you i will check again

OM-3kChcBjgFR2COfP, 2024-03-22:
[https://github.com/esphome/issues/issues/5598#issuecomment-2001904096](url)

ssieb, 2024-03-22:
The `[VV][i2c.arduino:151]: TX failed: unknown error 4` error is from the low-level ESP8266 library and means that the pin is being held low for some reason.
@OM-3kChcBjgFR2COfP if you change the log level to very_verbose and see that message, please come find me on the esphome discord.

OM-3kChcBjgFR2COfP, 2024-03-22:
> The `[VV][i2c.arduino:151]: TX failed: unknown error 4` error is from the low-level ESP8266 library and means that the pin is being held low for some reason. @OM-3kChcBjgFR2COfP if you change the log level to very_verbose and see that message, please come find me on the esphome discord.

Thanks for the answer, I changed the parameter a little later, I’ll write on discord

```
esp32-interrupt
States
Name	State	Actions
Mercury-230 Frequency	NA	
Mercury-230 Value Active	NA	
Mercury-230 Value Reactive	NA	
Mercury-230 Volts A	NA	
Mercury-230 Volts B	NA	
Mercury-230 Volts C	NA	
Mercury-230 Watts A	NA	
Mercury-230 Watts B	NA	
Mercury-230 Watts C	NA	
Mercury-230 Watts Summ	NA	
Mercury-230 Ampers A	NA	
Mercury-230 Ampers B	NA	
Mercury-230 Ampers C	NA	
Mercury-230 Ampers Summ	NA	
ESP32-Mercury-Restart	undefined	Press
Mercury-230 Serial Number		
Mercury-230 ESPHome Version	2024.2.2 Mar 22 2024, 19:25:20	
See ESPHome Web API for REST API documentation.

OTA Update
Файл не выбран
Debug Log
[V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[V][json:058]: Size after shrink 108 bytes
[V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[V][json:058]: Size after shrink 164 bytes
[V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[V][json:058]: Size after shrink 152 bytes
[V][json:058]: Size after shrink 152 bytes
[V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[V][json:058]: Size after shrink 156 bytes
[V][json:058]: Size after shrink 156 bytes
[V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[V][json:058]: Size after shrink 148 bytes
[V][json:058]: Size after shrink 148 bytes
[V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[V][json:058]: Size after shrink 148 bytes
[V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[V][json:058]: Size after shrink 148 bytes
[V][json:058]: Size after shrink 148 bytes
[V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[V][json:058]: Size after shrink 148 bytes
[V][json:058]: Size after shrink 148 bytes
[V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[V][json:058]: Size after shrink 148 bytes
[V][json:058]: Size after shrink 148 bytes
[V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[V][json:058]: Size after shrink 148 bytes
[V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[V][json:058]: Size after shrink 148 bytes
[V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[V][json:058]: Size after shrink 148 bytes
[V][json:058]: Size after shrink 148 bytes
[V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[V][json:058]: Size after shrink 148 bytes
[V][json:058]: Size after shrink 148 bytes
[V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[V][json:058]: Size after shrink 148 bytes
[V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[V][json:058]: Size after shrink 152 bytes
[V][json:058]: Size after shrink 152 bytes
[V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[... 28 lines cut ...]
[D][Mercury:205]: 0000067100: [<=] 1C 08 16 21 (480E)
[D][Mercury:205]: 0000067601: [<=] 1C 08 16 40 (89E6)
[0;38m[VV][scheduler:226]: Running interval 'update' with interval=10000 last_execution=58360 (now=68360)[0m
[VV][scheduler:226]: Running interval 'update' with interval=10000 last_execution=58360 (now=68360)
[0;38m[VV][scheduler:226]: Running interval '' with interval=10000 last_execution=64557 (now=74557)[0m
[VV][scheduler:226]: Running interval '' with interval=10000 last_execution=64557 (now=74557)
[D][Mercury:205]: 0000075099: [<=] 1C 08 01 (7606)
[D][Mercury:205]: 0000075600: [<=] 1C 05 00 00 (17B5)
[D][Mercury:205]: 0000076101: [<=] 1C 08 16 00 (8816)
[D][Mercury:205]: 0000076605: [<=] 1C 08 16 11 (481A)
[D][Mercury:205]: 0000077106: [<=] 1C 08 16 21 (480E)
[D][Mercury:205]: 0000077607: [<=] 1C 08 16 40 (89E6)
[0;38m[VV][scheduler:226]: Running interval 'update' with interval=10000 last_execution=68360 (now=78360)[0m
[VV][scheduler:226]: Running interval 'update' with interval=10000 last_execution=68360 (now=78360)
[0;38m[VV][scheduler:226]: Running interval '' with interval=10000 last_execution=74557 (now=84557)[0m
[VV][scheduler:226]: Running interval '' with interval=10000 last_execution=74557 (now=84557)
[D][Mercury:205]: 0000085096: [<=] 1C 08 01 (7606)
[D][Mercury:205]: 0000085597: [<=] 1C 05 00 00 (17B5)
[D][Mercury:205]: 0000086098: [<=] 1C 08 16 00 (8816)
[D][Mercury:205]: 0000086599: [<=] 1C 08 16 11 (481A)
[D][Mercury:205]: 0000087100: [<=] 1C 08 16 21 (480E)
[D][Mercury:205]: 0000087601: [<=] 1C 08 16 40 (89E6)
[0;38m[VV][scheduler:226]: Running interval 'update' with interval=10000 last_execution=78360 (now=88366)[0m
[VV][scheduler:226]: Running interval 'update' with interval=10000 last_execution=78360 (now=88366)
```

onurody, 2024-03-23:
I solved my problem, I threw the breadbord in the trash, I renewed all the cables, I changed my screen, I compiled the following code, as a result, esphome saw my sensor, now I am working to add the second sensor
i2c:
  sda: GPIO4
  scl: GPIO5

display:
  - platform: lcd_pcf8574
    dimensions: 20x4
    address: 0x27
    id: lcd_display
    lambda: |-
      

sensor:
  - platform: ina226
    address: 0x40
    shunt_resistance: 0.07465 ohm
    max_current: 3.2A
    # adc time used for both, Bus Voltage and Shunt Voltage
    adc_time: 140us
    adc_averaging: 128
    update_interval: 1s
    current:
      name: "INA226 Current"
      id: current
      filters: 
        - multiply: 100
    power:
      name: "INA226 Power"
      id: power
      filters: 
        - multiply: 300.24375
            
    bus_voltage:
      name: "INA226 Bus Voltage"
      id: voltage
      filters:
        - multiply: 3.24375
    shunt_voltage:
      name: "INA226 Shunt Voltage"