esphome/issues #3012: two uart sensor not working on esp32s2

shiwzhi, 2022-02-02:
### The problem

I'm trying to use ESPHome on esp32s2 board to connect two uart sensor PMS7003 and MH-Z19,  when using only one uart, both sensor works, when using two uart together, I can't get logs from ESPhome, but OTA works

### Which version of ESPHome has the issue?

2022.1.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

Wemos s2 mini

### Component causing the issue

uart sensor

### Example YAML snippet

```yaml
esphome:
  name: esp32s2co2

esp32:
  board: featheresp32-s2
  framework:
    type: esp-idf
    version: latest

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "HOME"
  password: "12345679"
  use_address: 192.168.2.220

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32S2Co2 Fallback Hotspot"
    password: "8cUe74jjHJmL"

i2c:
  sda: 3
  scl: 2
  scan: true
  id: bus_a

uart:
  - rx_pin: 21
    tx_pin: 18
    baud_rate: 9600
    id: pmsuart
  - rx_pin: 17
    tx_pin: 16
    baud_rate: 9600
    id: mhz19uart

sensor:
  - platform: mhz19
    uart_id: mhz19uart
    co2:
      name: "esp32s2 MH-Z19 CO2 Value"
    temperature:
      name: "esp32s2 MH-Z19 Temperature"
    update_interval: 10s
    automatic_baseline_calibration: true
  
  - platform: pmsx003
    uart_id: pmsuart
    type: PMSX003
    pm_1_0:
      name: "esp32s2 pm <1.0µm Concentration"
    pm_2_5:
      name: "esp32s2 pm <2.5µm Concentration"
    pm_10_0:
      name: "esp32s2 pm <10.0µm Concentration"
```

### Anything in the logs that might be useful for us?

```txt
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.2.220 using esphome API
WARNING Can't connect to ESPHome API for 192.168.2.220: Error connecting to ('192.168.2.220', 6053): [Errno 61] Connect call failed ('192.168.2.220', 6053)
INFO Trying to reconnect to 192.168.2.220 in the background
```

AndreasBVB, 2022-09-02:
Would like to see updates on this. Running into the same problem 2022.8.1. Thanks!

AndreasBVB, 2022-09-24:
Further info:
Switched to Espressif-Arduino, no such a problem:

platform_packages:
  - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.3
platform:
  - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream