esphome/issues #3015: RDM6300 no longer compiles

tomlut, 2022-02-04:
### The problem

The example code in the documents for the RDM6300 component:

```
rdm6300:
  # ...
  on_tag:
    then:
      - homeassistant.tag_scanned: !lambda 'return uint32_to_string(x);'
```
<https://esphome.io/components/binary_sensor/rdm6300.html#on-tag>

Now causes the build to fail with this error:

```
/config/esphome/nfc-scanner.yaml: In lambda function:
/config/esphome/nfc-scanner.yaml:98:32: error: 'uint32_to_string' was not declared in this scope
       - homeassistant.tag_scanned: !lambda 'return uint32_to_string(x);'
```

### Which version of ESPHome has the issue?

2022.1.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.1

### What platform are you using?

ESP32

### Board

mhetesp32minikit

### Component causing the issue

RDM6300

### Example YAML snippet

```yaml
rdm6300:
  uart_id: nfc_uart
  on_tag:
    then:
      - homeassistant.tag_scanned: !lambda 'return uint32_to_string(x);'
```
```

maxill1, 2022-02-05:
Same problem here with identical Esphome and HA versions

tomlut, 2022-03-21:
Still an issue with ESPHome 2022.3.0

tomlut, 2022-03-22:
The Docs are updated, solving this issue:

```
rdm6300:
  # ...
  on_tag:
    then:
      - homeassistant.tag_scanned: !lambda 'return to_string(x);'
```

closed 2022-03-22 by tomlut