esphome/issues #3016: Compile Error

coedshowers, 2022-02-05:
### The problem

I have a compiling error on ESP home after doing the newest update.
Running Raspberry pi4 with argon m.2 case with 180gb and mariaDB.
everything was fine before the update, was flashing some esp32 yesterday and now this. ( I’ll post logs at bottom ).
Everything I can find is a million post bitching and a few people saying reinstall (esphome) which I have it did not work, no real answers and a bunch of “try this, but I haven’t had the problem but try it anyways…” nothing has worked.
From what I gather the common theme is on pi 4 HA image its 64bit os and some toolkit package it needs to compile is only 32bit compatible but somehow somewhere HA has no option to change this and no control to install the package for 32bit from what I have read so far. Still no fixes from the forum post I’ve read that actually work.

INFO Reading configuration /config/esphome/aq-2.yaml…
INFO Generating C++ source…
INFO Compiling app…
Processing aq-2 (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.3.2)
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder → [Library Dependency Finder (LDF) — PlatformIO latest documentation](https://bit.ly/configure-pio-ldf)
Dependency Graph
|-- 1.2.2
|-- 1.0
|-- 1.0
|-- 1.0
|-- 2.1.0
| |-- 1.2.2
|-- 1.1.0
|-- 1.0
Compiling /data/aq-2/.pioenvs/aq-2/src/esphome/components/api/api_connection.cpp.o
cc1plus: internal compiler error: Illegal instruction
Please submit a full bug report,
with preprocessed source if appropriate.
See http://gcc.gnu.org/bugs.html for instructions.
Compiling /data/aq-2/.pioenvs/aq-2/src/esphome/components/api/api_frame_helper.cpp.o
cc1plus: internal compiler error: Illegal instruction
Please submit a full bug report,
with preprocessed source if appropriate.
See http://gcc.gnu.org/bugs.html for instructions.
Compiling /data/aq-2/.pioenvs/aq-2/src/esphome/components/api/api_pb2.cpp.o
cc1plus: internal compiler error: Illegal instruction
Please submit a full bug report,
with preprocessed source if appropriate.
See http://gcc.gnu.org/bugs.html for instructions.
*** [/data/aq-2/.pioenvs/aq-2/src/esphome/components/api/api_connection.cpp.o] Error 1
Compiling /data/aq-2/.pioenvs/aq-2/src/esphome/components/api/api_pb2_service.cpp.o
*** [/data/aq-2/.pioenvs/aq-2/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
*** [/data/aq-2/.pioenvs/aq-2/src/esphome/components/api/api_pb2.cpp.o] Error 1
cc1plus: internal compiler error: Illegal instruction
Please submit a full bug report,
with preprocessed source if appropriate.
See http://gcc.gnu.org/bugs.html for instructions.
*** [/data/aq-2/.pioenvs/aq-2/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
========================== [FAILED] Took 3.16 seconds ==========================

### Which version of ESPHome has the issue?

2022.1.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.2

### What platform are you using?

ESP32

### Board

esp32

### Component causing the issue

home assistant

### Example YAML snippet

```yaml
esphome:
  name: aq-office

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "XXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Aq-Office Fallback Hotspot"
    password: "TiH9asFoEkCz"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
This is why i hate home assistant and keep going back to smart things hub...
```

coedshowers, 2022-02-05:
figure out if you uninstall esphome, go and rename ( HA wouldnt let me delete ) but rename it anything then reinstall esphome and it worked again... its like every HA update is 5 new features and 109843092809 new bugs.

OttoWinter, 2022-02-05:
That's an issue in the toolchains platformio downloads to compile the project. Both the toolchain and platformio are out of our control (and would be an upstream issue) unfortunately.

closed 2022-07-05 by github-actions[bot]