esphome/issues #3019: teleinfo component not working

libussa, 2022-02-05:
### The problem

I am trying to use the [teleinfo component](https://esphome.io/components/sensor/teleinfo.html) to get information from an electrical counter.

Mine is using the "Standard" mode, as opposed to the historical one.

No data is visible in Home assistant sensors, except wifi RSSI and global status.

I added some debug to the uart part to see the actual data being received, which seems correct. See below for raw data.

### Which version of ESPHome has the issue?

2022.2.0-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.2.1

### What platform are you using?

ESP8266

### Board

d1 mini

### Component causing the issue

teleinfo

### Example YAML snippet

```yaml
esphome:
  name: erl
  platform: ESP8266
  board: d1_mini
    
# Enable logging
logger:
  baud_rate: 0
  level: VERBOSE

# Enable Home Assistant API
api:
  password: "secret"

ota:
  password: "secret"

wifi:
  ssid: "iot"
  password: "secret"
  manual_ip:
    static_ip: 10.10.10.140
    gateway: 10.10.10.1
    subnet: 255.255.255.0
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Erl Fallback Hotspot"
    password: "password"

captive_portal:

[... 307 lines cut ...]
[17:32:25][D][uart_debug:158]: <<< "IRMS1\t002\t0\r\n"
[17:32:25][D][uart_debug:158]: <<< "URMS1\t229\tG\r\n"
[17:32:25][D][uart_debug:158]: <<< "PREF\t06\tE\r\n"
[17:32:25][D][uart_debug:158]: <<< "PCOUP\t06\t_\r\n"
[17:32:25][D][uart_debug:158]: <<< "SINSTS\t00547\tV\r\n"
[17:32:25][D][uart_debug:158]: <<< "SMAXSN\tH220205160535\t01920\t8\r\n"
[17:32:25][V][component:199]: Component teleinfo took a long time for an operation (0.10 s).
[17:32:25][V][component:200]: Components should block for at most 20-30ms.
[17:32:25][D][uart_debug:158]: <<< "SMAXSN-1\tH220204201357\t03200\tL\r\n"
[17:32:25][D][uart_debug:158]: <<< "CCASN\tH220205173000\t00418\t>\r\n"
[17:32:25][D][uart_debug:158]: <<< "CCASN-1\tH220205170000\t00424\tV\r\n"
[17:32:25][D][uart_debug:158]: <<< "UMOY1\tH220205173000\t228\t0\r\n"
[17:32:25][D][uart_debug:158]: <<< "STGE\t003A4001\t>\r\n"
[17:32:25][V][component:199]: Component teleinfo took a long time for an operation (0.06 s).
[17:32:25][V][component:200]: Components should block for at most 20-30ms.
[17:32:25][D][uart_debug:158]: <<< "MSG1\t     PAS DE          MESSAGE    \t<\r\n"
[17:32:25][D][uart_debug:158]: <<< "PRM\t19474963772031\t@\r\n"
[17:32:25][D][uart_debug:158]: <<< "RELAIS\t000\tB\r\n"
[17:32:25][D][uart_debug:158]: <<< "NTARF\t01\tN\r\n"
[17:32:25][D][uart_debug:158]: <<< "NJOURF\t00\t&\r\n"
[17:32:25][D][uart_debug:158]: <<< "NJOURF+1\t00\tB\r\n"
[17:32:25][V][component:199]: Component teleinfo took a long time for an operation (0.08 s).
[17:32:25][V][component:200]: Components should block for at most 20-30ms.
[17:32:25][D][uart_debug:158]: <<< "PJOURF+1\t00008001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE\t9\r\x03"
```

0hax, 2022-02-06:
Hi,
I think it doesn't work because the tags your provided (HCHC HCHP and PAPP) are not in the output from your electrical counter.
If you add `EAST` for example, that should work.

libussa, 2022-02-07:
indeed, I should have looked at the datasheet....The tags completely changed between historical and standard mode, and I blindly copied the conf example.

Sorry for the noise!