esphome/issues #3024: RF Bridge send code errors

JPDMaker, 2022-02-06:
### The problem

Unable to send an RF code using the RF Bridge component connected through the UART bus.  Logs consistently show the error:

[W][rf_bridge:104]: Unknown action: 0xA5

I am able to receive codes in the esphome logs but unable to send the same codes. 

### Which version of ESPHome has the issue?

2022.1.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

NodeMCU ESP32   / 433rfbrigf r2 v1.0

### Component causing the issue

RF Bridge

### Example YAML snippet

```yaml
esphome:
  name: sonoff-rf
  platform: ESP32
  board: nodemcu-32s

ota:
  password: 

wifi:
  ssid: 
  password:

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: 
    password: 
    manual_ip:
      static_ip: 
      gateway: 
      subnet: 

captive_portal:

api:

uart:
  tx_pin: 17
  rx_pin: 16
  baud_rate: 19200

logger:
  baud_rate: 0
  level: very_verbose

    
rf_bridge:
  on_code_received:
    then:
      - homeassistant.event:
          event: esphome.rf_code_received
          data:
            sync: !lambda 'char buffer [10];return itoa(data.sync,buffer,16);'
            low: !lambda 'char buffer [10];return itoa(data.low,buffer,16);'
            high: !lambda 'char buffer [10];return itoa(data.high,buffer,16);'
            code: !lambda 'char buffer [10];return itoa(data.code,buffer,16);'
            
switch:
  - platform: template
    optimistic: true
    name: ShopVac
    icon: mdi:vacuum
    on_turn_on:
      - rf_bridge.send_code:
         sync: 0x1306
         low: 0x00B4
         high: 0x01EA
         code: 0xC19178
    on_turn_off:
      - rf_bridge.send_code:
         sync: 0x1306
         low: 0x00B4
         high: 0x01E0
         code: 0xC19174
```

### Anything in the logs that might be useful for us?

```txt
}
[12:56:03][D][switch:013]: 'ShopVac' Turning ON.
[12:56:03][D][switch:037]: 'ShopVac': Sending state ON
[12:56:03][D][rf_bridge:147]: Sending code: sync=0x1306 low=0x00B4 high=0x01EA code=0xC19178
[12:56:03][VV][uart.arduino_esp32:148]:     Flushing...
[12:56:03][VV][api.service:156]: send_switch_state_response: SwitchStateResponse {
  key: 1755628889
  state: YES
}
[12:56:03][V][component:199]: Component api took a long time for an operation (0.09 s).
[12:56:03][V][component:200]: Components should block for at most 20-30ms.
[12:56:03][VV][rf_bridge:023]: Processing byte: 0xAA
[12:56:03][VV][rf_bridge:137]: Parsed: 0xAA
[12:56:03][VV][rf_bridge:023]: Processing byte: 0xA5
[12:56:03][VV][rf_bridge:137]: Parsed: 0xA5
[12:56:03][VV][rf_bridge:023]: Processing byte: 0x13
[12:56:03][W][rf_bridge:104]: Unknown action: 0xA5
[12:56:03][VV][rf_bridge:108]: Parsed: 0x13
[12:56:04][VV][rf_bridge:023]: Processing byte: 0x06
[12:56:04][VV][rf_bridge:023]: Processing byte: 0x00
[12:56:04][VV][rf_bridge:023]: Processing byte: 0xB4
[12:56:04][VV][rf_bridge:023]: Processing byte: 0x01
[12:56:04][VV][rf_bridge:023]: Processing byte: 0xEA
[12:56:04][VV][rf_bridge:023]: Processing byte: 0xC1
[12:56:04][VV][rf_bridge:023]: Processing byte: 0x91
[12:56:04][VV][rf_bridge:023]: Processing byte: 0x78
[12:56:04][VV][rf_bridge:023]: Processing byte: 0x55
[12:56:04][V][component:199]: Component rf_bridge took a long time for an operation (0.21 s).
[12:56:04][V][component:200]: Components should block for at most 20-30ms.
```

### Additional information

[image]

nagyrobi, 2022-02-10:
@JPDMaker how did you connect?
- only 2 wires on the photo, shouldn't be 3? GND, RX, TX?
- how did you disable the onboard ESP82xx from the bridge board?

closed 2022-07-03 by github-actions[bot]

nagyrobi, 2022-10-10:
Also looking to combine multiple IoT devices into one, or create an Ethernet-based RF sender, it is indeed possible to use Sonoff RF Bridge with ESP32, following things are to be considered:
- you need the 4 pins between the RF chip and the switch (`GND`, `TX`, `RX`, `3V3`)
- put the switch into `OFF` position (away from these pins), it will disconnect the RF chip from the onboard ESP8285
- no need to power the Sonoff board separately, by applying 3.3V through the pins above, from the ESP32, the board will power up.
- configuring an RX and TX pair `uart` and `rf_bridge` component on the ESP32 as described [in the documentation](https://esphome.io/components/rf_bridge.html) will be functional.
- optionally/recommended to disable the onboard ESP8285 by pulling its `EN` pin to the ground (the chip-side of `R7`)

(tested with a WirelessTag WT32-ETH01 and a Sonoff RF Bridge R2 V1.0 2017.11.23)