esphome/issues #3025: CAN Bus Component doesn't work with ESP32

DieNummerEins, 2022-02-07:
### The problem

I can't use "canbus.send:" action. I think, there is a problem with can-id in the canbus.send line. I tried everything in your component description.

Furthermore, I need a wiring diagram to connect a few esp32 devices so they can "talk" via CAN bus. Which hardware do I need?

### Which version of ESPHome has the issue?

2022.1.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.3

### What platform are you using?

ESP32

### Board

DOIT ESP 32 DEVKIT V1

### Component causing the issue

CAN Bus

### Example YAML snippet

```yaml
esphome:
  name: esph15can01

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "esph15can01 Fallback Hotspot"
    password: !secret ota_password

captive_portal:

canbus:
  - platform: esp32_can
    tx_pin: GPIO1
    rx_pin: GPIO3
    can_id: 4
    bit_rate: 50kbps

button:
  - platform: template
    name: "CAN Bus Button"
    id: "can_bus_button"
    on_press:
      then:
      - logger.log: Button Pressed
      - canbus.send: 'hello'

switch:
  - platform: template
    name: esph15can01_sw01
    id: esph15can01_sw01
    optimistic: true
    inverted: true
    on_turn_on:
      then:
      #- canbus.send: 'switch turn on'
    on_turn_off:
      then:
      #- canbus.send: 'switch turn off'
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/esph15can01.yaml...
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 825, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 812, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 366, in command_run
    exit_code = write_cpp(config)
  File "/esphome/esphome/__main__.py", line 159, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 171, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 613, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 245, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 151, in wrapped
    await coro(conf)
  File "/esphome/esphome/components/template/button/__init__.py", line 13, in to_code
    await button.new_button(config)
  File "/esphome/esphome/components/button/__init__.py", line 107, in new_button
    await register_button(var, config)
  File "/esphome/esphome/components/button/__init__.py", line 102, in register_button
    await setup_button_core_(var, config)
  File "/esphome/esphome/components/button/__init__.py", line 88, in setup_button_core_
    await automation.build_automation(trigger, [], conf)
  File "/esphome/esphome/automation.py", line 350, in build_automation
    actions = await build_action_list(config[CONF_THEN], templ, args)
  File "/esphome/esphome/automation.py", line 323, in build_action_list
    action = await build_action(conf, templ, arg_type)
  File "/esphome/esphome/automation.py", line 316, in build_action
    ret = await builder(config, action_id, template_arg, args)
  File "/esphome/esphome/components/canbus/__init__.py", line 124, in canbus_action_to_code
    validate_id(config[CONF_CAN_ID], config[CONF_USE_EXTENDED_ID])
KeyError: 'can_id'
```

### Additional information

I have many ESP32-based multi-sensor in my house (with different sensors, switches, relays, buzzers...) and I would like to install much more of them in my house. Right now I am testing a lot and if everything is all right and the system is usable for my wife, children and guests then I install it everywhere. At the end there can be more than 50 devices in my home automation.

One of the tests is the ability to connect the esp-nodes via CAN bus, so they work even if no network and no home assistant is available. Another idea is to use bluetooth but I don't see at the moment that the ble component is able to make a "network", where I can define variables and push them between esphome-devices or something like this.

I would like to help you with this CAN component. Let me know if I can make detailed tests in my context.

blindlight86, 2022-02-08:
same problem

jesserockz, 2022-02-08:
Please try my PR https://github.com/esphome/esphome/pull/3173 and comment there if it fixes the issue.

```yaml
external_components:
  - source: github://pr#3173
    components: [canbus]
```

danielschramm, 2022-02-08:
Hello,
I already prepared some schematics for the ESP32 CAN Interface.
I will take the Task updating the docs, but it might take some weeks as I'm currently very busy.
Daniel

closed 2022-02-08 by jesserockz