esphome/issues #3026: Logging with MQTT throw some garbage

Francis-perso, 2022-02-08:
### The problem

I'm trying to use MQTT to have some log infos, but despite "esphome logs xxx.yaml" finds its children, the MQTT server (Mosquitto) shows some garbage.

### Which version of ESPHome has the issue?

2022.1.2

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

ESP01_1M

### Component causing the issue

logger with MQTT

### Example YAML snippet

```yaml
logger:
  baud_rate: 0
  level: INFO
  on_message:
    level: INFO
    then:
    - wait_until: mqtt.connected
    - mqtt.publish_json:
        topic: domoticz/in
        payload: |-
          root["command"] = "addlogmessage";
          root["message"] = message;
          root["level"] = to_string(level);
```

### Anything in the logs that might be useful for us?

```txt
[01:20:24][I][mqtt:215]: MQTT Connected!
[01:20:24][I][app:062]: setup() finished successfully!
[01:20:24][I][app:102]: ESPHome version 2022.1.2 compiled on Feb  8 2022, 01:14:04
[01:20:24][I][i2c.arduino:053]: Results from i2c bus scan:
[01:20:24][I][i2c.arduino:059]: Found i2c device at address 0x38
```

### Additional information

Seen in MQTT logs (via mosquitto_sub -t domoticz/in) :

{"command":"addlogmessage","message":"[I][mqtt:215]: MQTT Connected!","level":"3"}
{"command":"addlogmessage","message":"[I][app:062]: setup() finished successfully!","level":"3"}
{"command":"addlogmessage","message":"ssfully!","level":"3"}
{"command":"addlogmessage","message":"[I][app:102]: ESPHome version 2022.1.2 compiled on Feb  8 2022, 01:14:04","level":"3"}
{"command":"addlogmessage","message":".2 compiled on %s","level":"3"}
{"command":"addlogmessage","message":".2 compiled on %s","level":"3"}
{"command":"addlogmessage","message":"[I][i2c.arduino:053]: Results from i2c bus scan:","level":"3"}
{"command":"addlogmessage","message":"[I][i2c.arduino:059]: Found i2c device at address 0x38","level":"3"}
{"command":"addlogmessage","message":"dress 0x%02X","level":"3"}
{"command":"addlogmessage","message":"dress 0x%02X","level":"3"}
{"command":"addlogmessage","message":"dress 0x%02X","level":"3"}

OttoWinter, 2022-02-09:
`{"command":"addlogmessage","message":"[I][mqtt:215]: MQTT Connected!","level":"3"}`

that MQTT message is not generated by esphome; esphome would just publish

`[I][mqtt:215]: MQTT Connected!`

(without the JSON-like format). Maybe the domoticz software (or whatever software is publishing to the `domoticz/in` topic) is the issue?

OttoWinter, 2022-02-09:
The action `on_message` is only meant for low level operations like `uart.write` - that's because the buffer used to write the messages gets reused, and by the time `mqtt.publish_json` gets called the data might already be gone.

Francis-perso, 2022-02-09:
The problem is not that data is not published before getting lost, but that it is done multiple times, and not correctly after the first time.
As you can see in the ESPHome logs, we should have only 5 lines to be seen, and I see 11 lines received by MQTT brocker.

closed 2022-07-02 by github-actions[bot]