esphome/issues #3027: How do I convert my working non hex NEC remote protocol to work with new format?

poldim, 2022-02-08:
### The problem

I've had this config running on my fan for a few years.  I can't figure out how to convert my values to be *reversed*.  The device does not have a IR receiver, otherwise I would just test the receive function.  Since I made this thing a few years ago, I've forgotten exactly what I did to get the codes, but I know they've been working great.

I tried looking at some of the hints in [this](https://community.home-assistant.io/t/ir-change-in-nec-what-do-i-need-to-do/365840/16) thread.

[image]

### Which version of ESPHome has the issue?

2021.12.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.2.1

### What platform are you using?

ESP8266

### Board

D1

### Component causing the issue

remote_transmitter

### Example YAML snippet

```yaml
substitutions:
  platform: ESP8266
  board: d1_mini
  device_name: esphome_master_fan
  friendly_name: "Master Fan"
  reboot_timeout_wifi: 900s
  reboot_timeout_api: 1800s
  output_power: 17dB

packages:
  device_base: !include z_common.yaml

#################################

switch:
- platform: template
  id: fan_power_toggle
  turn_on_action:
    remote_transmitter.transmit_nec:
      address: 255
      command: 765

- platform: template
  name: "Fan Power"
  optimistic: True
  assumed_state: True
  turn_on_action:
    - switch.turn_on: fan_power_toggle
  turn_off_action:
    - switch.turn_on: fan_power_toggle

- platform: template
  name: Fan Oscilation
  turn_on_action:
    remote_transmitter.transmit_nec:
      address: 255
      command: 25245

- platform: template
  name: Fan Speed
  turn_on_action:
    remote_transmitter.transmit_nec:
      address: 255
      command: 8925

- platform: template
  name: Fan Mode
  turn_on_action:
    remote_transmitter.transmit_nec:
      address: 255
      command: 49725

- platform: template
  name: Fan Timer
  turn_on_action:
    remote_transmitter.transmit_nec:
      address: 255
      command: 43095

remote_transmitter:
  pin: D5
  carrier_duty_percent: 30%
```

jesserockz, 2022-02-08:
The data expected for nec is 16-bit, so 0xFF in your old config is actually 0x00FF, which needs to reverse taking into account the "leading" 0's

`0x00FF` = `0b0000000011111111`
Reversed:
`0b1111111100000000` = `0xFF00`

`765` = `0x02FD` = `0b0000001011111101`
Reversed:
`0b1011111101000000` = `0xBF40` = `48960`

poldim, 2022-02-08:
How are you doing the conversions?  Whatever I'm using isn't getting me the same numbers as you.  I'm using https://www.rapidtables.com/convert/number/ascii-hex-bin-dec-converter.html and https://onlinebinarytools.com/reverse-binary-numbers

255 works as 0xFF00
765 => 0x2FD, but adding a leading zero, 0x02FD make it equal 2253 in decimal => 00000010 11111101 => 10111111 01000000 => 0xBF 0x40 

25245 => 0xFC 0x2D => 11111100 00101101 => 10110100 00111111 => 0xB4 0x3F
8925 => 0x37C 0x05 => 1101111100 00000101 => 10100000 0011111011 => 0xA0 0x3E 0x03

The odd thing is that I got power/mode/timer working, but oscillation and speed arent

```yaml
substitutions:
  platform: ESP8266
  board: d1_mini
  device_name: esphome_master_fan
  friendly_name: "Master Fan"
  reboot_timeout_wifi: 900s
  reboot_timeout_api: 1800s
  output_power: 17dB

packages:
  device_base: !include z_common.yaml

#################################

switch:
- platform: template
  id: fan_power_toggle
  turn_on_action:
    remote_transmitter.transmit_nec:
      address: 0xFF00
      command: 0xBF40

- platform: template
  name: "Fan Power"
  optimistic: True
  assumed_state: True
  turn_on_action:
    - switch.turn_on: fan_power_toggle
  turn_off_action:
    - switch.turn_on: fan_power_toggle

- platform: template
  name: Fan Oscilation
  turn_on_action:
    remote_transmitter.transmit_nec:
      # address: 255
      address: 0xFF00
      command: 0xB43F #reversed hex
      # address: 0x00FF
      # command: 25245
      # command: 23715 #reversed decimal
      
- platform: template
  name: Fan Speed
  turn_on_action:
    remote_transmitter.transmit_nec:
      # address: 255
      address: 0xFF00
      command: 0xA03E #reversed hex
      # address: 0x00FF
      # command: 8925
      # command: 11985 #reversed decimal

- platform: template
  name: Fan Mode
  turn_on_action:
    remote_transmitter.transmit_nec:
      address: 0xFF00
      command: 0xBC43

- platform: template
  name: Fan Timer
  turn_on_action:
    remote_transmitter.transmit_nec:
      address: 0xFF00
      command: 0xEA15

remote_transmitter:
  pin: D5
  carrier_duty_percent: 30%
```

poldim, 2022-02-09:
Thanks to @jesserockz & @ssieb 

The solution was to enter my container via CLI and use python within the container to convert the values:

1. CLI into container
2. Type `python3`
3. Type `bin(<value>)` and replace value with the hex or decimal value.
4. Reverse this value right to left: `abc` becomes `cba`, the `0b` stays at the front.
5. Count the number of digits, if it's less than 16 digits, add zero's at the end to make it 16 digits
6. Type `hex(0b<value>)` and replace value with the reversed binary.
7. Resulting hex should work as your `command`.

```yaml
substitutions:
  platform: ESP8266
  board: d1_mini
  device_name: esphome_master_fan
  friendly_name: "Master Fan"
  reboot_timeout_wifi: 900s
  reboot_timeout_api: 1800s
  output_power: 17dB

packages:
  device_base: !include z_common.yaml

#################################

switch:
- platform: template
  id: fan_power_toggle
  turn_on_action:
    remote_transmitter.transmit_nec:
      address: 0xFF00
      command: 0xBF40

- platform: template
  name: "Fan Power"
  optimistic: True
  assumed_state: True
  turn_on_action:
    - switch.turn_on: fan_power_toggle
  turn_off_action:
    - switch.turn_on: fan_power_toggle

- platform: template
  name: Fan Oscilation
  turn_on_action:
    remote_transmitter.transmit_nec:
      address: 0xFF00
      command: 0xB946 #reversed hex for v2021.12
      
- platform: template
  name: Fan Speed
  turn_on_action:
    remote_transmitter.transmit_nec:
      # address: 255
      address: 0xFF00
      command: 0xBB44 #reversed hex for v2021.12
      # address: 0x00FF
      # command: 8925
      # command: 11985 #reversed decimal

- platform: template
  name: Fan Mode
  turn_on_action:
    remote_transmitter.transmit_nec:
      address: 0xFF00
      command: 0xBC43

- platform: template
  name: Fan Timer
  turn_on_action:
    remote_transmitter.transmit_nec:
      address: 0xFF00
      command: 0xEA15

remote_transmitter:
  pin: D5
  carrier_duty_percent: 30%
```

closed 2022-02-09 by poldim