esphome/issues #3028: Issues with BME280 probably related to i2c

Serios, 2022-02-08:
### The problem

After upgrade to the latest 2022.1.3 all of my bme280 sensors stopped reporting temp, humidity, pressure. Before that on 2021.12.3 everything was working normally

### Which version of ESPHome has the issue?

esphome v2022.1.3

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

esp-01

### Component causing the issue

BME280, i2c

### Example YAML snippet

```yaml
output:
  - platform: gpio
    pin: GPIO3
    id: bme280_supply

...

i2c:
  sda: GPIO0
  scl: GPIO2

...

platform: bme280
temperature:
  name: "$friendly_devicename Temperature"
  id: bme280_temperature
  oversampling: 2x
pressure:
  name: "$friendly_devicename Pressure"
  id: bme280_pressure
  oversampling: 2x
humidity:
  name: "$friendly_devicename Relative Humidity"
  id: bme280_humidity
  oversampling: 2x
address: 0x77
```

### Anything in the logs that might be useful for us?

```txt
[21:44:10][I][app:062]: setup() finished successfully!
[21:44:10][I][i2c.arduino:053]: Results from i2c bus scan:
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x08
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x09
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x0A
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x0B
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x0C
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x0D
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x0E
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x0F
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x10
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x11
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x12
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x13
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x14
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x15
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x16
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x17
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x18
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x19
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x1A
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x1B
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x1C
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x1D
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x1E
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x1F
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x20
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x21
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x22
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x23
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x24
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x25
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x26
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x27
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x28
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x29
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x2A
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x2B
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x2C
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x2D
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x2E
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x2F
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x30
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x31
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x32
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x33
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x34
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x35
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x36
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x37
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x38
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x39
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x3A
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x3B
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x3C
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x3D
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x3E
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x3F
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x40
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x41
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x42
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x43
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x44
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x45
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x46
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x47
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x48
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x49
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x4A
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x4B
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x4C
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x4D
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x4E
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x4F
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x50
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x51
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x52
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x53
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x54
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x55
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x56
[21:44:10][E][i2c.arduino:061]: Unknown error at address 0x57
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x58
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x59
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x5A
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x5B
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x5C
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x5D
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x5E
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x5F
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x60
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x61
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x62
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x63
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x64
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x65
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x66
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x67
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x68
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x69
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x6A
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x6B
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x6C
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x6D
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x6E
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x6F
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x70
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x71
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x72
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x73
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x74
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x75
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x76
[21:44:11][E][i2c.arduino:061]: Unknown error at address 0x77
[21:44:11][E][bme280.sensor:147]: Communication with BME280 failed!
```

### Additional information

I don't know if is relevant, but during compiling i can see following in the log:

`src/esphome/components/bme280/bme280.cpp:45:20: warning: 'const char* esphome::bme280::oversampling_to_str(esphome::bme280::BME280Oversampling)' defined but not used [-Wunused-function]
 static const char *oversampling_to_str(BME280Oversampling oversampling) {
                    ^
src/esphome/components/bme280/bme280.cpp:64:20: warning: 'const char* esphome::bme280::iir_filter_to_str(esphome::bme280::BME280IIRFilter)' defined but not used [-Wunused-function]
 static const char *iir_filter_to_str(BME280IIRFilter filter) {
                    ^
`

jcastro, 2022-02-11:
If you enable I2C scan is the address still 0x77? (mine is 0x76)
```
i2c:
  sda: D3
  scl: D4
  scan: True
```
If I don't set the address in the sensor section it won't work
```
sensor:
  - platform: bme280
    temperature:
      name: "BME280 Temperature"
      oversampling: 16x
    pressure:
      name: "BME280 Pressure"
    humidity:
      name: "BME280 Humidity"
    update_interval: 30s
    address: 0x76
```

Serios, 2022-02-11:
Hi @jcastro with address set as 0x77 the log returns

`[11:29:32][C][logger:233]: Logger:
[11:29:32][C][logger:234]:   Level: DEBUG
[11:29:32][C][logger:235]:   Log Baud Rate: 115200
[11:29:32][C][logger:236]:   Hardware UART: UART0
[11:29:32][C][i2c.arduino:037]: I2C Bus:
[11:29:32][C][i2c.arduino:038]:   SDA Pin: GPIO0
[11:29:32][C][i2c.arduino:039]:   SCL Pin: GPIO2
[11:29:32][C][i2c.arduino:040]:   Frequency: 50000 Hz
[11:29:32][C][i2c.arduino:046]:   Recovery: failed, SCL is held low on the bus
[11:29:32][I][i2c.arduino:053]: Results from i2c bus scan:
[11:29:32][I][i2c.arduino:059]: Found i2c device at address 0x19
[11:29:32][I][i2c.arduino:059]: Found i2c device at address 0x1A
[11:29:32][I][i2c.arduino:059]: Found i2c device at address 0x1B
[11:29:32][I][i2c.arduino:059]: Found i2c device at address 0x1C
[11:29:32][I][i2c.arduino:059]: Found i2c device at address 0x1D
[11:29:32][I][i2c.arduino:059]: Found i2c device at address 0x1E
[11:29:32][I][i2c.arduino:059]: Found i2c device at address 0x1F
[11:29:32][I][i2c.arduino:059]: Found i2c device at address 0x20
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x22
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x23
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x24
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x25
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x26
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x27
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x28
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x29
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x2A
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x2B
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x2C
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x2D
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x2E
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x2F
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x30
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x31
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x32
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x33
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x34
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x35
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x36
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x37
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x38
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x39
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x3A
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x3B
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x3C
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x3D
[11:29:32][E][i2c.arduino:061]: Unknown error at address 0x3E
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x3F
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x40
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x41
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x42
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x43
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x44
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x45
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x46
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x47
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x48
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x49
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x4A
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x4B
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x4C
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x4D
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x4E
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x4F
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x50
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x51
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x52
[11:29:33][E][i2c.arduino:061]: Unknown error at address 0x53
[... 187 lines cut ...]
[11:33:28][C][adc:057]:   Accuracy Decimals: 2
[11:33:28][C][adc:060]:   Pin: VCC
[11:33:28][C][adc:088]:   Update Interval: 60.0s
[11:33:28][C][bme280.sensor:143]: BME280:
[11:33:28][C][bme280.sensor:144]:   Address: 0x76
[11:33:28][C][bme280.sensor:156]:   IIR Filter: OFF
[11:33:28][C][bme280.sensor:157]:   Update Interval: 60.0s
[11:33:28][C][bme280.sensor:159]:   Temperature 'KWS Temperature'
[11:33:28][C][bme280.sensor:159]:     Device Class: 'temperature'
[11:33:28][C][bme280.sensor:159]:     State Class: 'measurement'
[11:33:28][C][bme280.sensor:159]:     Unit of Measurement: '°C'
[11:33:28][C][bme280.sensor:159]:     Accuracy Decimals: 1
[11:33:28][C][bme280.sensor:160]:     Oversampling: 2x
[11:33:28][C][bme280.sensor:161]:   Pressure 'KWS Pressure'
[11:33:28][C][bme280.sensor:161]:     Device Class: 'pressure'
[11:33:28][C][bme280.sensor:161]:     State Class: 'measurement'
[11:33:28][C][bme280.sensor:161]:     Unit of Measurement: 'hPa'
[11:33:28][C][bme280.sensor:161]:     Accuracy Decimals: 1
[11:33:28][C][bme280.sensor:162]:     Oversampling: 2x
[11:33:28][C][bme280.sensor:163]:   Humidity 'KWS Relative Humidity'
[11:33:28][C][bme280.sensor:163]:     Device Class: 'humidity'
[11:33:28][C][bme280.sensor:163]:     State Class: 'measurement'
[11:33:28][C][bme280.sensor:163]:     Unit of Measurement: '%'
[11:33:28][C][bme280.sensor:163]:     Accuracy Decimals: 1
[11:33:28][C][bme280.sensor:164]:     Oversampling: 2x`

jcastro, 2022-02-11:
I think this is the problem
`[11:33:27][C][i2c.arduino:046]:   Recovery: failed, SCL is held low on the bus`
Can you try changing SCL/SDA pins to for example 16/17 (on ESP32)

Serios, 2022-02-11:
> I think this is the problem `[11:33:27][C][i2c.arduino:046]: Recovery: failed, SCL is held low on the bus` Can you try changing SCL/SDA pins to for example 16/17 (on ESP32)

I see that too, and wondering where it came from since I didn't change the wiring nor the configuration. I can't change the pins, the since the board is cheapo esp01 not esp32

Serios, 2022-03-09:
Just updated to version 2022.2.6 and the issue still exist. Even tried to clean build files first and then recompile again, results are the same

Calimerorulez, 2022-03-14:
Same issue here with an SD1306 display and a SHT3xD sensor:

`[08:05:43][C][i2c.arduino:050]:   Recovery: failed, SDA is held low on the bus`

jcastro, 2022-03-14:
Same here with the SHT3X too, and still having it on the BME280 as well.

Calimerorulez, 2022-03-14:
When I power cycle the device it temporarily works dagain

Rasterpaal, 2022-04-01:
I've the same issue now with HASS and ESPHome 2022.3.2 
Is there any way we can downgrade the addon of ESPHome or install a different version so I can try to upload an older version of it?

zelig1971, 2022-04-01:
Same issue, ESPHome 2022.1.4, Home Assistant 2022.3.8. Both on ESP8266 and ESP32.

Serios, 2022-04-01:
I have my instance installed in venv and couldn't find any info how to downgrade to previous version of epshome. 

Since my home heating system depends on automations based on data from BME280 sensors, for the last couple of months it is pain in you know where, to fiddle with it, daily. I think there should be some some way to downgrade to 2021.12.3. 

I already know, that backup before upgrading is a must... but mistakes where made...

dadav, 2022-04-25:
I have the same error with different components:

```yaml
esphome:
  name: air1
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "yyy"
  password: !secret wifi_pass
  power_save_mode: none
  reboot_timeout: 30min
  fast_connect: on

# Enable logging
logger:

# Enable Home Assistant API
api:
  services:
    - service: sensair_background_calibrate
      then:
        - senseair.background_calibration: senseair_s8
    - service: sensair_background_calibration_result
      then:
        - senseair.background_calibration_result: senseair_s8

ota:
  password: "xxx"

captive_portal:

i2c:
  sda: D2
  scl: D1

font:
  - file: "fonts/monofont.ttf"
    id: opensans
    size: 10

display:
  - platform: ssd1306_i2c
    id: oled
    model: "SSD1306 64x48"
    pages:
      - id: page1
        lambda: |-
          it.printf(0, 0, id(opensans), "CO2: %.0fppm", id(co2).state);
          it.printf(0, 10, id(opensans), "PM25: %.0f", id(pm25).state);
          it.printf(0, 20, id(opensans), "Hmdty: %.0f", id(humidity).state);
          it.printf(0, 30, id(opensans), "Temp: %.0fC", id(temp).state);
# Maybe add a page later

interval:
  - interval: 5s
    then:
      - display.page.show_next: oled
      - component.update: oled

uart:
  - rx_pin: D5
    tx_pin: D6
    baud_rate: 9600
    id: uart1
    
  - rx_pin: D4
    tx_pin: D3
    baud_rate: 9600
[... 135 lines cut ...]
[08:04:56][I][i2c.arduino:056]: Unknown error at address 0x6A
[08:04:56][I][i2c.arduino:056]: Unknown error at address 0x6B
[08:04:56][I][i2c.arduino:056]: Unknown error at address 0x6C
[08:04:56][I][i2c.arduino:056]: Unknown error at address 0x6D
[08:04:56][I][i2c.arduino:056]: Unknown error at address 0x6E
[08:04:56][I][i2c.arduino:056]: Unknown error at address 0x6F
[08:04:56][I][i2c.arduino:056]: Unknown error at address 0x70
[08:04:56][I][i2c.arduino:056]: Unknown error at address 0x71
[08:04:56][I][i2c.arduino:056]: Unknown error at address 0x72
[08:04:56][I][i2c.arduino:056]: Unknown error at address 0x73
[08:04:56][I][i2c.arduino:056]: Unknown error at address 0x74
[08:04:56][I][i2c.arduino:056]: Unknown error at address 0x75
[08:04:56][I][i2c.arduino:056]: Unknown error at address 0x76
[08:04:56][I][i2c.arduino:056]: Unknown error at address 0x77
[08:04:56][I][i2c.arduino:060]: Found no i2c devices!
[08:04:56][D][pmsx003:204]: Got PM2.5 Concentration: 23 µg/m^3, Temperature: 0.0°C, Humidity: 0.0%
[08:04:56][D][sensor:113]: 'Büro Feinstaub <2.5µm': Sending state 23.00000 µg/m³ with 0 decimals of accuracy
```

I'm using: 

HA: 2022.4.6
Esphome: 2022.3.1

I didn't change any wireing nor configuration. I can't tell since which update this happens, because I noticed it just recently...

shakin89, 2022-07-07:
I have HA 2022.07 and the issue is still present.
I've resoldered all the wirings several times. but i can't resolve this issue.
Anyone know how to solve?

jkeys1989, 2022-07-11:
> I have HA 2022.07 and the issue is still present. I've resoldered all the wirings several times. but i can't resolve this issue. Anyone know how to solve?

I'm wondering if building a custom  sensor component will work.  I tried 2 different bme280's with no luck.

jkeys1989, 2022-07-11:
> 

I tried the following. 
i2c: 
# Enable logging
  sda: GPIO21
  scl: GPIO22
  scan: false 
  id: bus_a

sensor:
  - platform: bme280
    i2c_id: bus_a
    temperature:
      name: "BME280 Temperature"
    pressure:
      name: "BME280 Pressure"
    humidity:
      name: "BME280 Relative Humidity"
    address: 0x77
    update_interval: 15s

**Error below.**
[18:23:31][C][logger:276]:   Level: DEBUG
[18:23:31][C][logger:277]:   Log Baud Rate: 115200
[18:23:31][C][logger:278]:   Hardware UART: UART0
[18:23:31][C][i2c.arduino:038]: I2C Bus:
[18:23:31][C][i2c.arduino:039]:   SDA Pin: GPIO21
[18:23:31][C][i2c.arduino:040]:   SCL Pin: GPIO22
[18:23:31][C][i2c.arduino:041]:   Frequency: 50000 Hz
[18:23:31][C][i2c.arduino:050]:   Recovery: failed, SDA is held low on the bus
[18:23:31][C][bme280.sensor:148]: BME280:
[18:23:31][C][bme280.sensor:149]:   Address: 0x77
[18:23:31][E][bme280.sensor:152]: Communication with BME280 failed!

shakin89, 2022-07-12:
@Serios thanks for your tip. I'll try it for sure.
Meanwhile, the issue seems to be resolved for me.... 
i have an esp32 devkit c4, i'ìve changed pins scl to GPIO25 and sda to GPIO26. With this configuration the problem has gone...

Serios, 2022-07-12:
@jkeys1989 check the wiring and make sure that sda and scl are set to the proper pin numbers, also try to use 0x76 address in bme280 configuration. I don't know if update_interval and oversampling can cause this, but you can try to increase update interval to default 60s and decrease oversampling to 2x for example

jkeys1989, 2022-07-12:
> @jkeys1989 check your wiring and also make sure that sda and scl are set to the proper pin numbers, also try to use 0x76 address in bme280 configuration. I don't know if update_interval and oversampling can cause this, but you can try to increase update interval to default 60s and decrease oversampling to 2x for example

Thanks wiring is correct, but still the same error . Tried changing the address and updating interval.

Serios, 2022-07-12:
@jkeys1989 Do you have by any chance a spare BME280 module to test with, just to take out the possibility of faulty module out of the equation. Otherwise I'm really out of ideas what can cause this... Indeed there was an issue with the esphome code but it was resolved couple of releases ago.

marcinwasilewski, 2022-07-29:
I was experiencing exact the same issue on esp32-cam and BMP280 + STH30 but I found the solution in this thread: I disabled scanning in i2c. 
If I enable scanning, STH30 won't work together with BMP280 - but they will work separately. I can keep them wired, and both will be detected by i2c scan, but as soon as I initialize them both in the config it all breaks down at the scan.

shakin89, 2022-08-05:
> @Serios thanks for your tip. I'll try it for sure. Meanwhile, the issue seems to be resolved for me.... i have an esp32 devkit c4, i'ìve changed pins scl to GPIO25 and sda to GPIO26. With this configuration the problem has gone...

well, just one week of correct functioning... and then the bme280 has stopped working. Disabled scanning, remade the wiring for three time, my meteo station does not work as it should. Will try to buy a new spare one, but i'm thinking of replacing with something other.

jkeys1989, 2022-09-23:
> @jkeys1989 Do you have by any chance a spare BME280 module to test with, just to take out the possibility of faulty module out of the equation. Otherwise I'm really out of ideas what can cause this... Indeed there was an issue with the esphome code but it was resolved couple of releases ago.

I tried a different bme280 same thing. I also tried updating esphome, same error.

BobP2020, 2022-11-21:
I just ran across this discussion of problems with BME280 not working with current release of ESPHOME (22.11.1) as I am having the same problem. I am using an ESPRESSIF ESP32-WROOM 32U. I started with the default pin assignment,
i2c:
  sda: 21
  scl: 22
  scan: True
  id: bus_a
As suggested in the previous comment, I solved the problem by moving slc over to GPIO pin 18

My configuration file is now,
i2c:
  sda: 21
  scl: 18
  scan: True
  id: bus_a

Thanks for the discussion and thanks for insight to a workaround until the probable i2c bug in ESPHOME  (22.11.1) is resolved.

Micha-Btz, 2022-12-28:
I have this problem too, but I don't understand that I can chnage so easy the slc to an other pin. If I understand it correctly, my  wemos_d1_mini32 has only two i2c pins 21 & 22.

How can we help to find the root cause of this problem?
[blehome.yaml.zip](https://github.com/esphome/issues/files/10315659/blehome.yaml.zip)

jkeys1989, 2022-12-28:
Seems like the first 2 bme280's I purchased were defective.  I purchased 2 more from a different vendor and  working now.

Serios, 2022-12-29:
@jkeys1989 I'm glad you sorted it out, usually my go to rule when ordering some cheap modules is always buy+2. That way there is always a spare module, just in case...

federoRE, 2023-04-01:
Hi. IDK if it can be useful to someone but I want to share my config (used on a NodeMCU v3 ESP8266).
```yaml
i2c:
  sda: D2
  scl: D1
  scan: True
  
sensor:
  - platform: bmp280
    address: 0x76
    temperature:
      name: "Temperatura test"
      oversampling: 16x
    pressure:
      name: "Pressione test"
    update_interval: 60s
```

[image]
PS: make sure you are using the proper pins in case you are not using NodeMCU like me.

devsef, 2024-11-20:
Hi. I faced the same problem and in my case it is an ESP32 S3 Board and a AHT20+BMP280 Modul. After reading here and in ESPhome community a lot of people were assuming it is a wiring problem. In my case it was a voltage problem. The 5V connector on the board had only 1.35V. After connecting it to the 3.3V output the sensor started to work. And no matter if scan is true or what ever was mentioned here and there. So I recommend not only checking the wiring but also the voltage.