esphome/issues #3029: Brand new HiLetgo ESP-WROOM-32 crashes on boot with Wi-Fi enabled

ashald, 2022-02-09:
### The problem

Using following config:

Attempting to flash a brand new `HiLetgo ESP-WROOM-32 ESP32 ESP-32S` with below config using
```
$  sudo esphome run --device /dev/ttyUSB0 test-01.yaml
```
causes a crash on boot.

Removing Wi-Fi section it works OK and I can see the uptime sensor reporting updates over serial connection.

I have 2 boards like that running for about a year, and just bought a new one, but got the above issue with minimal config.
Suspected a faulty board an ordered another one, but got the same behavior.
Don't feel confident enough to try re-flashing any of the old boards at this point. 😅

Versions:
```
Package               Version
--------------------- -----------
aioesphomeapi         10.6.0
aiofiles              0.8.0
ajsonrpc              1.2.0
anyio                 3.5.0
asgiref               3.5.0
bitstring             3.1.9
bottle                0.12.19
certifi               2021.10.8
cffi                  1.15.0
charset-normalizer    2.0.10
click                 8.0.3
colorama              0.4.4
cryptography          3.4.8
ecdsa                 0.17.0
esphome               2022.1.2
esphome-dashboard     20220116.0
esptool               3.2
future                0.18.2
h11                   0.13.0
idna                  3.3
ifaddr                0.1.7
kconfiglib            13.7.1
marshmallow           3.14.1
noiseprotocol         0.3.1
paho-mqtt             1.6.1
Pillow                9.0.0
pip                   20.3.4
pkg-resources         0.0.0
platformio            5.2.4
protobuf              3.19.4
pycparser             2.21
pyelftools            0.27
pyparsing             2.3.1
pyserial              3.5
pytz-deprecation-shim 0.1.0.post0
PyYAML                6.0
reedsolo              1.5.4
requests              2.27.1
semantic-version      2.8.5
setuptools            44.1.1
six                   1.16.0
sniffio               1.2.0
starlette             0.17.1
tabulate              0.8.9
tornado               6.1
tzdata                2021.5
tzlocal               4.1
urllib3               1.26.8
uvicorn               0.16.0
voluptuous            0.12.2
[... 130 lines cut ...]
WARNING Decoded 0x400d9326: loopTask(void*) at /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:18
WARNING Decoded 0x40089afa: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[03:19:21]
[03:19:21]Rebooting...
[03:19:21]ets Jun  8 2016 00:22:57
[03:19:21]
[03:19:21]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[03:19:21]configsip: 0, SPIWP:0xee
[03:19:21]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[03:19:21]mode:DIO, clock div:2
[03:19:21]load:0x3fff0018,len:4
[03:19:21]load:0x3fff001c,len:1044
[03:19:21]load:0x40078000,len:10124
[03:19:21]load:0x40080400,len:5828
[03:19:21]entry 0x400806a8
[03:19:21][I][logger:214]: Log initialized
[03:19:21][I][app:029]: Running through setup()...
[03:19:21][C][wifi:037]: Setting up WiFi...
[03:19:22][D][wifi:370]: Sta\xe08~>\xcb\xeb\x81scan...
```
```

### Additional information

I flashed [waveshare Universal e-Paper Raw Panel Driver Board, ESP32 WiFi / Bluetooth Wireless](https://www.waveshare.com/e-Paper-ESP32-Driver-Board.htm) and `Lilygo T5-4.7` within last few days without an issue.

ashald, 2022-02-09:
Interestingly enough I git a boot button on the MCU and it booted OK. I tried pressing it again and it crashed again. Tried few more times - crashing again with different errors.
At this point I'm not sure if I got 2 faulty MCUs, or if there is some bug at play?
```
[03:32:18]ets Jun  8 2016 00:22:57
[03:32:18]
[03:32:18]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[03:32:18]configsip: 0, SPIWP:0xee
[03:32:18]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[03:32:18]mode:DIO, clock div:2
[03:32:18]load:0x3fff0018,len:4
[03:32:18]load:0x3fff001c,len:1044
[03:32:18]load:0x40078000,len:10124
[03:32:18]load:0x40080400,len:5828
[03:32:18]entry 0x400806a8
[03:32:18][I][logger:214]: Log initialized
[03:32:18][I][app:029]: Running through setup()...
[03:32:18][C][wifi:037]: Setting up WiFi...
[03:32:19][D][wifi:370]: Sta\xf2\xf0\xfc&\xcb\xeb\x81scan...
[03:32:19][D][sensor:125]: 'Uptime Sensor': Sending state 0.16900 s with 0 decimals of accuracy
[03:32:19]Guru Meditation Error: Core  panic'ed (Interrupt wdt timeout on CPU0)
[03:32:19]Core 0 register dump:
[03:32:19]PC       0x4013bf  PS       0x00060a4  A0       0x8013b4a A1       0x3fb890
[03:32:19]A2       0x3fab0  A3       0x00000000  A4       0x0005ae  A5       0x00000000
[03:32:19]A6       0x3fafc A7       0x00060023  A8       0x00000000  A9       0x00064ab
[03:32:19]A10      0x0005ae  A11      0x00000000  A12      0x00000001  A13      0x00000001
[03:32:19]A14      0x00060a0  A15      0x00000000  SAR      0x00000000  EXCCAUSE 0x00000005
[03:32:19]EXCVADDR 0x00000000  LBEG     0x4000ce  LEND     0x4000cf  LCOUNT   0xffff
[03:32:19]
[03:32:19]ELF file SHA256: 000000000000000
[03:32:19]
[03:32:19]Backtrace:0x4013bf:0x3fb8900x4013b47:0x3fb8b 0x40089aa0x3fb8d
[03:32:19]
[03:32:19]ore 1 register dump:
[03:32:19]PC       0x4000c71  PS       0x00060034  A0       0x8008a04  A1       0x3fbc0
[03:32:19]A2       0x3fb058  A3       0x3f13998  A4       0x00000014  A5       0x3fbe0
[03:32:19]A6       0x3fbe8  A7       0x00000001  A8       0x00000001  A9       0x3f13999
[03:32:19]A10      0x000000a  A11      0x00000000  A12      0x8008bb  A13      0x3fbe0
[03:32:19]A14      0x00000008  A15      0x00000001  SAR      0x0000001e EXCCAUSE 0x00000005
[03:32:19]EXCVADDR 0x00000000  LBEG     0x400014f  LEND     0x4000150d LCOUNT   0xffff
[03:32:19]
[03:32:19]ELF file SHA256: 000000000000000
[03:32:19]
[03:32:19]Backtrace:0x4000c71:0x3fbc00x4008a01:0x3fbc00x4008ca:0x3fbc00x4008c62:0x00000000
[03:32:19]
[03:32:19]Rebooting...
[03:32:26]ets Jun  8 2016 00:22:57
[03:32:26]
[03:32:26]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[03:32:26]configsip: 0, SPIWP:0xee
[03:32:26]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[03:32:26]mode:DIO, clock div:2
[03:32:26]load:0x3fff0018,len:4
[03:32:26]load:0x3fff001c,len:1044
[03:32:26]load:0x40078000,len:10124
[03:32:26]load:0x40080400,len:5828
[03:32:26]entry 0x400806a8
[03:32:27][I][logger:214]: Log initialized
[03:32:27][I][app:029]: Running through setup()...
[03:32:27][C][wifi:037]: Setting up WiFi...
[03:32:27][D][wifi:370]: StaW\xcb\xeb\x81scan...
[03:32:27][D][sensor:125]: 'Uptime Sensor': Sending state 0.17400 s with 0 decimals of accuracy
[03:32:27]\xf0\xaa\xc9ՁMeditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
[03:32:27]Memory dump at 0x4008c528: f01d0020 1b03Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
[03:32:27]Debug exception reason: BREAK instr
[03:32:27]Core 1 register dump:
[03:32:27]PC      : 0x400803c0  PS      : 0x00060036  A0      : 0x00000050  A1      : 0x3ffb1ce0
WARNING Decoded 0x400803c0: _DoubleExceptionVector at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:543
[03:32:27]A2      : 0x00000034  A3      : 0x00000000  A4      : 0x80081006  A5      : 0x3ffb1ce0
[03:32:27]A6      : 0x00000000  A7      : 0x00000000  A8      : 0x0000007e  A9      : 0x25440a08
[03:32:27]A10     : 0x00000000  A11     : 0x00000000  A12     : 0xfa000000  A13     : 0x00000000
[... 408 lines cut ...]
[03:32:41][I][logger:214]: Log initialized
[03:32:41][I][app:029]: Running through setup()...
[03:32:41][C][wifi:037]: Setting up WiFi...
[03:32:41][D][wifi:370]: Sta\xf2\xc0\x80\xf0\xf8\xe0\xe0\x80>\xfe8\x9cxcan...
[03:32:41]ur MdiatonEror Cre anc'd Inerup wt imou o CU1)
[03:32:41]or 1 egstr um:C     000c71 S     000004  A0    : x80874  A1    : x3fbb0
[03:32:41]2     0fb0058 3      0x3f413998 4      0x00000014  A5      0x3fffe0  A6     : xfbf78 7     : x0000001 8     : x0000001 9     : xf199  A10    : x00000a5 11     0x3ff1df  A12     0x8008b2bb  13     0x3fffe0  A14    : x0000008 15    : x0000001 AR    : x0000018  EXCCUSE 0x00000006
[03:32:41]ECVADR: x0000000 BEG   : x0002390 END   : x000239f LOUT   0x00000000
[03:32:41]
[03:32:41]LF fil SHA56: 000000000000
[03:32:41]
[03:32:41]Baktrace0x4000c210x3ff1bf0x4008a7010xffbc10 0x08c2ac:0x3fbc30 0x08c2620xffbc50 0407d4d:x3ff1d34
[03:32:41]
[03:32:41]Coe 0 rgisterdump
[03:32:41]P     : x0086f3a P      0x00060634 0      0x80089af  A1      0x3fb5c0
[03:32:41]A     : xfbdb8 3      0x3ffe200 4      0x00000000 5     : x000001a
[03:32:41]A      0x40107eb  A7      0x3fc24  8      0x3f409184 9      0x3ff5fa
[03:32:41]A0      0x00000017 11     0x00000000 12    : xfbd44 13    : x0000040
[03:32:41]A4    : x000003 15    : x0006f3  AR     0x0000001e ECCUSE 0x00000006
[03:32:41]ECVADR: x000000 LEG    0x4000c2e0 EN     0x4000c26  LCOUT   0xffffff
[03:32:41]
[03:32:41]LF fil SHA56: 0000000000000
[03:32:41]
[03:32:41]Bactrac: 0x4086f3a:x3fbf0 0x409af:03fbff0ebotng..
```

OttoWinter, 2022-02-09:
Do you know whether the ESP32 chip you got is a unicore chip (most ESP32 have 2 cores, but rarely there are also 1 core variants, which need a different config)

ashald, 2022-02-09:
AFAIK, it's a 2-core chip.
`esptool` says the same:
```
INFO Successfully compiled program.
esptool.py v3.2
Serial port /dev/ttyUSB0
Connecting............
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
```
Amazon link for reference https://www.amazon.com/gp/product/B0718T232Z

I was looking for other MCUs as an alternatives, and found that for _some other_ MCUs there is feedback that there is an issue with schematic, such that Wi-Fi activation creates a power surge that somehow crashes the ESP. And apparently people fix it by soldering some extra capacitor. Do you think something similar might be happening here? I wonder if I'm just unlucky and both of the units are from the same batch and have an issue (the other 4 of these units I bought a year ago so maybe something changed in the way they do them).

OttoWinter, 2022-02-09:
Yeah that could make sense, the crash does happen shortly after wifi is set up. Maybe in that case the flash reads get corrupted (or the instruction cache), resulting in the illegalinstruction exception

ashald, 2022-02-09:
If that's the case then there is nothing we can do to prevent that happening with software alone, right?
Thanks for looking into it!

closed 2022-02-09 by ashald

ashald, 2022-02-10:
@OttoWinter apologies for beating a dead horse, but I've got a follow-up question. I just sent back the MCUs, but then I noticed this on the product page from where I ordered them:
[image]

Unfortunately I do not exactly recall which one I got and I do not have a photo, but I have a feeling that both of the items I got were **32D**. Are you aware of any compatibility issues with `ESP32-WROOM-32D` which might've caused the above?

OttoWinter, 2022-02-16:
No there should not be an issue there, wroom-32d is supported