esphome/issues #3030: esp32-cam unstable connection, fails most of the time

adrianlzt, 2022-02-09:
### The problem

I have flashed esp32-cam in a AI-Thinker esp32 board and connected it to HA.

Most of the time the image it is not available, the calls to ``api/camera_proxy/`` return a HTTP 500 error.

I have disabled the cam in HA and use a python script with aioesphomeapi to test the camera alone, using a modification of the example script found in the repo ([used script](https://gist.github.com/d42b6854ce6f1e41e59144be20d3ff3c)).

The [output of the script](https://gist.github.com/adrianlzt/deafec4257ba4f1e6ccdd19b0058e6e0) after being executed during 40'.

In those 40' I was able to gather only an image 17 times, failing the rest, normally because ping timeout.

### Which version of ESPHome has the issue?

2022.1.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

ESP32-cam AI-Thinker

### Component causing the issue

cam

### Example YAML snippet

```yaml
esphome:
  name: esp32-cam

esp32:
  board: esp32cam
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "REDACTED"
  password: "REDACTED"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Cam Fallback Hotspot"
    password: "REDACTED"

captive_portal:

esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

  # Image settings
  name: 32Cam
```

### Additional information

Connecting the camera with esphome cli to get the logs works sometimes:
```
$ esphome logs conf/esp32cam.yaml
INFO Reading configuration conf/esp32cam.yaml...
WARNING GPIO0 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Found multiple options, please choose one:
  [1] /dev/ttyUSB0 (SimTech, Incorporated)
  [2] /dev/ttyUSB1 (SimTech, Incorporated)
  [3] /dev/ttyUSB2 (SimTech, Incorporated)
  [4] /dev/ttyUSB3 (SimTech, Incorporated)
  [5] /dev/ttyUSB4 (SimTech, Incorporated)
  [6] Over The Air (esp32-cam.local)
(number): 6
INFO Starting log output from esp32-cam.local using esphome API
INFO Successfully connected to esp32-cam.local
[10:34:10][I][app:102]: ESPHome version 2022.1.3 compiled on Feb  9 2022, 08:18:48
[10:34:11][C][wifi:488]: WiFi:
[10:34:22][C][wifi:350]:   Local MAC: REDACTED
[10:34:22][C][wifi:351]:   SSID: 'REDACTED'
[10:34:24][C][wifi:352]:   IP Address: 192.168.1.25
[10:34:25][C][wifi:354]:   BSSID: REDACTED
[10:34:25][C][wifi:355]:   Hostname: 'esp32-cam'
INFO esp32-cam.local: Ping timed out!
INFO Disconnected from ESPHome API for esp32-cam.local
WARNING Disconnected from API
INFO Successfully connected to esp32-cam.local
[10:34:50][D][esp32_camera:168]: Got Image: len=13041
[10:35:03][D][esp32_camera:168]: Got Image: len=13168
[10:35:12][D][esp32_camera:168]: Got Image: len=13138
[10:35:19][D][esp32_camera:168]: Got Image: len=13139
[10:35:30][D][esp32_camera:168]: Got Image: len=13155
[10:35:40][D][esp32_camera:168]: Got Image: len=13216
[10:35:52][D][esp32_camera:168]: Got Image: len=13194
[10:36:00][D][esp32_camera:168]: Got Image: len=13098
[10:36:11][D][esp32_camera:168]: Got Image: len=13196
[10:36:25][D][esp32_camera:168]: Got Image: len=13175
[10:36:30][D][esp32_camera:168]: Got Image: len=13229
[10:36:40][D][esp32_camera:168]: Got Image: len=13169
[10:36:51][D][esp32_camera:168]: Got Image: len=13139
[10:36:59][D][esp32_camera:168]: Got Image: len=13122
[10:37:10][D][esp32_camera:168]: Got Image: len=13146
[10:37:20][D][esp32_camera:168]: Got Image: len=13096
[10:37:32][D][esp32_camera:168]: Got Image: len=13187
[10:37:41][D][esp32_camera:168]: Got Image: len=13166
[10:37:50][D][esp32_camera:168]: Got Image: len=13160
[10:38:00][D][esp32_camera:168]: Got Image: len=13157
```

```
$ esphome logs conf/esp32cam.yaml
...
INFO Starting log output from esp32-cam.local using esphome API
WARNING Can't connect to ESPHome API for esp32-cam.local: Timeout while connecting to ('192.168.1.25', 6053)
INFO Trying to reconnect to esp32-cam.local in the background
```

```
 $ esphome logs conf/esp32cam.yaml
...
INFO Starting log output from esp32-cam.local using esphome API
WARNING Can't connect to ESPHome API for esp32-cam.local: Hello timed out
INFO Trying to reconnect to esp32-cam.local in the background

````

u17194907425, 2022-09-18:
I have the same issue. The device starts working and after some sort time no image is retrieved.

Btw, I notice that both camera chip and the microcontroller become quite hot over time. It's that possible root cause of the issue? It's there any way to confirm that throttling is happening due to overheat?

adrianlzt, 2022-09-19:
In [PR 2813](https://github.com/esphome/esphome/pull/2813) there is an interesting analysis that could be related (small number of sockets).

He suggest moving to Arduino 2 (https://registry.platformio.org/tools/platformio/framework-arduinoespressif32/installation).

Looks like some work is being done in [PR #3564](https://github.com/esphome/esphome/pull/3564).

Maybe testing against jesserockz-2022-203 branch could give us some light.