esphome/issues #3031: ESP32-C3 Arduino Core 2.0.0

roomme13, 2022-02-09:
### The problem

Hi,

does anyone success installing esp32-c3 with arduino core 2.0.0 ?
i want to use mqtt, if i use idf platform its not available, but when i use arduino core 2.0.0 it have an error

if i dont specify the source :
`Error: Could not find the package with 'platformio/framework-arduinoespressif32 @ ~3.20000.0' requirements for your system 'linux_armv7l'`

if i use the source,
`Error: This board doesn't support arduino framework!`

is there away to make it work? or it still not supported yet?

### Which version of ESPHome has the issue?

2022.1.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

ESP32-C3

### Example YAML snippet

```yaml
esphome:
  name: c3test
  platformio_options:
    board_build.flash_mode: dio
  on_boot:
    - priority: -300
      then:
        - script.execute: checking
        
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino
    version: 2.0.0
    source: https://github.com/espressif/arduino-esp32.git
```

MFlasskamp, 2022-02-16:
With a more recent version of Arduino Core it's working. This is the snippet I'm using with LilyGo TTGO T-01C3 ESP32-C3 module:

```yaml
esphome:
  name: myesp32c3
  platformio_options:
    board_build.variant: esp32c3
    board_build.flash_mode: dio

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino
    version: dev
    platform_version: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
```

I then flashed the `firmware-factory.bin` with esptool.

roomme13, 2022-02-17:
@MFlasskamp i use same board but revision version maybe TTGO T-IO Plus

`INFO Reading configuration /config/c3.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing myesp32c3 (board: esp32-c3-devkitm-1; framework: arduino; platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream)
--------------------------------------------------------------------------------
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-arduinoespressif32 0.0.0+sha.50e9772 
 - toolchain-riscv32-esp 8.4.0+2021r2
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <WiFi> 2.0.0
|-- <ESPmDNS> 2.0.0
|-- <Update> 2.0.0
Compiling .pioenvs/myesp32c3/src/esphome/components/adc/adc_sensor.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
Compiling .pioenvs/myesp32c3/src/esphome/components/api/api_connection.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
Compiling .pioenvs/myesp32c3/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
*** [.pioenvs/myesp32c3/src/esphome/components/adc/adc_sensor.cpp.o] Error 127
Compiling .pioenvs/myesp32c3/src/esphome/components/api/api_pb2.cpp.o
*** [.pioenvs/myesp32c3/src/esphome/components/api/api_connection.cpp.o] Error 127
*** [.pioenvs/myesp32c3/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
sh: 1: riscv32-esp-elf-g++: not found
*** [.pioenvs/myesp32c3/src/esphome/components/api/api_pb2.cpp.o] Error 127
========================== [FAILED] Took 3.34 seconds ==========================`

But i got this error with your template,
i can get it work with another template but the I2C or MQTT never worked for me.

MFlasskamp, 2022-02-18:
Your board is different, but that's not the problem. There is something mixed up with your esphome build. Did you try "Clean Build Files" in ESPHome?
Your config file is `/config/c3.yaml` but the working directory is `myesp32c3` (name from my example).

The example extended with MQTT and I2C:
```yaml
esphome:
  name: myesp32c3
  platformio_options:
    board_build.variant: esp32c3
    board_build.flash_mode: dio

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino
    version: dev
    platform_version: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass

mqtt:
  broker: !secret mqtt_ip

i2c:
  sda: GPIO8
  scl: GPIO2
```
And for reference the compile log:
```
INFO Reading configuration /config/myesp32c3.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING GPIO8 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO8 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing myesp32c3 (board: esp32-c3-devkitm-1; framework: arduino; platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream)
--------------------------------------------------------------------------------
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-arduinoespressif32 0.0.0+sha.6a7bcab 
 - toolchain-riscv32-esp 8.4.0+2021r2
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 2.0.0
|-- <ESPmDNS> 2.0.0
|-- <AsyncMqttClient-esphome> 0.8.6
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <Wire> 2.0.0
|-- <ArduinoJson> 6.18.5
Compiling .pioenvs/myesp32c3/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/myesp32c3/src/esphome/components/esp32/gpio_arduino.cpp.o
[... 99 lines cut ...]
Compiling .pioenvs/myesp32c3/FrameworkArduino/esp32-hal-sigmadelta.c.o
Compiling .pioenvs/myesp32c3/FrameworkArduino/esp32-hal-spi.c.o
Compiling .pioenvs/myesp32c3/FrameworkArduino/esp32-hal-time.c.o
Compiling .pioenvs/myesp32c3/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pioenvs/myesp32c3/FrameworkArduino/esp32-hal-tinyusb.c.o
Compiling .pioenvs/myesp32c3/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pioenvs/myesp32c3/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pioenvs/myesp32c3/FrameworkArduino/firmware_msc_fat.c.o
Compiling .pioenvs/myesp32c3/FrameworkArduino/libb64/cdecode.c.o
Compiling .pioenvs/myesp32c3/FrameworkArduino/libb64/cencode.c.o
Compiling .pioenvs/myesp32c3/FrameworkArduino/main.cpp.o
Compiling .pioenvs/myesp32c3/FrameworkArduino/stdlib_noniso.c.o
Compiling .pioenvs/myesp32c3/FrameworkArduino/wiring_pulse.c.o
Compiling .pioenvs/myesp32c3/FrameworkArduino/wiring_shift.c.o
Archiving .pioenvs/myesp32c3/libFrameworkArduino.a
Linking .pioenvs/myesp32c3/firmware.elf
RAM:   [=         ]  11.0% (used 36120 bytes from 327680 bytes)
Flash: [====      ]  39.0% (used 715040 bytes from 1835008 bytes)
Building .pioenvs/myesp32c3/firmware.bin
esp32_create_combined_bin([".pioenvs/myesp32c3/firmware.bin"], [".pioenvs/myesp32c3/firmware.elf"])
Flash params set to 0x022f
Wrote 0xbfaa0 bytes to file /config/.esphome/build/myesp32c3/.pioenvs/myesp32c3/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 58.01 seconds =========================
INFO Successfully compiled program.
```

roomme13, 2022-02-20:
@MFlasskamp Yes always clean build first.
BTW i tried your configuration, but the result same.
```
INFO Reading configuration /config/testgain.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing myesp32c3 (board: esp32-c3-devkitm-1; framework: arduino; platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream)
--------------------------------------------------------------------------------
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-arduinoespressif32 0.0.0+sha.50e9772 
 - toolchain-riscv32-esp 8.4.0+2021r2
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 2.0.0
|-- <ESPmDNS> 2.0.0
|-- <AsyncMqttClient-esphome> 0.8.6
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <Wire> 2.0.0
|-- <ArduinoJson> 6.18.5
Compiling .pioenvs/myesp32c3/src/esphome/components/esp32/core.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
Compiling .pioenvs/myesp32c3/src/esphome/components/esp32/gpio_arduino.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
Compiling .pioenvs/myesp32c3/src/esphome/components/esp32/gpio_idf.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
*** [.pioenvs/myesp32c3/src/esphome/components/esp32/core.cpp.o] Error 127
*** [.pioenvs/myesp32c3/src/esphome/components/esp32/gpio_arduino.cpp.o] Error 127
*** [.pioenvs/myesp32c3/src/esphome/components/esp32/gpio_idf.cpp.o] Error 127
Compiling .pioenvs/myesp32c3/src/esphome/components/esp32/preferences.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
*** [.pioenvs/myesp32c3/src/esphome/components/esp32/preferences.cpp.o] Error 127
========================== [FAILED] Took 3.07 seconds ==========================
```
created new yaml and used your config , only change the pin for SDA & SCL,

OttoWinter, 2022-02-20:
The toolchains are managed by platformio and uploaded by espressif afaik. Those errors point to an issue with how they were uploaded to those projects.

But considering arduino-esp32 @ 2.0.0 has not officially been released yet by platformio (unfortunately), that could just be an issue that will be solved by the time it's released

jddonovan, 2023-06-09:
I realise this is old, but still open.

1st: idf version has MQTT support now.

2nd: First time c3 targeted compiled happily with the arduino libraries. I used the default parameters:

esphome:
  name: hx711
  friendly_name: HX711

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

- Using ESPHOME v2023.6.0-dev.
- Arduino libs are pulled from espressif, looks to be  2.0.9 
- platformio/espressif32@5.3.0

Try again, and perhaps close the ticket?