esphome/issues #3033: IF with multi condition

Pro-Control, 2022-02-11:
### The problem

Hi
as the code below I try to turn off the relay switch when my all light goes off after 1m but is not work

any help

thanks

### Which version of ESPHome has the issue?

v2022.1.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.2.5

### What platform are you using?

ESP8266

### Board

d1

### Component causing the issue

Automations Switch Light 

### Example YAML snippet

```yaml
light:

  - platform: rgb
    name: "$devicename Right Strip"
    id: right_strip_rgb
    red: right_strip_red
    green: right_strip_green
    blue: right_strip_blue
    on_turn_on:
    - switch.turn_on: relay
    on_turn_off:
    - if:
        condition:
          - light.is_off: "right_strip_rgb"
          - light.is_off: "lift_strip_rgb"
          - light.is_off: "down_strip_rgb"
          - light.is_off: "top_strip_ww"
          - light.is_off: "down_strip_ww"
        then:
          - delay: 60s
          - switch.turn_off: relay

  - platform: rgb
    name: "$devicename Lift Strip"
    id: lift_strip_rgb
    red: lift_strip_red
    green: lift_strip_green
    blue: lift_strip_blue
    on_turn_on:
    - switch.turn_on: relay
    on_turn_off:
    - if:
        condition:
          - light.is_off: "right_strip_rgb"
          - light.is_off: "lift_strip_rgb"
          - light.is_off: "down_strip_rgb"
          - light.is_off: "top_strip_ww"
          - light.is_off: "down_strip_ww"
        then:
          - delay: 60s
          - switch.turn_off: relay

  - platform: rgb
    name: "$devicename Down Strip"
    id: down_strip_rgb
    red: down_strip_red
    green: down_strip_green
    blue: down_strip_blue
    on_turn_on:
    - switch.turn_on: relay
    on_turn_off:
    - if:
        condition:
          - light.is_off: "right_strip_rgb"
          - light.is_off: "lift_strip_rgb"
          - light.is_off: "down_strip_rgb"
          - light.is_off: "top_strip_ww"
          - light.is_off: "down_strip_ww"
        then:
          - delay: 60s
          - switch.turn_off: relay

  - platform: monochromatic
    name: "$devicename Top W W"
    id: top_strip_ww
    output: top_w_w
    on_turn_on:
    - switch.turn_on: relay
    on_turn_off:
    - if:
        condition:
          - light.is_off: "right_strip_rgb"
          - light.is_off: "lift_strip_rgb"
          - light.is_off: "down_strip_rgb"
          - light.is_off: "top_strip_ww"
          - light.is_off: "down_strip_ww"
        then:
          - delay: 60s
          - switch.turn_off: relay

  - platform: monochromatic
    name: "$devicename Down W W"
    id: down_strip_ww
    output: down_w_w
    on_turn_on:
    - switch.turn_on: relay
    on_turn_off:
    - if:
        condition:
          - light.is_off: "right_strip_rgb"
          - light.is_off: "lift_strip_rgb"
          - light.is_off: "down_strip_rgb"
          - light.is_off: "top_strip_ww"
          - light.is_off: "down_strip_ww"
        then:
          - delay: 60s
          - switch.turn_off: relay
```

nagyrobi, 2022-02-15:
You need either:
```
    on_turn_off:
    - if:
        condition:
          or:
            - light.is_off: "right_strip_rgb"
            - light.is_off: "lift_strip_rgb"
            - light.is_off: "down_strip_rgb"
            - light.is_off: "top_strip_ww"
            - light.is_off: "down_strip_ww"
```
or
```
    on_turn_off:
    - if:
        condition:
          and:
            - light.is_off: "right_strip_rgb"
            - light.is_off: "lift_strip_rgb"
            - light.is_off: "down_strip_rgb"
            - light.is_off: "top_strip_ww"
            - light.is_off: "down_strip_ww"
```

Pro-Control, 2022-02-20:
Thanks, @nagyrobi  I tried this method but not work as supposed most of time just take the first condition in my case is "right_strip_rgb" 
So I ended with this solution

`
power_supply:
  - id: 'power_supply_relay'
    enable_time: 0s
    keep_on_time: 60s
    pin:
      number: GPIO14
      inverted: false

light:

  - platform: rgb
    name: "$devicename Right Strip"
    id: "right_strip_rgb"
    red: right_strip_red
    green: right_strip_green
    blue: right_strip_blue
    effects:
      - random:
output:

  - platform: esp8266_pwm
    id: down_w_w
    pin: GPIO12
    power_supply: power_supply_relay
`

Pro-Control, 2022-02-20:
Thanks, @nagyrobi I tried this method but not work as supposed most of time just take the first condition in my case is "right_strip_rgb"
So I ended with this solution
``
power_supply:
  - id: 'power_supply_relay'
    enable_time: 0s
    keep_on_time: 60s
    pin:
      number: GPIO14
      inverted: false

light:

  - platform: rgb
    name: "$devicename Right Strip"
    id: "right_strip_rgb"
    red: right_strip_red
    green: right_strip_green
    blue: right_strip_blue
    effects:
      - random:
      
output:    
  - platform: pca9685
    channel: 0
    id: right_strip_blue
    power_supply: power_supply_relay
    
  - platform: pca9685
    channel: 1
    id: right_strip_green
    power_supply: power_supply_relay
    
  - platform: pca9685
    channel: 2
    id: right_strip_red
    power_supply: power_supply_relay
``

closed 2022-07-05 by github-actions[bot]