esphome/issues #3040: ESP32 Nextion reset very often

kulichs, 2022-02-13:
### The problem

It regularly resets every 1900ms

### Which version of ESPHome has the issue?

2022.1.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.2.2

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

nextion

### Example YAML snippet

```yaml
esphome:
  name: esp32-next-test

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

# Enable logging
logger:
#  level: VERY_VERBOSE
# Enable Home Assistant API
api:

ota:
  password: "58431bc57a8c33fb3af8b6c4f8fe638b"

wifi:
  ssid: !secret wifi_name
  password: !secret wifi_password
  
  # Optional manual IP
  manual_ip:
    static_ip: 192.168.223.116
    gateway: 192.168.223.1
    subnet: 255.255.255.0
    dns1: 192.168.223.1

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Next-Test Fallback Hotspot"
    password: "TmVCSOmuNiMW"

captive_portal:

time:
  - platform: sntp
    id: sntp_time
    timezone: Europe/Prague
[... 155 lines cut ...]
WARNING Decoded 0x40088bdc: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
WARNING Decoded 0x40088e59: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
WARNING Decoded 0x40161203: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
WARNING Decoded 0x4016124a: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
WARNING Decoded 0x40160b65: __cxa_allocate_exception at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_alloc.cc:313
WARNING Decoded 0x40160c3c: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54
WARNING Decoded 0x400db8b3: esphome::nextion::Nextion::add_no_result_to_queue_(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /config/esphome/.esphome/build/esp32-next-test/src/esphome/components/nextion/nextion.cpp:192
WARNING Decoded 0x400db9aa: esphome::nextion::Nextion::add_no_result_to_queue_with_command_(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /config/esphome/.esphome/build/esp32-next-test/src/esphome/components/nextion/nextion.cpp:938
WARNING Decoded 0x400dbb1d: esphome::nextion::Nextion::add_no_result_to_queue_with_printf_(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, ...) at /config/esphome/.esphome/build/esp32-next-test/src/esphome/components/nextion/nextion.cpp:982
WARNING Decoded 0x400ddb7a: esphome::nextion::Nextion::set_component_text(char const*, char const*) at /config/esphome/.esphome/build/esp32-next-test/src/esphome/components/nextion/nextion_commands.cpp:230
WARNING Decoded 0x400ddbed: esphome::nextion::Nextion::set_component_text_printf(char const*, char const*, ...) at /config/esphome/.esphome/build/esp32-next-test/src/esphome/components/nextion/nextion_commands.cpp:230
WARNING Decoded 0x400e5c9f: std::_Function_handler<void (esphome::nextion::Nextion&), setup()::{lambda(esphome::nextion::Nextion&)#1}>::_M_invoke(std::_Any_data const&, esphome::nextion::Nextion&) at /config/esphome/.esphome/build/esp32-next-test/src/esphome/core/gpio.h:62
 (inlined by) _M_invoke at /data/cache/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
WARNING Decoded 0x400daccf: std::function<void (esphome::nextion::Nextion&)>::operator()(esphome::nextion::Nextion&) const at /config/esphome/.esphome/build/esp32-next-test/src/esphome/components/nextion/nextion.cpp:192
 (inlined by) esphome::nextion::Nextion::update() at /config/esphome/.esphome/build/esp32-next-test/src/esphome/components/nextion/nextion.cpp:139
WARNING Decoded 0x400dacda: non-virtual thunk to esphome::nextion::Nextion::update()
WARNING Decoded 0x4016b321: std::_Function_handler<void (), esphome::PollingComponent::call_setup()::{lambda()#1}>::_M_invoke(std::_Any_data const&) at /config/esphome/.esphome/build/esp32-next-test/src/esphome/core/component.cpp:135
 (inlined by) _M_invoke at /data/cache/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
WARNING Decoded 0x400811da: std::function<void ()>::operator()() const at /config/esphome/.esphome/build/esp32-next-test/src/esphome/core/scheduler.cpp:104
 (inlined by) esphome::Scheduler::call() at /config/esphome/.esphome/build/esp32-next-test/src/esphome/core/scheduler.cpp:196
WARNING Decoded 0x400e3b99: esphome::Application::loop() at /config/esphome/.esphome/build/esp32-next-test/src/esphome/core/application.cpp:69
WARNING Decoded 0x400e5e8a: loop() at /config/esphome/.esphome/build/esp32-next-test/src/esphome/core/gpio.h:62
WARNING Decoded 0x400f0ae1: loopTask(void*) at /data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:23
WARNING Decoded 0x40089e6a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

paul-at-pebbletech, 2022-02-17:
@kulichs Suggest you add a update interval to the nextion platform section

`  - platform: nextion
    id: nextion1
    update_interval: 10s
    uart_id: uart_2`

kulichs, 2022-02-18:
@paul-at-pebbletech After the adjustment, the restart is every 4300 seconds

paul-at-pebbletech, 2022-02-19:
@kulichs I would also change the baud_rate also no hyphen on the id 
```
uart:
    id: uart_2
    rx_pin: GPIO16
    tx_pin: GPIO17
    baud_rate: 115200
```

Here is my main display loop
```
- platform: nextion
    id: disp1
    uart_id: uart_2
    update_interval: 10s
    tft_url: 'http://rhome:8123/local/boiler5.tft'
    
    lambda: |-
        ESP_LOGD("main", "enter display loop");
        if (id(delay_disp))
              {
                  id(delay_disp) = false;
                  ESP_LOGD("main", "display loop first run");
              }
        else
              {
                  ESP_LOGD("main", "display loop main run");
                  if (id(first_page)) 
                    {
                          it.goto_page("weather");
                          auto time = id(rtc_time).now();
                          auto day_text =  time.strftime("%a");
                          it.set_component_text("dayw",day_text.c_str());
                          it.set_component_text_printf("hourw","%02d:",time.hour);
                          it.set_component_text_printf("minutew","%02d",time.minute);
                          it.set_component_text_printf("statuslabelw","s: %s",id(bsched).state.c_str());
                          it.set_component_background_color("maxTemp", Color(int(id(max_background_color))) );
                          it.set_component_text_printf("maxTemp","%.0f",id(today_max).state);
                          it.set_component_background_color("minTemp", Color(int(id(min_background_color))) );
                          it.set_component_text_printf("minTemp","%.0f",id(today_min).state);
                          it.set_component_background_color("actualTemp", Color(int(id(current_background_color)) ) );
                          it.set_component_text_printf("actualTemp","%.0f",id(toptemp).state);
                          it.set_component_font_color("rainText","0");
                          it.set_component_background_color("rainText", Color(int(id(rain_background_color)) ) );
                          it.set_component_text_printf("rainText","%.0f",id(today_rain).state);
                          it.set_component_font_color("uvIndex","0");
                          it.set_component_background_color("uvIndex", Color(int(id(uv_background_color)) ) );
                          it.set_component_text_printf("uvIndex","%.0f",id(today_uv).state);
                          it.send_command_printf  ("weatherPicture.pic=%.0f",id(today_icon).state);
                          it.send_command_printf  ("windPicture.pic=%.0f",id(today_wind_icon).state);
                          it.set_component_background_color("windSpeed","65535");
                          it.set_component_text_printf("windSpeed","%.0f",id(today_wind).state); 
                          id(adjust_buttons_pagew).execute();
                          id(adjust_icons).execute();
                      } 
                  else 
                     {
                          it.goto_page("temperature");    
                          it.set_component_text_printf("topt","%2.1f",id(toptemp).state);
                          it.set_component_text_printf("bottemp","%2.1f",id(temperature_office).state);
                          it.set_component_text_printf("ladtemp","%2.1f",id(temperature_office).state);
                          it.set_component_text_printf("furtemp","%2.1f",id(temperature_office).state);
                          it.set_component_text_printf("rettemp","%2.1f",id(temperature_office).state);
                          it.set_component_text_printf("blendtemp","%2.1f",id(toptemp).state);
                          auto time = id(rtc_time).now();
                          auto day_text =  time.strftime("%a");
                          it.set_component_text("day",day_text.c_str());
                          it.set_component_text_printf("hour","%02d:",time.hour);
                          it.set_component_text_printf("minute","%02d",time.minute);
                          id(adjust_buttons_pageb).execute();
                          id(adjust_icons).execute();
                      }
     
                   id(first_page) = !id(first_page);  // Switch page
                  ESP_LOGD("main", "EXIT display loop");
              }
```

SenexCrenshaw, 2022-06-21:
Seems though you are running out of memory. 
``(1906685) MDNS: Cannot allocate memory (line: 2588, free heap: 324 bytes)``

There is a way to increase the heap size as well. I dont have the information handy at the moment though. 

Odd since you are not running a whole lot on it. What is in?
``
id(adjust_buttons_pagew).execute();
id(adjust_icons).execute();
``

I havent been very active to be patient if i dont respond. Make sure you dont hold the display loop for too long. Sometimes its better to just update the components if things actually change as well. so maybe dont change the background if it was set already...

obert0, 2022-06-27:
I have the same issue. It seem to be a memory leak du to the nextion component. I finaly pinpout the Nextion platform to drain the memory after commenting all the YAML.
The following graph show the heap free of my esp32 with some changes I make in the esphome Nextion config. First section is update internval 5sec, second with update internval 30sec and thurd without the lambda code to update the text in the Nextion.
[image]

I hope you could find the issue to this problem.

paul-at-pebbletech, 2022-06-29:
@SenexCrenshaw  `id(adjust_buttons_pagew).execute(); id(adjust_icons).execute();` are scripts that set the right background images for buttons. But I posted my code as an example of what works to help @kulichs.
BTW I had to completely give up using the display with ESP8266 as I had the random reboot problem. But since I moved to ESP32 it has been fine

obert0, 2022-07-04:
I think I resolve my issue. I had put the baud setting in nextion editor inside the initialisation of the page0 instead of in the program.s. No more merory leak and no more reboot.

kulichs, 2022-07-07:
@obert0 thanks. It solved my problem.