esphome/issues #3041: esp32 cam snapshot doesn't take new image

augonis, 2022-02-13:
### The problem

Camera snapshot service requests new image, but saves last sate image instead of waiting for new/requested image.

To replicate point camera at phone with timer running, trigger snapshot service from HA at known timer time, look at the time in saved image.

In my experiments I concluded that camera does get request for new image and takes it, but HA impatiently saves the last image to file (usually the last idle image).

### Which version of ESPHome has the issue?

2022.1.4

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.12.10

### What platform are you using?

ESP32-IDF

### Board

ESP32 CAM AI Thinker Board

### Component causing the issue

camera

### Example YAML snippet

```yaml
esphome:
  name: nomad_camera
  platform: ESP32
  board: esp32cam

wifi:
  ssid: "?????"
  password: "?????"

logger:
api:
  password: "??????"
ota:
  password: "??????"
  
  
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]

  # the order of the data_pins is significant, don't mix up the order

  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  resolution: 800x600
  name: esp_cam02
  idle_framerate: 0.1fps
```

Micha70, 2022-02-27:
I have noticed same behavior:

In esphome:

`esphome:
  name: esp32-cam
  platform: ESP32
  board: esp32dev

  on_boot:
    then:
        - logger.log: Waiting for a client
        - wait_until:
            api.connected:
        - logger.log: A client has connected, waiting a little
        - logger.log: LED on
        - output.turn_on: esp32cam_light
        - delay: 1000ms
        - logger.log: Calling snapshot service
        - homeassistant.service:
            service: camera.snapshot
            data:
              entity_id: camera.birdbox_camera
              filename: '/config/www/tmp/snapshot-birdbox.jpg'
        - delay: 1000ms
        - logger.log: Calling local_file service
        - homeassistant.service:
            service: local_file.update_file_path
            data:
              entity_id: camera.birdbox_snapshot
              file_path: '/config/www/tmp/snapshot-birdbox.jpg'
        - logger.log: Called snapshot service
        - delay: 1000ms
        - logger.log: Sleeping
        - output.turn_off: esp32cam_light
`

In homeassistant: 
`  allowlist_external_dirs:
    - /config/www/tmp/

camera:
  - platform: local_file
    name: birdbox Snapshot
    file_path: /config/www/tmp/snapshot-birdbox.jpg
`

No snapshot is taken after reboot.

nvtkaszpir, 2023-09-09:
try to set `idle_framerate: 1 fps` which is maximum allowed, it looks like then the images are taken once per second so in worst scenario it should be around 2s delay